esphome/issues #6817: ili9xxx blank screen since ESP Home 2025.2.0

Mikael2235, 2025-02-26:
### The problem

Hello, 

Since the last migration of ESP Home 2025.2.0, and after recompilation, my screen ili9xxx (ESP32 2,8'' CYD) doesn't work anymore, my yaml code has not change except I would like to display ssid network. at the compilation this monday 24th Febr it has been compiled with new ESP Home 2025.2.0

I checked the release note for version 2025.2.0, and note for ili9xxx, I have added **psram:** in my yaml code, but same white screen. logs is below : 
[09:53:55][C][psram:020]: PSRAM:
[09:53:55][C][psram:021]:   Available: NO

I also tried to add **color_palette: 8bit** but still the same blank screen. 
Tried today the latest ESP Home 2025.2.1 without more success

Much appreciated if anyone can help me to solve this issue ? 
Good day,

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ili9xxx

### YAML Config

```yaml
psram:

display:
  - platform: ili9xxx
    model: ili9341
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    auto_clear_enabled: False
    update_interval: never
    dimensions:
      height: 240
      width: 320
    rotation: 90
    invert_colors: false
    color_palette: 8bit  # Forcer le mode 8 bits si nécessaire
    lambda: |-
```

### Anything in the logs that might be useful for us?

```txt
[10:03:41][C][spi:068]: SPI bus:
[10:03:41][C][spi:069]:   CLK Pin: GPIO14
[10:03:41][C][spi:070]:   SDI Pin: GPIO12
[10:03:41][C][spi:071]:   SDO Pin: GPIO13
[10:03:41][C][spi:076]:   Using HW SPI: SPI
[10:03:41][C][spi:068]: SPI bus:
[10:03:41][C][spi:069]:   CLK Pin: GPIO25
[10:03:41][C][spi:070]:   SDI Pin: GPIO39
[10:03:41][C][spi:071]:   SDO Pin: GPIO32
[10:03:41][C][spi:076]:   Using HW SPI: HSPI
[10:03:41][C][ili9xxx:091]: ili9xxx
[10:03:41][C][ili9xxx:091]:   Rotations: 90 °
[10:03:41][C][ili9xxx:091]:   Dimensions: 240px x 320px
[10:03:41][C][ili9xxx:092]:   Width Offset: 0
[10:03:41][C][ili9xxx:093]:   Height Offset: 0
[10:03:41][C][ili9xxx:102]:   Color mode: 8bit 332 mode
[10:03:41][C][ili9xxx:108]:   Data rate: 40MHz
[10:03:41][C][ili9xxx:111]:   CS Pin: GPIO15
[10:03:41][C][ili9xxx:112]:   DC Pin: GPIO2
[10:03:41][C][ili9xxx:114]:   Color order: BGR
[10:03:41][C][ili9xxx:115]:   Swap_xy: NO
[10:03:41][C][ili9xxx:116]:   Mirror_x: YES
[10:03:41][C][ili9xxx:117]:   Mirror_y: NO
[10:03:41][C][ili9xxx:118]:   Invert colors: NO
[10:03:41][C][ili9xxx:123]:   Update Interval: never
[10:03:41][C][ledc.output:180]: LEDC Output:
[10:03:41][C][ledc.output:181]:   Pin GPIO21
[10:03:41][C][ledc.output:182]:   LEDC Channel: 0
[10:03:41][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[10:03:41][C][ledc.output:184]:   Phase angle: 0.0°
[10:03:41][C][ledc.output:185]:   Bit depth: 16
[10:03:41][C][ledc.output:180]: LEDC Output:
[10:03:41][C][ledc.output:181]:   Pin GPIO4
[10:03:41][C][ledc.output:182]:   LEDC Channel: 1
[10:03:41][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[10:03:41][C][ledc.output:184]:   Phase angle: 0.0°
[10:03:41][C][ledc.output:185]:   Bit depth: 16
[10:03:41][C][ledc.output:180]: LEDC Output:
[10:03:41][C][ledc.output:181]:   Pin GPIO16
[10:03:41][C][ledc.output:182]:   LEDC Channel: 2
[10:03:41][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[10:03:41][C][ledc.output:184]:   Phase angle: 0.0°
[10:03:41][C][ledc.output:185]:   Bit depth: 16
[10:03:41][C][ledc.output:180]: LEDC Output:
[10:03:41][C][ledc.output:181]:   Pin GPIO17
[10:03:41][C][ledc.output:182]:   LEDC Channel: 3
[10:03:41][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[10:03:41][C][ledc.output:184]:   Phase angle: 0.0°
[10:03:41][C][ledc.output:185]:   Bit depth: 16
[10:03:41][C][light:092]: Light 'Display Backlight'
[10:03:41][C][light:094]:   Default Transition Length: 1.0s
[10:03:41][C][light:095]:   Gamma Correct: 2.80
[10:03:41][C][light:092]: Light 'LED'
[10:03:41][C][light:094]:   Default Transition Length: 1.0s
[10:03:41][C][light:095]:   Gamma Correct: 2.80
[10:03:41][C][psram:020]: PSRAM:
[10:03:41][C][psram:021]:   Available: NO
[10:03:41][C][xpt2046:062]: XPT2046:
[10:03:41][C][xpt2046:064]:   IRQ Pin: GPIO36
[10:03:41][C][xpt2046:065]:   X min: 280
[10:03:41][C][xpt2046:066]:   X max: 3860
[10:03:41][C][xpt2046:067]:   Y min: 340
[10:03:41][C][xpt2046:068]:   Y max: 3860
[10:03:41][C][xpt2046:070]:   Swap X/Y: NO
[10:03:41][C][xpt2046:071]:   Invert X: YES
[10:03:41][C][xpt2046:072]:   Invert Y: NO
[10:03:41][C][xpt2046:074]:   threshold: 400
[10:03:41][C][xpt2046:076]:   Update Interval: 0.050s
```

ssieb, 2025-02-26:
You have no update interval.  How are you updating the screen?

Mikael2235, 2025-02-26:
@ssieb, I have remove update_interval for some tests. With **update_interval: 1s** it is same behavior, not totally white screen, but vertical lines on 3/4 of the screen.

Mikael2235, 2025-02-26:
Dans mes derniers logs je constate cela : 
[11:47:06][C][ili9xxx:091]: ili9xxx
[11:47:06][C][ili9xxx:091]:   Rotations: 90 °
[11:47:06][C][ili9xxx:091]:   Dimensions: 240px x 320px
[11:47:06][C][ili9xxx:092]:   Width Offset: 0
[11:47:06][C][ili9xxx:093]:   Height Offset: 0
[11:47:06][C][ili9xxx:099]:   Color mode: 16bit
[11:47:06][C][ili9xxx:108]:   Data rate: 40MHz
[11:47:06][C][ili9xxx:111]:   CS Pin: GPIO15
[11:47:06][C][ili9xxx:112]:   DC Pin: GPIO2
[11:47:06][C][ili9xxx:114]:   Color order: BGR
[11:47:06][C][ili9xxx:115]:   Swap_xy: NO
[11:47:06][C][ili9xxx:116]:   Mirror_x: YES
[11:47:06][C][ili9xxx:117]:   Mirror_y: NO
[11:47:06][C][ili9xxx:118]:   Invert colors: NO
[11:47:06][C][ili9xxx:121]:   => Failed to init Memory: YES!
[11:47:06][C][ili9xxx:123]:   Update Interval: 1.0s
**[11:47:06][E][component:082]:   Component display is marked FAILED**

ssieb, 2025-02-26:
`Failed to init Memory: YES!`
You need to find some way to free up more RAM.

clydebarrow, 2025-02-27:
1. According to the log you do *not* have 8 bit mode selected:

> [11:47:06][C][ili9xxx:099]: Color mode: 16bit

2. Rather than `rotation` use `transform`.

This config works for me, giving a landscape presentation. If it's upside down set `mirror_x` and `mirror_y` to false.

```
display:
  - platform: ili9xxx
    model: ili9341
    id: main_lcd
    cs_pin: GPIOxx
    dc_pin: GPIOxx
    reset_pin: GPIOxx
    auto_clear_enabled: false
    update_interval: 10s
    invert_colors: false
    dimensions:
      height: 240
      width: 320
    transform:
      swap_xy: true
      mirror_y: true
      mirror_x: true
    color_palette: 8bit
    show_test_card: true
```

jurgen2005, 2025-03-06:
"According to the log you do not have 8 bit mode selected"
This fixed my issue too! Thx!

closed 2025-03-06 by clydebarrow