esphome/issues #6820: PZEM-004T (V4) irregular reading from software serial port

jopl, 2025-02-27:
### The problem

Hi friends,
I have got the following issue:
Based on the yaml code:

```
...
sensor:
  - platform: pzemac
    id: pzemac_2
    address: 0x2
    modbus_id: modbus_pzem
    update_interval: 15s
    voltage:
      name: "PZEM-004T Load Voltage"
      id: pzem_load_voltage
...
```

ESPHome (with ESP32) regularly (every 15 seconds) sends the following code (from log):

...
[06:33:15][D][uart_debug:114]: >>> 02:04:00:00:00:0A:70:3E
...

but serial port reading usually occures after more than one sent message and and response from the module is multiple:

...
[06:33:30][D][uart_debug:114]: >>> 02:04:00:00:00:0A:70:3E
[06:33:30][D][pzemac:049]: PZEM AC: V=232.4 V, I=1.468 A, P=269.5 W, E=5420.0 Wh, F=50.0 Hz, PF=0.79
[06:33:30][D][sensor:093]: 'PZEM-004T Load Voltage': Sending state 232.39999 V with 1 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Current': Sending state 1.46800 A with 3 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Active Power': Sending state 269.50000 W with 2 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Energy': Sending state 5420.00000 Wh with 0 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Power Factor': Sending state 0.79000  with 2 decimals of accuracy
[06:33:30][D][pzemac:049]: PZEM AC: V=232.4 V, I=1.430 A, P=258.7 W, E=5421.0 Wh, F=50.0 Hz, PF=0.78
[06:33:30][D][sensor:093]: 'PZEM-004T Load Voltage': Sending state 232.39999 V with 1 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Current': Sending state 1.43000 A with 3 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Active Power': Sending state 258.70001 W with 2 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Energy': Sending state 5421.00000 Wh with 0 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Power Factor': Sending state 0.78000  with 2 decimals of accuracy
[06:33:30][D][pzemac:049]: PZEM AC: V=232.4 V, I=1.493 A, P=275.5 W, E=5422.0 Wh, F=50.0 Hz, PF=0.79
[06:33:30][D][sensor:093]: 'PZEM-004T Load Voltage': Sending state 232.39999 V with 1 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Current': Sending state 1.49300 A with 3 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Active Power': Sending state 275.50000 W with 2 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Energy': Sending state 5422.00000 Wh with 0 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Power Factor': Sending state 0.79000  with 2 decimals of accuracy
[06:33:30][D][pzemac:049]: PZEM AC: V=232.4 V, I=1.472 A, P=270.7 W, E=5423.0 Wh, F=50.0 Hz, PF=0.79
[06:33:30][D][sensor:093]: 'PZEM-004T Load Voltage': Sending state 232.39999 V with 1 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Current': Sending state 1.47200 A with 3 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Active Power': Sending state 270.70001 W with 2 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Energy': Sending state 5423.00000 Wh with 0 decimals of accuracy
[06:33:30][D][sensor:093]: 'PZEM-004T Load Power Factor': Sending state 0.79000  with 2 decimals of accuracy
[06:33:30][D][uart_debug:114]: <<< 02:04:14:09:14:05:BC:00:00:0A:87:00:00:15:2C:00:00:01:F4:00:4F:00:00:2A:3D:02:04:14:09:14:05:96:00:00:0A:1B:00:00:15:2D:00:00:01:F4:00:4E:00:00:99:98:02:04:14:09:14:05:D5:00:00:0A:C3:00:00:15:2E:00:00:01:F4:00:4F:00:00:3D:55:02:04:14:09:14:05:C0:00:00:0A:93:00:00:15:2F:00:00:01:F4:00:4F:00:00:EE:01:02:04:14:09:14:05:C5:00:00:0A:95:00:00:15:30:00:00:01:F4:00
...

Seems to me serial port buffer cannot recognize module response correctly and reading is waiting for some unclear moment ?

Is this behaviour standard, please? And if it is not, is there any solution to read module response after every request, please?

Thanks for response in advance.
Kind regards, Josef.

### Which version of ESPHome has the issue?

ESPHome 2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2025.2

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-02-27:
What are the uart and modbus configs?
What is between these?
```
[07:00:45][D][uart_debug:114]: >>> 02:04:00:00:00:0A:70:3E
...
[07:01:00][D][uart_debug:114]: >>> 02:04:00:00:00:0A:70:3E
```

jopl, 2025-02-27:
Config in yaml:

uart:
  - id: uart_pzem
     tx_pin: ${pzem_tx_pin}
     rx_pin: ${pzem_rx_pin}
     baud_rate: 9600
     debug:
        direction: BOTH

modbus:
  - id: modbus_pzem
     uart_id: uart_pzem

The ESP32 board is dedicated also for collecting data from inverter, therefore there are many lines not related (hopefully) to this issue.

ssieb, 2025-02-27:
Is the inverter using a different uart?  I can't see any way that it's not the pzem delaying the data send, but I don't know why it would do that.  Do you have another ESP32 that you could use to listen in to the uart lines just to check?  Or do you have a logic analyzer?

jopl, 2025-02-27:
Dear ssieb,
thank you so much for your replies and advices!
The inverter uses different UART. It has dedicated hardware one.
To be more specific, I've got 2 of PZEMs on the software uart bus described above. Both of them work similarly, i.e irregularly. But even if I don't use one of them (by yaml configuration change), behavior is the same.
What is raising to my mind is some type of sensor's data filtering, as when there are big changes on power for example, I would say that frequency of responses increases ... But it's just my perception, not exact measurements. But there is no such filter applied in the yaml code.
I could also measure and try to identify it by analyzer ;-), weekend will be here soon :-)))
Regards, Josef

ssieb, 2025-02-27:
There is no software uart on ESP32.  What if you physically remove one of the PZEMs from the uart bus?

jopl, 2025-02-27:
Ok, I'll try it and let you know ;-)
Thx. J.

randybb, 2025-02-27:
I don't see such behavior, I have 3 pzem's connected via one bus (this one is from mains, then a second setup for an electric heater).

```
[16:45:24][D][uart_debug:114]: >>> 0A:04:00:00:00:0A:71:76
[16:45:24][D][pzemac:049]: PZEM AC: V=238.6 V, I=2.388 A, P=499.3 W, E=7492378.0 Wh, F=50.0 Hz, PF=0.88
[16:45:24][D][sensor:093]: 'Mains L1 Voltage': Sending state 238.60001 V with 1 decimals of accuracy
[16:45:24][D][sensor:093]: 'Mains L1 Current': Sending state 2.38800 A with 3 decimals of accuracy
[16:45:24][D][sensor:093]: 'Mains L1 Power': Sending state 499.29999 W with 2 decimals of accuracy
[16:45:24][D][sensor:093]: 'Mains L1 Energy': Sending state 7492378.00000 Wh with 0 decimals of accuracy
[16:45:24][D][sensor:093]: 'Mains L1 Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[16:45:24][D][sensor:093]: 'Mains L1 Power Factor': Sending state 0.88000  with 2 decimals of accuracy
[16:45:24][W][component:237]: Component modbus took a long time for an operation (64 ms).
[16:45:24][W][component:238]: Components should block for at most 30 ms.
[16:45:24][D][uart_debug:114]: <<< 0A:04:14:09:52:09:54:00:00:13:81:00:00:53:1A:00:72:01:F4:00:58:00:00:B3:CD
[16:45:25][D][uart_debug:114]: >>> 0B:04:00:00:00:0A:70:A7
[16:45:25][D][pzemac:049]: PZEM AC: V=237.8 V, I=0.625 A, P=130.5 W, E=7624749.0 Wh, F=50.0 Hz, PF=0.88
[16:45:25][D][sensor:093]: 'Mains L2 Voltage': Sending state 237.80000 V with 1 decimals of accuracy
[16:45:25][D][sensor:093]: 'Mains L2 Current': Sending state 0.62500 A with 3 decimals of accuracy
[16:45:25][D][sensor:093]: 'Mains L2 Power': Sending state 130.50000 W with 2 decimals of accuracy
[16:45:25][D][sensor:093]: 'Mains L2 Energy': Sending state 7624749.00000 Wh with 0 decimals of accuracy
[16:45:25][D][sensor:093]: 'Mains L2 Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[16:45:25][D][sensor:093]: 'Mains L2 Power Factor': Sending state 0.88000  with 2 decimals of accuracy
[16:45:25][W][component:237]: Component modbus took a long time for an operation (56 ms).
[16:45:25][W][component:238]: Components should block for at most 30 ms.
[16:45:26][D][uart_debug:114]: <<< 0B:04:14:09:4A:02:71:00:00:05:19:00:00:58:2D:00:74:01:F4:00:58:00:00:8C:55
[16:45:28][D][uart_debug:114]: >>> 0C:04:00:00:00:0A:71:10
[16:45:28][D][pzemac:049]: PZEM AC: V=238.8 V, I=0.866 A, P=191.3 W, E=7308186.0 Wh, F=50.0 Hz, PF=0.93
[16:45:28][D][sensor:093]: 'Mains L3 Voltage': Sending state 238.80000 V with 1 decimals of accuracy
[16:45:28][D][sensor:093]: 'Mains L3 Current': Sending state 0.86600 A with 3 decimals of accuracy
[16:45:28][D][sensor:093]: 'Mains L3 Power': Sending state 191.30000 W with 2 decimals of accuracy
[16:45:28][D][sensor:093]: 'Mains L3 Energy': Sending state 7308186.00000 Wh with 0 decimals of accuracy
[16:45:28][D][sensor:093]: 'Mains L3 Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[16:45:28][D][sensor:093]: 'Mains L3 Power Factor': Sending state 0.93000  with 2 decimals of accuracy
[16:45:28][W][component:237]: Component modbus took a long time for an operation (63 ms).
[16:45:28][W][component:238]: Components should block for at most 30 ms.
[16:45:28][D][uart_debug:114]: <<< 0C:04:14:09:54:03:62:00:00:07:79:00:00:83:9A:00:6F:01:F4:00:5D:00:00:B7:79
```

jopl, 2025-02-27:
Dear randybb,
may I ask you for configuration yaml file content, please?
And do you use PZEM version 4?
Regards, Josef.

ssieb, 2025-02-28:
> TX of each pzem is connected via a diode.

You have a pullup as well, right?

randybb, 2025-02-28:
Found a schematics which I made 6 years ago - I think I am using both, the standard 5V unmodified for mains monitoring and 3V3 version for heating. And yes, there is a pullup resistor.

[image]

jopl, 2025-02-28:
Dear all,
version 4 hasn't got optocouplers, instead of them, it has pi121U31 isolator as input / output buffer (active in LO as well as HI levels on TX). This is the only difference from version 3, as far as I know.
Therefore diodes and pull up resistor is a need in my case, it's clear ;-). But I guess it's not need in case  of optocouplers TX, as it has open collectors as output. But it's true, to have LED for identification of active module is advantage. And next reason is level difference (5V and 3V3) on the same bus. It's also clear and specific here.
Will let you know if the measured results during weekend. Still I have no idea what could be reason of my issue :-(.
Regards, Josef.

randybb, 2025-02-28:
Ah, this one? https://vi.aliexpress.com/item/1005005021309263.html hm. Never seen that before. There is no capacities dropper, so it is a bit thinner. Would expect they will finally create a 3phase device, but no :(

Try it with one device if multiple devices on that bus are not causing issues. The diodes are good so you are not feeding the other outputs.

jopl, 2025-03-01:
Hi friends,
you are right in every aspect, diodes are need (not resistor, as it is already in place on the board (see schematic)
And as I tried to simplified the situation, I didn't mention the power source as difference. This is switched-mode power supply and it reflects decrease in self-consumption from version 3 (0.4 W) to 0.1W in this PZEM version 4.
In the meantime, I tried to experiment with number of connected PZEMs to the bus and finally seems to me the issue is connected to the capacitance and impedance of RX/TX lines. I should say I have got the level converters for these lines on my ESP32 board interface and probably 10k pullup resistors are insufficient.
I will measure it by oscilloscope, but first I have to galvanically isolate the mains, as the ESP is connected to the inverter as well.

PZEM-004T version 4.0 output schematic:
[image]

jopl, 2025-03-03:
Hi,
just to followup my issue ... the irregular readings were identified to be caused by terrible noise / interferrence from inverter power supply, which I used for all the interface. As there are frequent spikes of 6V peak to peak (10 us), this noise is spreading everywhere and is detected by ESP32 serial port as data, I think. Therefore I start to rework all the electronics connected to inverter, using isolators ADM3251E and external power supply. I will update this issue once I have final solution. From the first tests I can say it is trully helpful to have power source isolated from the inverter (POW-HVM10.2M).

jopl, 2025-03-05:
Hi,
finally the issue is solved.
There were two reasons for strange behaviour.
First was described previously, i.e. the noise from power source (inverter's 12V). It has been diminshed by external linear voltage regulator of 3.3V with LC filters for each of both PZEM-004T modules. I added also conversion of RS232 TTL bus of ESP32 to RS485 and back to RS232 TTL at the side of PZEM-004T modules. With that, I confirmed the one PZEM is working like a charm, but when I connected the second one, communication stopped again. Therefore I tried to found out the reason for that and I identified it. The RS485 convertor uses 4069 CMOS chip as the invertors (with output current of each invertor approx. 2 mA) and the datasheet's pi121U31 input current is 0.5 μA (range −10 to 10 μA). It means everything seems to be OK with two of pi121U31s. But the input current of pi12U31 is the static current and as the communication speed is 9600 bps, the current could be in several mA per one PZEM module. And this was clearly the reason for working of 1 PZEM, but not 2 of them. Therefore I solved it by replacing 4069 with 74HC14 (schmidt invertors in HC technology with output current per inverter approx. 20 mA) and everythig works now fluently.
See the final schematic of PZEM modules side.
Regards, Josef.

[image]

jopl, 2025-03-05:
The issue is therefore solved and closed.

closed 2025-03-05 by jopl

NonaSuomy, 2025-06-18:
Can you show a picture of the real final setup?