esphome/issues #6821: [Custom Component] Build error on ESPHome 2025.02

ESP4Ever, 2025-02-27:
### The problem

src/main.cpp: In function 'void setup()':
src/main.cpp:401:67: error: invalid new-expression of abstract class type 'esphome::dtr0xx_io::dtr0xx_ioGPIOPin'
dtr0xx_io_dtr0xx_iogpiopin_id = new dtr0xx_io::dtr0xx_ioGPIOPin();
^
In file included from src/esphome.h:19,
from src/main.cpp:3:
src/esphome/components/dtr0xx_io/dtr0xx_io.h:53:7: note: because the following virtual functions are pure within 'esphome::dtr0xx_io::dtr0xx_ioGPIOPin':
class dtr0xx_ioGPIOPin : public GPIOPin, public Parented<dtr0xx_ioComponent> {
^~~~~~~~~~~~~~~~
In file included from src/esphome/core/hal.h:4,
from src/esphome/core/application.h:7,
from src/esphome/components/api/api_connection.h:9,
from src/esphome.h:3,
from src/main.cpp:3:
src/esphome/core/gpio.h:61:23: note: 'virtual esphome::gpio::Flags esphome::GPIOPin::get_flags() const'
virtual gpio::Flags get_flags() const = 0;
^~~~~~~~~
src/main.cpp:430:69: error: invalid new-expression of abstract class type 'esphome::dtr0xx_io::dtr0xx_ioGPIOPin'
dtr0xx_io_dtr0xx_iogpiopin_id_2 = new dtr0xx_io::dtr0xx_ioGPIOPin();
^
*** [.pioenvs/esphome-web-3991a0/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.90 seconds ==========================

### Which version of ESPHome has the issue?

2025.02.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

dingtian.esp32-8-relay-board

### Component causing the issue

source: github://kecajtop/dtr0xx_io@master

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

Hello!

I'm not sure where the issue is and what has been changed in the ESPHome 2025.02 causing this error

clydebarrow, 2025-02-27:
You will need to get the external component maintainer to update their code.

DaniilMir, 2025-03-03:
This is a related topic https://github.com/kecajtop/dtr0xx_io/issues/10

closed 2025-03-03 by clydebarrow