esphome/issues #6829: Shelly Dimmer component - `warmup_brightness` default value mismatch- Documentation vs. Source Code

forrel35, 2025-03-01:
### The problem

The documentation for the `shelly_dimmer` component for Shelly Dimmer 2 states that the **default value for the `warmup_brightness` parameter is 0.**

However, analysis of the component's source code and logs from a Shelly Dimmer 2 device indicate that **the actual default value of `warmup_brightness` in the source code is 100.**

https://github.com/esphome/esphome/blob/f11ad9ad5b5c01712822670df6507f752abe926d/esphome/components/shelly_dimmer/shelly_dimmer.h#L67

This is confirmed by the fact that **in the Shelly Dimmer 2 device logs, when the `warmup_brightness` parameter is not defined in the YAML configuration, the value 100 appears.** This suggests that the component defaults to using a value of 100, which contradicts the information provided in the documentation.

**Expected behavior:**

The documentation for the `shelly_dimmer` component should be consistent with the source code and **correctly indicate the default value for the `warmup_brightness` parameter.**  If the default value in the source code is 100, the documentation should be updated to reflect this value. Alternatively, if the intended default value is 0, the component's source code should be corrected to align with the documentation.

**Actual behavior:**

*   **The documentation of the `shelly_dimmer` component states the default `warmup_brightness` value: 0.**
*   **The component's source code (and device logs) suggest the default `warmup_brightness` value: 100.**

**Steps to reproduce**

1. Configure the `shelly_dimmer` component in the YAML configuration, **without defining the `warmup_brightness` parameter.**
2. Flash the ESPHome firmware onto a Shelly Dimmer 2 device.
3. Check the **ESPHome logs** (ESPHome Dashboard -> Logs) or **serial console logs** of the Shelly Dimmer 2 device during startup.
4. In the logs, search for information related to the `shelly_dimmer` component and the `warmup_brightness` parameter.
5. Observe that **the `warmup_brightness` value in the logs is 100.**
6. Check the **`shelly_dimmer` component documentation** and verify the information about the default `warmup_brightness` value.
7. Observe that **the documentation states a default value of 0.**

### Which version of ESPHome has the issue?

2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.5

### What platform are you using?

ESP8266

### Board

Shelly Dimmer 2

### Component causing the issue

shelly_dimmer

### YAML Config

```yaml
substitutions:
  device_name: Shelly Dimmer 2 ID54
  device_name_lower: shelly-dimmer2-id54
  max_temp: "80" # °C

esphome:
  name: ${device_name_lower}

esp8266:
  board: esp01_1m
  restore_from_flash: true

wifi:
  ssid: !secret wifi_iot_parter_ssid
  password: !secret wifi_iot_parter_password
  domain: !secret local_domain

  manual_ip:
    static_ip: 192.168.48.154
    gateway: 192.168.48.5
    subnet: 255.255.255.0
    dns1: 192.168.48.5

  ap:
    ssid: ${device_name_lower}
    password: !secret fallback_hotspot_password
    manual_ip:
      static_ip: 192.168.4.2
      gateway: 192.168.4.1
      subnet: 255.255.255.0

captive_portal:

logger:
    baud_rate: 0

api:
  encryption:
    key: !secret api_password
  reboot_timeout: 0s

ota:
  - platform: esphome
    password: !secret ota_password

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${device_name} - IP Address
    ssid:
      name: ${device_name} - Connected SSID
    bssid:
      name: ${device_name} - Connected BSSID
    mac_address:
      name: ${device_name} - Device MAC Address

  - platform: version
    name: ${device_name} - ESPHome Version

switch:
  - platform: gpio
    name: ${device_name} - Automations ON/OFF
    icon: mdi:cogs
    pin: GPIO0 # use dummy GPIO so we can use restore_mode default_on
    restore_mode: RESTORE_DEFAULT_ON
    id: automations_switch

button:
  - platform: restart
    id: restart_button
    name: ${device_name} - Restart

uart:
    tx_pin: 1
    rx_pin: 3
    baud_rate: 115200

light:
    - platform: shelly_dimmer
      name: ${device_name}
      id: dimmer
      power:
        name: ${device_name} - Light Power
      voltage:
        name: ${device_name} - Light Voltage
      current:
        name: ${device_name} - Light Current
      min_brightness: 20
      max_brightness: 2500
      leading_edge: false #choose between leading edge and trailing edge (use trailing edge for led dimming)
      restore_mode: RESTORE_DEFAULT_OFF
      default_transition_length: 1s
      gamma_correct: 2.8 #change this to your liking. Default value is 2.8, but 0 prevents some brightness pops for me
      firmware:
        version: "51.6"
        update: false

sensor: #Important: don't change this sensor-part unless you know what you are doing. These sensors will shut the light down when overheating temperature is reached.
  # NTC Temperature
  - platform: ntc
    sensor: temp_resistance_reading
    name: Temperature
    id: temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
    on_value:
      then:
        - if:
            condition:
              - sensor.in_range:
                  id: temperature
                  above: ${max_temp}
              - light.is_on: dimmer
            then:
              - light.turn_off: 
                  id: dimmer
              - logger.log: "Switch turned off because temperature exceeded ${max_temp}°C"
              - homeassistant.service:
                  service: persistent_notification.create
                  data:
                    title: Message from ${device_name}
                  data_template:
                    message: Switch turned off because temperature exceeded ${max_temp}°C
    on_value_range:
      - above: ${max_temp}
        then:
          - logger.log: "Temperature exceeded ${max_temp}°C"
          - homeassistant.service:
              service: persistent_notification.create
              data:
                title: Message from ${device_name}
              data_template:
                message: Temperature exceeded ${max_temp}°C

  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm

  - platform: adc
    id: temp_analog_reading
    pin: A0

  - platform: uptime
    name: ${device_name} - Uptime Sensor
    id: uptime_sec

  - platform: wifi_signal
    name: ${device_name} - WiFi Signal
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[16:54:31][C][shelly_dimmer:118]: ShellyDimmer:
[16:54:31][C][shelly_dimmer:119]:   NRST Pin: GPIO5
[16:54:31][C][shelly_dimmer:120]:   BOOT0 Pin: GPIO4
[16:54:31][C][shelly_dimmer:122]:   Leading Edge: NO
[16:54:31][C][shelly_dimmer:123]:   Warmup Brightness: 100
[16:54:31][C][shelly_dimmer:126]:   Minimum Brightness: 20
[16:54:31][C][shelly_dimmer:127]:   Maximum Brightness: 2500
[16:54:31][C][shelly_dimmer:129]:   Update Interval: 10.0s
[16:54:31][C][shelly_dimmer:131]:   STM32 current firmware version: 51.6 
[16:54:31][C][shelly_dimmer:132]:   STM32 required firmware version: 51.6
```

swoboda1337, 2025-03-01:
The source code value is irrelevant as the python default will override it: https://github.com/esphome/esphome/blob/f11ad9ad5b5c01712822670df6507f752abe926d/esphome/components/shelly_dimmer/light.py#L158

Can you can create docs PR? 

https://esphome.io/guides/contributing.html

forrel35, 2025-03-02:
> The source code value is irrelevant as the python default will override it: https://github.com/esphome/esphome/blob/f11ad9ad5b5c01712822670df6507f752abe926d/esphome/components/shelly_dimmer/light.py#L158
> 
> Can you can create docs PR?
> 
> https://esphome.io/guides/contributing.html

Yes, I am able to edit the ESPHome documentation. However, it's not clear whether the default value of 100 for warmup_brightness is intentional or if it is a bug and the default should indeed be 0.

If the author intended 0 as the default value, then a change in the source code would be necessary, and not just in the documentation.