esphome/issues #6830: I can't get new Sonoff S31s to provide any power data or voltage readings

grabdavid, 2025-03-02:
### The problem

First of all i'm a total noob so please exuse me ... I installed Tasmota on 4 Sonoff S31s and they worked perfectly.  I just ordered 4 new ones but I couldn't get any power or voltage readings.  So I took ther opportunity to learn and switch them to ESPHome but I am still getting no data. From my research, I suspect there might be an issue with the UART which was changed to EVEN in 2024.  I would like to try ODD and NONE but I get the compiler error " cse7766 requires parity EVEN for the uart referenced by uart_id".  I am trying to learn how to override this and I would appreciate any input regarding the CSE7766 and tips on how I can debug the S31s.  
CSE7766 hard to use since ESPHome 2024.2.0 #5501:  https://github.com/esphome/issues/issues/5501
even parity is required:  https://github.com/esphome/esphome-docs/pull/4308

### Which version of ESPHome has the issue?

 ESPHome Docker image version  2025.2.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

 Home Assistant Docker base version 2024.12.1

### What platform are you using?

ESP8266

### Board

Sonoff S31 POW V1 2 2019-01-23

### Component causing the issue

CSE7766

### YAML Config

```yaml
# Basic Config

substitutions:
  devicename: s31-6849
  friendly_devicename: "${devicename} Unit 1"
  update_interval: 60s
  
# Basic Config
esphome:
  name: ${devicename}

esp8266:
  board: esp12e
  early_pin_init: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password  
  use_address: "${devicename}.grabberville.net"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename} Hotspot"
    password: !secret wifi_password 

logger:
  baud_rate: 0
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
  
  # Remove this line if you're not using Home Assistsant or your switch will restart every now and again
api:
# password: "asdopw7jm2340gws"
[... 184 lines cut ...]
[17:33:30][C][api:141]:   Address: s31-b581.grabberville.net:6053
[17:33:30][C][api:143]:   Using noise encryption: YES
[17:33:30][C][wifi_info:012]: WifiInfo SSID 'Sonoff S31 Unit 3: SSID'
[17:33:30][C][wifi_info:012]:   Icon: 'mdi:wifi-settings'
[17:33:30][C][wifi_info:013]: WifiInfo BSSID 'Sonoff S31 Unit 3: BSSID'
[17:33:30][C][wifi_info:013]:   Icon: 'mdi:wifi-settings'
[17:33:30][C][wifi_info:011]: WifiInfo Scan Results 'Sonoff S31 Unit 3: Wifi Scan'
[17:33:30][C][wifi_info:011]:   Icon: 'mdi:wifi-refresh'
[17:33:30][C][wifi_info:010]: WifiInfo IPAddress 'Sonoff S31 Unit 3: IP'
[17:33:30][C][wifi_info:010]:   Icon: 'mdi:ip-outline'
[17:33:30][C][wifi_signal.sensor:010]: WiFi Signal 'Sonoff S31 Unit 3: WiFi Signal'
[17:33:30][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[17:33:30][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[17:33:30][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[17:33:30][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[17:33:59][D][sensor:093]: 'Sonoff S31 Unit 3: WiFi Signal': Sending state -34.00000 dBm with 0 decimals of accuracy
```

### Additional information

I have tried both no load and with a load. Normal functioning S31s will display line voltage even if there is nothing plugged in ( 0 current )

[image]

Thank you for your dedication to making ESPHome an incredible product.

ssieb, 2025-03-02:
That sensor requires EVEN parity.  Are you sure you didn't get the S31 lite by accident since they don't work with Tasmota either?

grabdavid, 2025-03-02:
> That sensor requires EVEN parity. Are you sure you didn't get the S31 lite by accident since they don't work with Tasmota either?

Yes I checked that. The circuit board has the word POW on it and the lite model is missing the power chip (next to the button). They are working now. I think the problem was that board type. It needs to be esp01_1m. Thanks for your input !!
 
I want to determine if this device will handle solar panels. I have my 16 panels in 4 groups of 4 which operate around 80 volts but they can drop to almost 0 with clouds.  I wonder if this power chip will work at very low voltages.