esphome/issues #6831: type RGB24 is removed; replace with 'type: RGB'

jojkopeter, 2025-03-02:
### The problem

After updating ESPhome to version 2025.2.1, I get the following error message when updating the LED clock:
Image type RGB24 is removed; replace with 'type: RGB'.

`INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/led-clock.yaml...
Failed config

animation: [source /config/esphome/led-clock.yaml:130]
  - file: animations/Kotik.gif
    id: kotik_animation
    
    Image type RGB24 is removed; replace with 'type: RGB'.
    type: RGB24
 `

How can I solve this error?
If I switch to type: RGB I get this error message
issued:
mapping values ​​are not allowed here
  in "/config/esphome/led-clock.yaml", line 132, column 12

### Which version of ESPHome has the issue?

2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1_mini

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-02:
Show what you tried changing it to.  That's a formatting error.

ssieb, 2025-03-03:
Neither one of those is correct.  You need to fix your formatting.

closed 2025-03-03 by ssieb

LarsBorris, 2025-03-04:
> Neither one of those is correct. You need to fix your formatting.

Maybe give a hint to the correct format?

ssieb, 2025-03-04:
Line things up like you must have everywhere else in your config.
Clearly the original config was working, so I don't know how it got messed up by changing a few characters.

LarsBorris, 2025-03-04:
This code has not been touched, only thing that happened, was upgrading ESPhome to higher version.
Now it will not compile.

Can you tell me what the correct syntax is in the new version?

[image]

ssieb, 2025-03-04:
It's not that change.  Check the logs.

LarsBorris, 2025-03-04:
Can you please elaborate on your response?
What do you mean by, its not that change?

The type was RGB24 before and will not compile on newest version of ESPhome.
If I change RGB24 to RGB, the code compiles but does not display the image on my device.

(I am not a coder, ESPhome changes breaking my devices are quite hard for me to troubbleshoot)

ssieb, 2025-03-04:
I'm saying that this change has nothing to do with it not working.  Did you read the changelog?  If you need more help, come ask on the esphome discord server.

jojkopeter, 2025-03-04:
> Keines davon ist richtig. Sie müssen Ihre Formatierung korrigieren.

Can you please tell me where the formatting is not correct here:

`substitutions:
  name: led-clock

esphome:
  name: ${name}
  min_version: 2024.6.0
  on_boot:
    then:
      - display.page.show: splashscreen
      - light.turn_on: 
          id: clock_rgb
          brightness: 50%

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.1.222
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "&{name} Fallback"
    password: !secret ota_password
    channel: 4

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

time:
 # - platform: homeassistant
 #   id: homeassistant_time
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Berlin"
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org
    on_time_sync:
      then:
        - globals.set:
            id: time_synced
            value: 'true'
        - if:
            condition:
              lambda: 'return id(time_synced);'
            then:
              - lambda: |-
                  auto time = id(sntp_time).now();
[... 360 lines cut ...]
                break;              
              default:
                it.image(digit_location_X[i], digit_location_Y[i], id(number_00), id(color_black), color);
                break;
            }
          }         
        }

        if (id(digit_animationFrame) > 8) {
          id(previousTimeValue)[0] = timeValue[0];
          id(previousTimeValue)[1] = timeValue[1];
          id(previousTimeValue)[2] = timeValue[2];
          id(previousTimeValue)[3] = timeValue[3];
          id(digit_animationFrame) = 0;
          id(digit_animationStart)[0] = false;
          id(digit_animationStart)[1] = false;
          id(digit_animationStart)[2] = false;
          id(digit_animationStart)[3] = false;
        }
        else id(digit_animationFrame)++;
        
        if (enableColon) {
          it.draw_pixel_at(colon_location_X, 1, color);
          it.draw_pixel_at(colon_location_X, 3, color);
        }`

ssieb, 2025-03-04:
It looks ok, what's the error?  But please come to the esphome discord server for this.

jojkopeter, 2025-03-06:
It's clear to me too, but with type: RGB24 there was no error for type image look here:
animation:
- file: "animations/Kotik.gif"
id: kotik_animation
type: RGB24 # change here to type: RGB

image:
- file: "animations/Numbers_anim/00.png"
id: number_00
- file: "animations/Numbers_anim/01.png"
id: number_01
- file: "animations/Numbers_anim/10.png"
id: number_10
- file: "animations/Numbers_anim/11.png"
id: number_11
- file: "animations/Numbers_anim/12.png"
id: number_12
- file: "animations/Numbers_anim/21.png"
id: number_21
  - file: "animations/Numbers_anim/22.png"

randybb, 2025-03-06:
It looks like it is not clear for you - `type (Required):` so in this case it is not passing the validation. For all images you have no `type:`, for that animiation you had, but a wrong one which is not supported anymore..

LarsBorris, 2025-03-06:
> It looks like it is not clear for you - `type (Required):` so in this case it is not passing the validation. For all images you have no `type:`, for that animiation you had, but a wrong one which is not supported anymore..

This is exactly the problem. The syntax RGB24 worked in previous release. It does not in the current. Using RGB without 24 does not work. The image doesn’t display. What is the correct syntax? I can’t make it any simpler. Thanks.

ssieb, 2025-03-06:
The problem is that you aren't using the "rgb".  You have no `type:` set.
You need to read the [changelog](https://esphome.io/changelog/2025.2.0.html#image-components).  If you follow the link there, you can find there's a shortcut to make it easier as well.

LarsBorris, 2025-03-08:
I managed to fix my issue.
It was indeed not related to the RGB syntax as @ssieb suggested.
The real issue causing the white screen for me was that I needed the color_palette: 8BIT syntax:

display:
  - platform: ili9xxx
    invert_colors: false
    model: TFT 2.4
    cs_pin: 13
    dc_pin: 14
    reset_pin: 33
    rotation: 180
    update_interval: 10s
    color_palette: 8BIT

Now my power meter is working again.
Thank you for trying to lead me in the right direction, eventhough I was resisting.

bazcurtis, 2025-03-26:
I have the same issue with the Image type RGB24 is removed; replace with 'type: RGB'.  type: RGB24 error. I assume I have to edit this part of the configuration.

```
display:
  - platform: ili9xxx
    id: s3_box_lcd
    model: S3BOX
    invert_colors: false
    data_rate: 40MHz
    cs_pin: 5
    dc_pin: 4
    reset_pin:
      number: 48
      inverted: true
    update_interval: never
```

I am not sure what to add. From the above, I assume color_palette, but not sure which option to pick. Any help would be appreciated.

ssieb, 2025-03-26:
@bazcurtis that error is unrelated to the display config.  Read the changelog and if you still need help, come ask on discord.

bazcurtis, 2025-03-26:
@ssieb I did check the change log, but found not reference to RGB. If you could give me a hint at which part of the change log I am looking at I will give it a try, if that fails I will pop over to Discord

ssieb, 2025-03-26:
I see that one wasn't listed in the changelog.  It's just a message that tells you exactly what to do.
https://esphome.io/components/image#configuration-variables

bazcurtis, 2025-03-27:
Thanks for the help. I hit two more errors after removing the 24.

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.

and then after fixing that, I got this. 

src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope

In the end I found another post that pointed me here - [New Firmware](https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml)