esphome/issues #6832: ESPHome Device Builder addon gone

ps915, 2025-03-02:
### The problem

The Add-on ESPHome Device Builder was removed from Add-on Store in Home Assistant!

I got an repair message, that the Ad-on is no longer availble. 

I am running HA OS on Proxmox on a ASUS NUC 13 Pro (Intel Core i3-1315U)

### Which version of ESPHome has the issue?

2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.5

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]

[image]

[image]****

matze191174, 2025-03-02:
I have the same problem.

uowiec, 2025-03-03:
I have the exactly same problem, also on Proxmox

[image]

maxus-git, 2025-03-03:
Same Problem here:

I have two instances - In no. 1 the addon is updateable from 2025.2.1 to 2025.2.2 and in instance no. 2 the addon (2025.2.0) is no longer avaiable in the addon-store.

## Info no. 1 (Addon available and updatable):
[image]

## Info no. 2 (Addon gone):
[image]

Looks identical to me.

randybb, 2025-03-03:
should be fixed in HA 2025.2.5 https://github.com/home-assistant/supervisor/issues/5699

maxus-git, 2025-03-03:
Also here - fixed with Supervisor 2025.03.0

ps915, 2025-03-03:
fixed indeed! Thanks!

closed 2025-03-03 by ps915