esphome/issues #6834: New ESPhome devices will not show/update via HA/Settings notifications - only older/existing?

NOTORIOUSVR, 2025-03-03:
### The problem

Not exactly sure where/why/how this is happening. I did a quick search on here and google and didn’t immediately turn up anything - more just about not being able to update ESPhome devices at all however that is not my issue.

I can update all of my ESPhome devices through the webconsole, however HA normally notifies of ‘updates’ for all devices including ESPhome devices after updating core ESPhome to a new version.

What I noticed is that all my old devices show up in HA for requiring an update after the core update which is how it’s always been, however the newer devices I’ve added to ESPhome do not show up there. I have to go to the webpanel of ESPhome to update the devices for some reason?

I see functionally no difference in code between and old and new device,.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.5

### What platform are you using?

ESP8266

### YAML Config

```yaml
name: lr-bt-proxy
  friendly_name: lr_bt_proxy

esp32:
  board: mhetesp32devkit
  framework:
    type: esp-idf

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_pass

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "blahblahblah123"

ota:
  - platform: esphome
    password: "blahblah1234"

web_server:
 port: 80
  
## BLUETOOTH HA PROXY
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

### Anything in the logs that might be useful for us?

```txt
I wish I could identify when this started (in what version) but others are starting to report the same thing.

https://community.home-assistant.io/t/new-esphome-devices-will-not-show-update-via-ha-settings-notifications-only-older-existing/
```

ornago, 2025-03-04:
Same here, since I updated and reflashed the config for all of my esp. There is one device asking for an update in settings which I can't update due to a special component included. Comparing that config with the other devices, I don't see any difference why it broke update through settings.

randybb, 2025-03-04:
check in HA under your device if the update entity is enabled. I think it is now disabled by default and you need to explicitly enable it as many people did not like to be notified.

NOTORIOUSVR, 2025-03-04:
> check in HA under your device if the update entity is enabled. I think it is now disabled by default and you need to explicitly enable it as many people did not like to be notified.

I've looked everything I believe is logical - can you specify where I should be looking in HA for this?

randybb, 2025-03-04:
Settings > Devices & services > ESPHome (click on the  devices) and then on your device. Under Configuration you will see a disabled entity, which is the firmware one

NOTORIOUSVR, 2025-03-04:
> Settings > Devices & services > ESPHome (click on the devices) and then on your device. Under Configuration you will see a disabled entity, which is the firmware one

The only thing that is there is:

**Allow the device to perform Home Assistant actions.**

Which was already enabled and from what I understand has nothing to do with the upgrade mechanism.

Either way that is not the fix.

randybb, 2025-03-04:
not configure, but you need to see your entities
[image]

then select your device and then
[image]

NOTORIOUSVR, 2025-03-04:
Here are some examples of my entitites on that screen which DO NOT show in the HA update screen:

https://imgur.com/G5FtHqf

https://imgur.com/tyWL7sv

https://imgur.com/9w7rUew

and an example of one that DOES show in the HA update screen as it is an old ESPHome device:

https://imgur.com/M0jubJF

I don't see  much differece except for the BT Proxies which seem to not show any data or have any working switches and show disconnected for some reason - that is also new since changes required by ESPhome.

randybb, 2025-03-04:
But in the last case you have that Firmware entity enabled, in the previous three it is grayed out - disabled.

NOTORIOUSVR, 2025-03-04:
> But in the last case you have that Firmware entity enabled, in the previous three it is grayed out - disabled.

Incorrect - on the 2nd one it too is enabled

NOTORIOUSVR, 2025-03-04:
I do agree that there is something missing between ESPhome and HA and for some reason when I add new units it doesn't always show Firmware status.  Why that is I do not know.

NOTORIOUSVR, 2025-03-05:
> Can see this discord thread: https://discord.com/channels/429907082951524364/1342164927703552154
> 
> I am guessing there is an issue with the second change here: [home-assistant/core#120907](https://github.com/home-assistant/core/pull/120907) [home-assistant/core#129751](https://github.com/home-assistant/core/pull/129751)
> 
> I also do not have any firmware update entities

That looks exactly like the issue I am experiencing....  Does anyone know if this is a plug in issue or a core HA issue yet?

swoboda1337, 2025-03-05:
> That looks exactly like the issue I am experiencing.... Does anyone know if this is a plug in issue or a core HA issue yet?

Not that I know of, should open one.

fusionstream, 2025-04-06:
Enabling the "Firmware" entity fixed it for me. But I have to go into it 1 by 1.