esphome/issues #6836: The png image decoder is not working.

plplaaa2, 2025-03-04:
### The problem

The png image decoder is not working.

### Which version of ESPHome has the issue?

ESPHome 2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.5

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3 8R16N

### Component causing the issue

online image

### YAML Config

```yaml
http_request:

text_Sensor:
  - platform: homeassistant
    entity_id: weather.naver_weather_1
    name: naver weather 
    id: naver_weather_1
    on_value:
      - online_image.release: weather
      - if:
          condition:            
            and:              
              - lambda: 'return id(naver_weather_1).state == "cloudy";'
              - sun.is_below_horizon:
          then:
            - online_image.set_url:
                id: weather
                url: "http://192.168.x.x:8123/local/weather/cloudy.png"
            - component.update: weather
            - lvgl.image.update:
                id: home_weather_icon
                src: weather

online_image:
  - url: "http://192.168.x.x:8123/local/weather/sunny.png"
    id: weather
    format: PNG
    type: RGB565
    resize: 150x150
    transparency: alpha_channel
    on_download_finished:
      - component.update: my_display

lvgl:
  pages:
    - id: weather_page
      widgets:
      - obj:
          x: 0
          y: 30
          align: TOP_LEFT
          styles: time_style
          width: 480
          height: 450
          widgets:
            - obj:
                x: 0
                y: 0
                width: 480
                height: 250
                styles: card
                widgets:
                  - obj:
                      x: 0
                      y: 0
                      width: 200
                      height: 150
                      align: TOP_LEFT
                      styles: weather_style
                      widgets:
                        - image:
                            x: -10
                            y: 0
                            align: TOP_MID
                            bg_opa: TRANSP
                            id: home_weather_icon
                            src: weather
                            antialias: true
```

### Anything in the logs that might be useful for us?

```txt
full yaml

https://github.com/plplaaa2/esphome-craft-480-kc
```

### Additional information

Log

```
[23:46:56][I][online_image:102]: Updating image http://192.168.1.102:8123/local/weather/cloudy.png
[23:46:56][D][online_image:154]: Starting download
[23:46:56][D][online_image:171]: Allocating PNG decoder
[23:46:56][E][online_image.png:046]: PNG decoder engine not initialized!
[23:46:56][I][online_image:102]: Updating image http://192.168.1.102:8123/local/weather/cloudy.png
[23:46:56][D][online_image:154]: Starting download
[23:46:56][D][online_image:171]: Allocating PNG decoder
[23:46:56][E][online_image.png:046]: PNG decoder engine not initialized!

```

ssieb, 2025-03-04:
For some reason, one device has enough free RAM and the other doesn't.  There's a PR to use PSRAM for the decoder, so that should help when it's included.
https://github.com/esphome/esphome/pull/8354

closed 2025-03-04 by ssieb

plplaaa2, 2025-03-05:
> For some reason, one device has enough free RAM and the other doesn't. There's a PR to use PSRAM for the decoder, so that should help when it's included. [esphome/esphome#8354](https://github.com/esphome/esphome/pull/8354)

Thank you for your response. I only have 8MB of NAND among the LVGL devices I have, so even if I put the weather image icon in, the capacity is close to 90%, so the Online image component is my savior. I hope this PR will be applied in the next update.

ssieb, 2025-03-05:
You could try using JPG or BMP for now if possible.  I think those decoders use less RAM.