esphome/issues #6838: Nextion - touch_sleep_timeout no longer works

jpg6363, 2025-03-06:
### The problem

since version 2025, touch_sleep_timeout: 60 no longer works on Nextion platform

### Which version of ESPHome has the issue?

2025

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.0

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

nextion

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

jpg6363, 2025-03-07:
esphome:
  name: nextion32
  friendly_name: Nextion32

esp32:
  board: esp32dev
  framework:
    type: arduino

.....

#------------------------------------------------------------    
display:
#------------------------------------------------------------  
  - platform: nextion
    id: nextion32
    **touch_sleep_timeout: 60**
    wake_up_page: 0
    brightness: 70%
    update_interval: 3s

argonaute199, 2025-03-13:
I have the same issue since 2025.1 with my Nextion : touch_sleep_timeout doesn't work anymore, so the touchpad stays on.
My full project with code is published in french in this blog : https://www.hacf.fr/esphome-ecran-tactile-nextion/
Let me know if you need hep for tests.

jpg6363, 2025-03-13:
Salut,

Tu as fait un super travail donc je me suis inspiré.

Je pense que la V 2025 a apporté un bug sur les Nextion car j'ai vu d'autres personnes qui parlaient de ce problème. Pour l'instant je suis revenu à la 2024. Je ne sais pas bien comment avancer car les développeurs n'ont pas répondu au ticket

Cordialement

argonaute199, 2025-03-13:
Merci ! J'ai essayé différentes choses comme spécifier directement la variable "thsp" dans l'éditeur Nextion, mais rien n'y fait. Le trafic ESPHome doit empêcher la mise en veille (?)

Jemago, 2025-03-23:
I have the same problem. I tried different things but nothing worked. Any ideas?
Right now, I have a command to turn off the screen at night and turn it on in the morning, but it's not the same-

edwardtfn, 2025-05-05:
Similar issue:
- #5320

edwardtfn, 2025-05-05:
Are you sure this worked before? I don't see any change related to this on the ESPHome side.

argonaute199, 2025-05-05:
For information, I proposed workarounds here (sorry in french)
https://forum.hacf.fr/t/article-ecran-tactile-nextion-avec-esphome-deporter-vos-commandes/10622/111

jpg6363, 2025-05-05:
Merci beaucoup pour ta réponse argonaute199

jpg6363, 2025-05-05:
j'ai mis en place la 1ere méthode, tout fonctionne parfaitement. Encore merci pour ton travail

Jemago, 2025-05-05:
Muchas gracias, merci beaucoup argonaute199, me ha funcionado la solución que has dicho, la primera.

jpg6363, 2025-05-06:
oui autant pour moi le problème est apparu depuis la version 2025 - je suis passé de ESPHome Device Builder 2024.12.4 à la 2025.4.1