esphome/issues #6839: fatal error: Killed signal terminated program cc1plus (stuck with firmware upgrade)

guidokieboom, 2025-03-06:
### The problem

While trying to upgrade device with ESPHOME 2024.12 to 2025.2.2 I get fatal errors and firmware update fails. I've tried both wireless (via wifi) and connected to USB on RPi4-2Gb. The free memory of the RPi4 is approx. 1Gb.
Could anyone help me in directing to a solution or other way to obtain support?

[image]

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.03

### What platform are you using?

ESP32

### Board

ESP32WROOM

### YAML Config

```yaml
esphome:
  name: esp32-04
  comment: BT proxy 

esp32:
  board: esp32dev
  framework:
    type: esp-idf

substitutions:
  hostname: esp32-04

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.2.126   
    gateway: 192.168.2.254
    subnet: 255.255.255.0
  ap:
    ssid: "ESP32_04 Fallback Hotspot"
    password: "xxxxxxxxx"

captive_portal:

web_server:
  port: 80

logger: 
  level: DEBUG

api:
  encryption: 
    key: "xxxxxxxxxxxxxxxxxxxxx"
ota:
#  safe_mode: True
  platform: esphome
  password: !secret esphome_password
  
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: false

bluetooth_proxy:

binary_sensor:
  - platform: status
    name: $hostname Status
    
switch:
  - platform: restart
    name: $hostname restart

sensor:
  - platform: wifi_signal
    name: $hostname WiFi
    update_interval: 600s
```

### Anything in the logs that might be useful for us?

```txt
firmware updates have been working well for past years in earlier versions.
```

ssieb, 2025-03-06:
If it's been working before, you've been lucky.  You don't have enough RAM for compiling.  You will need to do the compiling on another device.
You could try setting the addon to a single compile process, but that will still likely crash and will be very slow.

closed 2025-03-06 by ssieb

guidokieboom, 2025-03-07:
Thanks @ssieb for responding. It seems that I have free memory while compiling (see below graph). Another esp32 compile did work well; this one that worked had a longer yaml configuration (sorry, but I could not get the indent shown properly - in my yaml file it is correct). Maybe bit different. The first one compiling well, the second one having fatal errors while compiling.

`esp32:  
    board: nodemcu-32s
  `

`esp32:
  board: esp32dev
  framework:
    type: esp-idf`

[image]

Would you have other directions?

ssieb, 2025-03-07:
My suggestion is to install esphome on another computer for compiling.  It's not going to work well on that pi.

guidokieboom, 2025-03-07:
thanks for the advise. I then will have to purchase a new RPi. I failed to setup HA and ESPHOME on a window machine. I just commented out the framework and type esp-idf lines and it compiled well again. I also now have BT proxy working well as well (and that is sole purpose of this esp32 board)

ssieb, 2025-03-07:
HA can stay on the pi.  You just need to install esphome somewhere else.