esphome/issues #6840: id(sntp_time).now() returns in UTC after waking up from deep sleep w/o network

SodaWithoutSparkles, 2025-03-07:
### The problem

After the time have been acquired, I put the ESP32-C3 to sleep. Upon waking up, I check the current hour and determine what action shall be taken.

From the logs, I can see that the `id(currTime).now().hour` was returning UTC time instead of local time (UTC+8), until the network is connected and time synchronized.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

ESP32C3 Super mini (Using esp32-c3-devkitm-1)

### Component causing the issue

time/sntp

### YAML Config

```yaml
:
  name: music-player
  friendly_name: music player
  platformio_options:
    board_build.flash_mode: dio
    board_build.mcu: esp32c3

  on_boot: 
    priority: 500
    then:
      # - if:
      #     condition:
      #       - lambda: 'return id(needTurnOffDF);'
      #     then: 
      #       - dfplayer.sleep
      #       - delay: 1s
      #       - output.turn_off: dfplayerEnable
      #       - globals.set: 
      #           id: needTurnOffDF
      #           value: '0'
      - if:
          condition:
            - binary_sensor.is_on: keepAwake
          then:
            - deep_sleep.prevent: deep_sleep_1
            - logger.log:
                format: "Last time fix is %u."
                args: [ 'id(lastTimeFix)']
            - wifi.enable: 
            - globals.set: 
                id: lastTimeFix
                value: '0'
          else: 
            - wifi.disable: 
            - lambda: 'id(lastTimeFix) += 1;'
            - if: 
                condition: 
[... 330 lines cut ...]
  sleep_duration: 24h
  wakeup_pin: 
      number: 1
      allow_other_uses: True
```

### Anything in the logs that might be useful for us?

```txt
[17:57:57][D][wifi:472]: Disabling WIFI...
[17:57:57][D][main:045]: Last time fix is 1.
[17:57:57][D][button:010]: 'Play target song' Pressed.
[17:57:57][D][main:1018]: Hour: 9, DoW: 6
...
[17:58:02][I][wifi:617]: WiFi Connected!
...
[17:58:02][D][sntp:066]: Synchronized time: 2025-03-07 17:58:01
...
[17:58:03][D][custom:358]: Time is now: 17:58:03 (Fri)
[17:58:03][D][main:361]: Sleeping for 116 s
```

### Additional information

The esp is supposed to instruct a DFPlayer Mini to play a song every hour between 11:00 and 18:00 local time on Mon-Fri. To conserve battery, deep_sleep is used whenever it it not playing anything.

SodaWithoutSparkles, 2025-03-07:
I think the issue was the TZ info can only be loaded after wifi is connected?

dwmw2, 2025-04-07:
> I think the issue was the TZ info can only be loaded after wifi is connected?

No, I think it can (and should) be loaded much earlier than that.

But yes, you're right that the problem is that TZ info *is* only loaded after wifi is connected. 

It's done when sntp is set up, which is at priority `BEFORE_CONNECTION`. At that point, `RealTimeClock::apply_timezone()` calls `setenv()` to set the `$TZ` environment variable.

Until then, `$TZ` isn't set, and all calls into the C library to get the "local" time will return GMT.

The correct fix is probably to set `$TZ` *much* earlier in startup, not from the sntp setup at all.

Partly because `setenv()` isn't defined to be thread-safe. If you call it from a thread while there are other threads already running, those other threads may crash if they call `getenv()` while your `setenv()` is running. And ESPHome does start a bunch of threads (some of them behind the scenes like the esp-idf MQTT one).

dwmw2, 2025-04-07:
Reducing the priority of your on_boot method to something lower than BEFORE_CONNECTION (220) would probably work around this. As would explicitly calling `setenv(TZ,...)` for yourself in that method.

closed 2025-04-09 by jesserockz