esphome/issues #6842: ESP8266 - Captive Portal Blank Page

hiscorebob, 2025-03-07:
### The problem

Captive Portal in AP Mode is blank. A forced refresh only shows "data:text/html" on an Android Mobile Device.
Also the page http://192.168.4.1 does not load in a browser when connected to the AP.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini v4.0

### Component causing the issue

captive_portal

### YAML Config

```yaml
wifi:
  ssid: "WIFINAME"
  password: "PASSWORD"
  fast_connect: True
  reboot_timeout: 0s
  ap:
    ssid: "APNAME"
    password: password
    #ap_timeout: 20s
captive_portal:

web_server:
  log: False
  ota: False
  version: 3
```

### Anything in the logs that might be useful for us?

```txt
No error in logs. AP Web Server Startup is logged, as well as connected Client
```

### Additional information

Tried also with web_server version2, same thing.

Dudhr, 2025-06-29:
You've probably moved on after 3 months, but:

`web_server:`
 `  local: true`

hiscorebob, 2025-07-19:
using local: true in web_server component did not change anything, problem was still present.

After reading the release notes of current 2025.7.x release (major improvements on performance and memory management), I decided to try again.

And I can confirm that with latest release, the captive Portal now works again for my project on an ESP8266!

So a big Thank you! to the ESPHome Devs for such an important release (at least for me)