esphome/issues #6843: Display doesn't work anymore after update to 2025.2 (ili9xxx)

andreas-bulling, 2025-03-07:
### The problem

I have two displays based on the ili9xxx platform (model: ILI9342). Everything was working fine with the 2025.1 but after updating to 2025.2 the display, upon a reboot, only shows noisy/scrumbled pixels instead of the text I want to show. I didn't change the yaml so the error must be caused by the new version. The touchscreen still works as before.

```
display:
  - platform: ili9xxx
    model: ILI9342
    spi_id: lcd
    cs_pin:
      number: GPIO15
      ignore_strapping_warning: true
    dc_pin:
      number: GPIO02
      ignore_strapping_warning: true
    rotation: 180
    color_order: RGB
    invert_colors: false
    lambda: |-
       SOME_DISPLAY_CODE

# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "${device_upper} Backlight"
    id: back_light
    restore_mode: RESTORE_DEFAULT_OFF

  - platform: rgb
    name: "${device_upper} LED"
    red: output_red
    id: led
    green: output_green
    blue: output_blue
    restore_mode: ALWAYS_OFF
```

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

andreas-bulling, 2025-03-07:
no. Now I did. Thanks for the hint

andreas-bulling, 2025-03-08:
Reopening this because adding `psram:` doesn't help.

ssieb, 2025-03-08:
If it's an original ESP32, it probably doesn't have PSRAM.  Read the rest of the section in the changelog and check the device logs.

closed 2025-03-08 by ssieb

andreas-bulling, 2025-03-08:
The device was working with the older esphome version for months. So the ESP is definitely able to drive the display. It's an integrated display with ESP32

andreas-bulling, 2025-03-08:
color_palette: 8BIT did the trick