esphome/issues #6845: ESPHome device builder keeps getting stuck in Initializing

RobbertvanOeveren, 2025-03-08:
### The problem

Hi,

When i try to create a new device, the Initializing pop-up comes up. I have waited for more than 20 mins, and it never seems to get out of that loop. 

might this be related to running HA on a RPI4? on the system page, it seems to shoot towards approx 1 gb of ram usage, and the keeps there. the screen says it has 4 gb, but i have never seen it use it all...

if you need any additional info, please shout, id really like to program my board :-)

to prevent issues with the board, i have tried several; boards, cables etc.
ESP8266 (Wemos D1)
ESP32-S2 (Wemos S2)
ESP32-WROOM Devkit
ESP32-S3

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP8266

### Board

ESP32-S2; ESP8266; ESP32-S3

### YAML Config

```yaml
N.A.
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-08:
Compiling on a pi is going to take a long time.  Skip the initial install, then try compiling after so you can see the compile log.

RobbertvanOeveren, 2025-03-09:
i have now installed a backup on a mini-pc, with 8 gb of ram in a proxmox VM. 
the initializing fase still last for more than i care to keep the prgm button pressed,

ssieb, 2025-03-09:
Why are you holding the button?  Did you try what I said?

RobbertvanOeveren, 2025-03-09:
> Why are you holding the button? Did you try what I said?

i installed the initial install via the web.esphome.io, then it discovers the new ESP. still no option to take it over (button greyed out)
might it be that the secrets is not correctly formatted?

or did i misunderstand your "skip the inital install" ? whenever i press the add device button, i do not get any options, just a stalled loading screen.

ssieb, 2025-03-09:
Could you please come ask on the esphome discord server?

RobbertvanOeveren, 2025-03-09:
> Could you please come ask on the esphome discord server?

i have just joined, will create a thread there,

ssieb, 2025-03-09:
It was an issue with bad data in the secrets file.

closed 2025-03-09 by ssieb