esphome/issues #6848: SGP30 sensor chip should be read every second, independent of the update_interval setting

FvanEgmond, 2025-03-08:
### The problem

According to the hardware specs, the SGP30 sensor only provides correct values if this is polled every second. 
in ESPHome, this functions well when the update_interval setting is set to 1 second. 
But this drowns Home Assistant (recorder) with values, so I do like the option to set the interval to higher settings, e.g. 60 seconds. 
But then the code should function to still poll the chip every second, but only _report_ the value every 60 seconds. 
Can the code be altered to do so?   
Thanks in advance!

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP32

### Board

esp32-wroom-32u

### Component causing the issue

sgp30

### YAML Config

```yaml
- platform: sgp30
    eco2:
      name: "Study CO2"
      accuracy_decimals: 1
    tvoc:
      name: "Study TVOC"
      accuracy_decimals: 1
    store_baseline: yes
    address: 0x58
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt

```

azide0x37, 2025-03-09:
You can use sensor filters. 
https://esphome.io/components/sensor/#throttle-average

`# Basic
- platform: sgp30
    eco2:
      name: "Study CO2"
      accuracy_decimals: 1
    tvoc:
      name: "Study TVOC"
      accuracy_decimals: 1
    store_baseline: yes
    address: 0x58
    update_interval: 1s # still poll sensor at 1Hz
    filters:
      - throttle_average: 60s # only send one update per minute
        
# Simple Moving Average
- platform: sgp30
    eco2:
      name: "Study CO2"
      accuracy_decimals: 1
    tvoc:
      name: "Study TVOC"
      accuracy_decimals: 1
    store_baseline: yes
    address: 0x58
    update_interval: 1s # still poll sensor at 1Hz
    filters:
      - sliding_window_moving_average: # average last 60 measurements; update once per minute
        window_size: 60
        send_every: 60`

FvanEgmond, 2025-03-09:
Hi @azide0x37, Thanks for thinking with me! 
This works, although with a slight change: you need to specify the filters per output: 

  - platform: sgp30
    eco2:
      name: "Gameroom CO2"
      accuracy_decimals: 1
      filters:
        - skip_initial: 3
        - throttle: 60s
    tvoc:
      name: "Gameroom TVOC"
      accuracy_decimals: 1
      filters:
        - skip_initial: 3
        - throttle: 60s
    store_baseline: yes
    address: 0x58
    update_interval: 1s

closed 2025-03-09 by FvanEgmond