esphome/issues #6849: No OTA update for Ready-Made Project Bluetooth proxy After update Esphome Builder

Bandit3112011, 2025-03-09:
### The problem

No update available

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP32

### Board

M5Stack Atom Echo 

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]

ssieb, 2025-03-10:
I don't know what you're showing there.  And you haven't really said what the problem is.

cybergrimes, 2025-03-11:
> I don't know what you're showing there. And you haven't really said what the problem is.

Not issue poster but HA is stating to update Bluetooth proxies to version 2025.2.2 or later. The web installer for Bluetooth proxies is currently installing 2025.2.1