esphome/issues #6850: Esphome Builder should notify if browser doesn't support web serial

thibaultmol, 2025-03-09:
### The problem

I noticed that if you try to flash an esp device while in the 'esphome builder' section in home assistant , it won't notify you that your browser doesn't support web serial. the actual esphome web interface does, but the 'esphome builder' section in the home assistant ui won't give you this error/warning

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

The style of error that you get in esphome:

[image]

while in esphome builder I just get this screen (forever... because nothing will happen)

[image]

[image]

thibaultmol, 2025-07-08:
https://nostalebots.xyz/

thibaultmol, 2025-07-08:
But why can't it show the same error as esphome which does properly inform me that it's not supported

ssieb, 2025-07-08:
You're misunderstanding.  You're trying to use the serial port on whatever system the builder is running on, not your browser.

ssieb, 2025-07-08:
It literally says "server port".