esphome/issues #6852: Communication with Tuya MCU not possible - ESP32/ESP8266 not working

Wolly300, 2025-03-09:
### The problem

Hello togehter,

I'm trying to get rid of the Tuya cloud and replace it with ESPHome. My first device/ test is the MiBoxer WiFI 5 in 1 LED Strip Controller.
Model: WL5
Chip: CB3S
CB3S: V2.1.6
MCU: V1.0.1
MCU communication: 115200 baudrate

I tried to flash the chip with Tuya-Cloudcutter, but unfortunately, the device is not opening an access point. So I removed the original chip and tried to replace it with an ESP. 

Before I removed the chip, I already tried to read the communication between the SB3S and the MCU on the board (used a micrograpper probe and connected it to an UART to USB converter). 

The [TuyaMCUAnalyzer](https://github.com/openshwprojects/TuyaMCUAnalyzer) is unfortunately not working. I can read transmitted packages (CB3S Tx -> MCU Rx), but there are always 5 packages in the Rx ringbuffer. 

So tried the SerialTool software with HEX outbut. Now I can see something, but the content makes no sense:

[image]

[image]

Interestingly, on the other side (CB3S Rx <- MCU Tx) is no communication. (Also tested 9600 here)
But the original chip works fine and can control everything via app and HomeAssistant via LocalTuya.

I'm using ESPHome through the Home Assistant extension.
I first tried an ESP8266 (because I have like 20 of them and need to use them before they end like some stuff from my grandpa) with the knowledge that it uses a software UART for communication. 

Hooked it up, and it was not working.

[image]

That's my ESPHome config:

```
# serial connection to Tuya MCU
uart:
  rx_pin: 13
  tx_pin: 15
  baud_rate: 115200

# Register the Tuya MCU connection
tuya:

globals:
  # workaround for the issue where the sensor is not updated on initialization
  # it's safer to assume it's open rather than closed in such case
  # https://github.com/esphome/issues/issues/6503
  - id: initial_state
    type: bool
    restore_value: no
    initial_value: 'true'

binary_sensor:
  - platform: tuya
    id: mcu_contact
    sensor_datapoint: 101
    on_state:
      if:
        condition:
          lambda: 'return id(initial_state);'
        then:
          lambda: 'id(initial_state) = false;'

  - platform: template
    id: contact
    lambda: 'return id(initial_state) || id(mcu_contact).state;'

cover:
  - platform: template
    device_class: garage
    id: main_cover
    name: ""
    lambda: "return id(contact).state ? COVER_OPEN : COVER_CLOSED;"

light:
  - platform: "tuya" # dont forget to set the correct mode whit the button on the controller
    name: "WB5 LED Controller"
    color_interlock: true # infinite loop isue when false (false is default)
    gamma_correct: 1 # has built in gamma correct wil cause issuses when changed
    default_transition_length: 0s # has built in transitions
    switch_datapoint: 20

    # for brightness
    #dimmer_datapoint: 22
    #min_value: 0
    #max_value: 1000 # Max value is 1000 (default is 255)

    # for Correlated Color Temperature(CCT) or white channels
    #color_temperature_datapoint: 23
    #color_temperature_invert: true #set to true if connected as stated on the controller
    #color_temperature_max_value: 1000 # Max value is 1000 (default is 255)
    #cold_white_color_temperature: 6000 K # Change depending on LED strip
    #warm_white_color_temperature: 2700 K # Change depending on LED strip

    # for RGB channels
    color_datapoint: 24
    color_type: hsv
```

I tried it again with an ESP32 C3 MINI to have native UART connection, but same result:

````
[21:55:19][C][logger:177]: Logger:
[21:55:19][C][logger:178]:   Max Level: DEBUG
[21:55:19][C][logger:179]:   Initial Level: DEBUG
[21:55:19][C][logger:181]:   Log Baud Rate: 0
[21:55:19][C][logger:182]:   Hardware UART: USB_CDC
[21:55:19][C][uart.arduino_esp32:151]: UART Bus 0:
[21:55:19][C][uart.arduino_esp32:152]:   TX Pin: GPIO2
[21:55:19][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[21:55:19][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[21:55:19][C][uart.arduino_esp32:157]:   Baud Rate: 115200 baud
[21:55:19][C][uart.arduino_esp32:158]:   Data Bits: 8
[21:55:19][C][uart.arduino_esp32:159]:   Parity: NONE
[21:55:19][C][uart.arduino_esp32:160]:   Stop bits: 1
[21:55:19][C][template.binary_sensor:028]: Template Binary Sensor 'contact'
[21:55:19][C][template.cover:073]: Template Cover 'ESP32 C3 MINI'
[21:55:19][C][template.cover:073]:   Device Class: 'garage'
[21:55:19][C][light:092]: Light 'WB5 LED Controller'
[21:55:19][C][light:094]:   Default Transition Length: 0.0s
[21:55:19][C][light:095]:   Gamma Correct: 1.00
[21:55:19][C][tuya.light:101]: Tuya Dimmer:
[21:55:19][C][tuya.light:106]:    Switch has datapoint ID 20
[21:55:19][C][tuya.light:109]:    Color has datapoint ID 24
[21:55:19][C][tuya.binary_sensor:017]: Tuya Binary Sensor:
[21:55:19][C][tuya.binary_sensor:018]:   Binary Sensor has datapoint ID 101
[21:55:19][C][captive_portal:089]: Captive Portal:
[21:55:19][C][mdns:116]: mDNS:
[21:55:19][C][mdns:117]:   Hostname: esp32-c3-mini
[21:55:19][C][esphome.ota:073]: Over-The-Air updates:
[21:55:19][C][esphome.ota:074]:   Address: esp32-c3-mini.local:3232
[21:55:19][C][esphome.ota:075]:   Version: 2
[21:55:19][C][esphome.ota:078]:   Password configured
[21:55:19][C][safe_mode:018]: Safe Mode:
[21:55:19][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:55:19][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:55:19][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:55:19][C][api:140]: API Server:
[21:55:19][C][api:141]:   Address: esp32-c3-mini.local:6053
[21:55:19][C][api:143]:   Using noise encryption: YES
[21:55:19][C][tuya:041]: Tuya:
[21:55:19][C][tuya:044]:   Initialization failed. Current init_state: 0
[21:55:19][C][tuya:049]:   If no further output is received, confirm that this is a supported Tuya device.
[21:55:19][E][tuya:464]: Initialization failed at init_state 0
[21:55:34][E][tuya:464]: Initialization failed at init_state 0
[21:55:49][E][tuya:464]: Initialization failed at init_state 0
[21:56:04][E][tuya:464]: Initialization failed at init_state 0
[21:56:19][D][light:036]: 'WB5 LED Controller' Setting:
[21:56:19][D][light:047]:   State: OFF
[21:56:19][D][tuya:623]: Setting datapoint 20 to 0
[21:56:19][W][tuya:626]: Setting unknown datapoint 20
[21:56:19][E][tuya:464]: Initialization failed at init_state 0
[21:56:21][E][tuya:464]: Initialization failed at init_state 0
[21:56:30][D][light:036]: 'WB5 LED Controller' Setting:
[21:56:30][D][light:047]:   State: ON
[21:56:30][D][tuya:668]: Setting datapoint 24 to 0000000003E8
[21:56:30][W][tuya:671]: Setting unknown datapoint 24
[21:56:30][D][tuya:623]: Setting datapoint 20 to 1
[21:56:30][W][tuya:626]: Setting unknown datapoint 20
[21:56:32][E][tuya:464]: Initialization failed at init_state 0
[21:56:33][E][tuya:464]: Initialization failed at init_state 0
[21:56:35][E][tuya:464]: Initialization failed at init_state 0
```

For every "[E][tuya:464]: Initialization failed at init_state 0" in the log, I can see a new transmitted package in SerialTool. The value is always FF

Does anyone know why the communication with the MCU is not working?
I don't know what else I could try. 
And why do I get this "Setting unknown datapoint xx"?

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP32

### Board

ESP32-C3-MINI1, ESP8266

### Component causing the issue

MiBoxer WL5

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-09:
You didn't need to replace the chip.  esphome will run on the CB3S module.

Wolly300, 2025-03-09:
> You didn't need to replace the chip. esphome will run on the CB3S module.

Right now I'm not really able to flash one.

ssieb, 2025-03-09:
Please come ask for help on the esphome discord server.

ssieb, 2025-03-10:
Resolved.  It was wrong pins.

closed 2025-03-10 by ssieb