esphome/issues #6857: Sensor SGP30 doesn't report correct values even after baseline

boggiano, 2025-03-11:
### The problem

ESPHOME  Version: 2025.2.1

I am using an ESP32 S3 mini with a SGP30 sensor via I2C.
The sensor is recognized and everything seems to be working.
I waited for 12 hours for calibration and from the logs, I actually see:

`"09:19:25	[I]	[sgp30:128]Current eCO2 baseline: 0x0018, TVOC baseline: 0x000E"`

But even smoking on it I get always the same values (or very close to these):
```
09:21:25	[D]	[sgp30:292]	 Got eCO2=401.0ppm TVOC=0.0ppb
09:21:25	[D]	[sensor:094]	 'CO2 SGP30_1': Sending state 401.00000 ppm with 1 decimals of accuracy
09:21:25	[D]	[sensor:094]	 'TVOC SGP30_1': Sending state 0.00000 ppb with 1 decimals of accuracy
```

I tried with Arduino IDE (with an esp8266) with a simple code using Rob Tillaart's library and Iìve got  consistent results, so the sensor is working.

The SGP30 is the only device in this configuration (I've removed the SHT30, just to be sure).

### Which version of ESPHome has the issue?

2025.2.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-S3 Mini

### Component causing the issue

SGP30

### YAML Config

```yaml
substitutions:
  device_internal_name: esp32_s3_barom_1
  device_sampling_time: 30s

esphome:
  name: ${device_internal_name}

esp32:
  board: esp32-s3-devkitc-1

#${device_internal_name}

#################################
#          WI-FI                #
#################################

# MAC Local MAC: 24:EC:4A:2F:2C:24

# ✅ Configura WiFi e OTA
wifi:
  ssid: "Mynet"
  password: "mypass"

  manual_ip:
    static_ip: 192.168.188.107  # IP statico 
    gateway: 192.168.188.1
    subnet: 255.255.255.0

#################################
#          OTA                  #
#################################

ota:
  platform: esphome

#################################
#         WEB SERVER            #
#################################

web_server:
  port: 80

#################################
#            LOGS               #
#################################

logger:
  level: DEBUG

#################################
#             HA API            #
#################################

api:
  reboot_timeout: 0s #Evitiamo il reboot se non connesso a HA

#################################
#         BUTTON RESTART        #
#################################

button:
  platform: restart
  name: "Riavvia ${device_internal_name}"
  id: device_restart

#################################
#                I2C            #
#################################

# ✅ Configura I²C su ESP32-S3 Mini (SDA: GPIO8, SCL: GPIO9)
i2c:
  sda: 8
  scl: 9
  scan: true  # Cerca automaticamente i dispositivi I²C

#################################
#         SENSORI TESTO         #
#################################
# Text Sensors (per informazioni)

text_sensor:
  - platform: wifi_info
    ip_address:
      name: IP Address
      id: ${device_internal_name}_ip_address
    ssid:
      name: Connected SSID
      id: ${device_internal_name}_connected_ssid
    mac_address:
      name: Mac Wifi Address
      id: ${device_internal_name}_mac_address

  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

  #-------------------------------------------------------------------------------
  # Custom Text sensors
  #-------------------------------------------------------------------------------
  - platform: template
    name: Uptime Human Readable
    id: ${device_internal_name}_uptime_human
    icon: mdi:clock-start

################################################################################
# Switch
################################################################################
switch:
  - platform: restart
    name: "Restart"
    id: ${device_internal_name}_device_restart

#################################
#         SENSORI               #
#################################

# ✅ Sensore Uptime

#  - platform: uptime
#    type: seconds
#    name: Uptime ${device_internal_name}

  - platform: uptime
    name: "Uptime Sensor"
    id: ${device_internal_name}_uptime_sensor
    update_interval: ${device_sampling_time}
    internal: true
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: ${device_internal_name}_uptime_human
            state: !lambda |-
              int seconds = round(id(${device_internal_name}_uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

# ✅ Sensore GY-SGP30 (Qualità dell'aria: CO2 e TVOC)

  - platform: sgp30
    eco2:
      name: "CO2 SGP30_1"
      id: ${device_internal_name}_eco2 
      accuracy_decimals: 1
    tvoc:
      name: "TVOC SGP30_1"
      id: ${device_internal_name}_tvoc
      accuracy_decimals: 1

    baseline:
      eco2_baseline: 0x001C
      tvoc_baseline: 0x0010

#    compensation:
#      temperature_source: sht3xd_temperature
#      humidity_source: sht3xd_humidity

    address: 0x58
    store_baseline: yes
    update_interval: ${device_sampling_time}

  # ✅ Sensore sht3xd (Temperatura, Umidità)

#  - platform: sht3xd

#    temperature:
#      name: "Temperatura sht3xd_1"
#      id: sht3xd_temperature
#      filters:
#        - offset: -1.0  # ✅ Corregge l'errore del sensore

#    humidity:
#      name: "Umidità sht3xd_1"
#      id: sht3xd_humidity

#    address: 0x44
#    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt

```

simenopsahl, 2025-04-09:
I and someone else had this problem also
https://github.com/esphome/issues/issues/6848  
It seems that the sensor is only accurate when measuring every second, but you can use a filter to throttle the reported value.
```YAML
sensor:
  - platform: sgp30
    i2c_id: $tvoc_i2c_bus
    id: sgp_sensor
    eco2:
      id: eco2_sgp30
      name: "eCO2 SGP30"
      device_class: "carbon_dioxide"
      state_class: "measurement"
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30
    tvoc:
      id: tvoc_sgp30
      name: "TVOC SGP30"
      device_class: "volatile_organic_compounds_parts"
      state_class: "measurement"
      unit_of_measurement: "ppm"
      filters:
        # PPB to PPM
        - multiply: 0.001
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30
    address: 0x58
    update_interval: 1s
    compensation:
      temperature_source: temperature_local
      humidity_source: humidity_local
```
Or
```YAML
  - platform: sgp30
    eco2:
      name: "Study CO2"
      accuracy_decimals: 1
      filters:
        - throttle: 60s
    tvoc:
      name: "Study TVOC"
      accuracy_decimals: 1
      filters:
        - throttle: 60s
    store_baseline: yes
    address: 0x58
    update_interval: 1s
```

boggiano, 2025-04-14:
That fixed the issue!
Thanks!

closed 2025-04-14 by boggiano