esphome/issues #6858: wk2132 i2c to uart converter

edouardreg, 2025-03-11:
### The problem

Hello,

I use a DFR0627 (DFROBOT gravity module) wich is based on a WK2132 i2c to uart converter.
The purpose is to add two more uart ports to my esp32 wich is monitoring several water tanks levels.

I already use two JSN-SR04T sensors to two esp32 hardwares uart with succes

But with the wk2132_i2c: component it seems impossible to use the JSN-SR04T sensor on the uarts created by the converter.
As soon as I indicate as "uart_id: " for the JSN-04T one of the wk2132's uarts id there is a "required key missin" error in the yaml editor.

I think it's perhaps because the JSN-SR04T needs to define rx and tx pins and don't know how manage the virtual ports ?

How can we resolve this ?

Thanks by advance

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.1

### What platform are you using?

ESP32

### Board

esp32 devkitc v4

### Component causing the issue

platform: "jsn_sr04t"

### YAML Config

```yaml
esphome:
  name: test
  friendly_name: Test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "J9bC3TCR3KZNKFB3SLVvVcv1bgdmExI0Jced7eVKCpk="

ota:
  - platform: esphome
    password: "4707758c3950dc1c3f0f046956c725b8"

 
wifi:
  ap: {}

captive_portal:

i2c:
  sda: GPIO26
  scl: GPIO27
  scan: true
  id: bus_i2c_b
  frequency: 800kHz

wk2132_i2c:
  - address: 0x70
    id: wk2132_bridge_i2c
    uart:
      - id: i2c_uart_0
        channel: 0
        baud_rate: 9600

      - id: i2c_uart_1
        channel: 1
        baud_rate: 9600
     
sensor:
# JSN-SR04M2 
  - platform: "jsn_sr04t"
    name: "Distance_SR04M_cuves"
    uart_id: i2c_uart_0
    update_interval: 3s
    state_class: measurement
    unit_of_measurement: "m"
```

### Anything in the logs that might be useful for us?

```txt
message "required key not provided." at line 46 (uart:) into the esphome's yaml editor
```

edouardreg, 2025-03-18:
After that, if using esp-idf  

`esp32:
  board: esp32dev
  framework:
    type: esp-dif`

It's not working. With arduino framework, it's OK : you get the JSN-SR04T value, but also an error 
`[E][weikai:406]: Receive frame error FSR=00101000` with each distance value received...

timdaman, 2025-05-27:
I have another data point here. I am seeing exactly the same behavior.

My configuration is very large (I have most of the houses sensors on one micro controller) so I will post extracts and add the full YAML content to the end.

Here is my wk2132 configuration
```
  - address: 0x70
    i2c_id: i2c_uart_expander_bus
    id: wk2132_bridge_i2c_2
    uart:
      - id: rainbarrel_level_uart
        channel: 0
        baud_rate: 9600
```

When I added that it passed validation. However when I try to use it with a02yyuw

```
sensor:
  - platform: a02yyuw
    uart_id: rainbarrel_level_uart
    name: rainbarrel airgap
    unit_of_measurement: "cm"
    accuracy_decimals: 00
    filters:
      - multiply: 0.1 # Convert MM to CM
      - throttle: 60s
```

The uart key on `wk2132_bridge_i2c_2` raises a problem `required key not provided.`

[image]

When I remove the `uart_id` from "rainbarrel airgap" the error clears on `wk2132_bridge_i2c_2`

[image]

**Full YAML below**
```
substitutions:
  devicename: basement-1
  name: basement-1

esphome:
  name: ${devicename}
  on_boot:
    priority: 800
    then:
      - if:
          condition:
            binary_sensor.is_on: humidifier_open_sensor
          then:
            - button.press: humidifer_close_drain_button

packages:
  wesp32:  !include packages/wesp32.yaml
  # bluetooth:  !include packages/esp32_bluetooth_proxy.yaml
  mmwave: !include packages/ld2410.yaml

uart:
  - tx_pin: GPIO33
    rx_pin: GPIO34
    baud_rate: 256000
    id: diningroom_mmwave_uart

dfrobot_sen0395:
  - id: bathroom_mmwave
    uart_id: bathroom_mmwave_uart

i2c:
[... 437 lines cut ...]
    ld2410_id: diningroom_mmwave
    timeout:
      name: Diningroom timeout
    max_move_distance_gate:
      name: Diningroom max move distance gate
    max_still_distance_gate:
      name: Diningroom max still distance gate

text_sensor:
  - platform: template
    name: "Kitchen AQI Description"
    icon: "mdi:air-filter"
    id: kitchen_aqi_text
  - platform: template
    name: "NOC AQI Description"
    icon: "mdi:air-filter"
    id: noc_aqi_text

interval:
  - interval: 48h
    then:
      - button.press: humidifer_open_drain_button
      - delay: 100s
      - button.press: humidifer_close_drain_button
```

timdaman, 2025-05-27:
Not a CPP person though sometimes I play one. ;)

Looking at the weikai component I see the following
```
this->data_bits_ = 8;  // always equal to 8 for WeiKai (cant be changed)
```
https://github.com/esphome/esphome/blob/73771d5c50467a771d76cca1efe9e489662b09f7/esphome/components/weikai/weikai.cpp#L285C1-L285C75

I wonder if this explains how we got here. Since stop bits can't be set the validator disallows it.
Could adding 
```
CONF_DATA_BITS = "data_bits"
```
https://github.com/esphome/esphome/blob/73771d5c50467a771d76cca1efe9e489662b09f7/esphome/components/weikai/__init__.py#L19C1-L19C15

and 
```
                cv.Optional(CONF_DATA_BITS, default=1): cv.one_of(8, int=True),
```
https://github.com/esphome/esphome/blob/73771d5c50467a771d76cca1efe9e489662b09f7/esphome/components/weikai/__init__.py#L62

resolve this bug? 

That would allow validation work but since there is one hard coded value we won't need to do any of the other configuration processing that the other configuration attributes need.

I don't have a esphome dev environment to validate.

timdaman, 2025-05-27:
I hot-patched my esphome container with the change I mentioned above and validation now passed. I was able to flash my device and it is now return data as expected.