esphome/issues #6863: Font not rendering correctly in 2025.3.0 beta

skgsergio, 2025-03-13:
### The problem

I have two TTF fonts that stopped rendering in with the beta. Probably related to https://github.com/esphome/esphome/pull/8300

I've built the project using `esphome` command line in debug and I noticed the glyphs are getting wrong width and height, both set at 1.

Research context: https://github.com/lubeda/EspHoMaTriXv2/issues/283

### Which version of ESPHome has the issue?

2025.3.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.2

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

font

### YAML Config

```yaml
font:
  - id: font_matrixchunky6
    file: resources/fonts/MatrixChunky6.ttf
    size: 6
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_matrixchunky8
    file: resources/fonts/MatrixChunky8.ttf
    size: 8
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
```

### Anything in the logs that might be useful for us?

```txt
- 2025.3.0b2

DEBUG Running to_code in esphome.components.font (num 16)
DEBUG Adding: // font:
DEBUG Adding: //   id: font_matrixchunky6
//   file:
//     path: /config/esphome/resources/fonts/MatrixChunky6.ttf
//     type: local
//   size: 6
//   glyphs:
//   - '|¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ'
//   glyphsets: []
//   ignore_missing_glyphs: false
//   bpp: 1
//   extras: []
//   raw_data_id: uint8_t_id
//   raw_glyph_id: font_glyphdata_id
DEBUG Adding: static const uint8_t uint8_t_id[] PROGMEM = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
DEBUG Registered variable uint8_t_id of type uint8_t
DEBUG Adding: static const font::GlyphData font_glyphdata_id[] = {font::GlyphData{
  .a_char = (const uint8_t *)" ",
  .data = uint8_t_id + 0,
  .advance = 5,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}, font::GlyphData{
  .a_char = (const uint8_t *)"!",
  .data = uint8_t_id + 1,
  .advance = 3,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}, [...], font::GlyphData{
  .a_char = (const uint8_t *)"\342\202\254",
  .data = uint8_t_id + 101,
  .advance = 9,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}};
DEBUG Registered variable font_glyphdata_id of type font::GlyphData
DEBUG Adding global: font::Font *font_matrixchunky6;
DEBUG Adding: font_matrixchunky6 = new font::Font(font_glyphdata_id, 102, 6, 6, 1);
DEBUG Registered variable font_matrixchunky6 of type font::Font
DEBUG  -> finished

- 2025.2.2

DEBUG Running to_code in esphome.components.font (num 16)
DEBUG Adding: // font:
DEBUG Adding: //   id: font_matrixchunky6
//   file:
//     path: /config/resources/fonts/MatrixChunky6.ttf
//     type: local
//   size: 6
//   glyphs:
//   - '|¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ'
//   glyphsets: []
//   ignore_missing_glyphs: false
//   bpp: 1
//   extras: []
//   raw_data_id: uint8_t_id
//   raw_glyph_id: font_glyphdata_id
DEBUG Adding: static const uint8_t uint8_t_id[] PROGMEM = {0x00, 0x00, 0xA8, 0x80, 0xAA, 0x00, 0x00, 0x6C, 0xE6, 0xC0, 0xA2, 0x48, 0xA0, 0x52, 0x44, 0x00, 0x89, 0x28, 0x00, 0xA4, 0xA0, 0x00, 0x4E, 0x40, 0x00, 0xA0, 0xE0, 0x00, 0x80, 0x22, 0x48, 0x80, 0xEA, 0xAA, 0xE0, 0x4C, 0x44, 0xE0, 0xE2, 0xE8, 0xE0, 0xE2, 0xE2, 0xE0, 0xAA, 0xE2, 0x20, 0xE8, 0xE2, 0xE0, 0xE8, 0xEA, 0xE0, 0xE2, 0x22, 0x20, 0xEA, 0xEA, 0xE0, 0xEA, 0xE2, 0xE0, 0x88, 0x00, 0x24, 0x84, 0x20, 0xE0, 0xE0, 0x00, 0x84, 0x24, 0x80, 0xE2, 0x40, 0x40, 0x76, 0xAD, 0x0E, 0x00, 0xEA, 0xEA, 0xA0, 0xEA, 0xCA, 0xE0, 0xEA, 0x8A, 0xE0, 0xCA, 0xAA, 0xE0, 0xE8, 0xE8, 0xE0, 0xE8, 0xE8, 0x80, 0xE4, 0x2D, 0x2F, 0x00, 0x94, 0xBD, 0x29, 0x00, 0xE4, 0x44, 0xE0, 0x22, 0x2A, 0xE0, 0xAA, 0xCA, 0xA0, 0x88, 0x88, 0xE0, 0x8B, 0x6A, 0xA2, 0x88, 0x00, 0x96, 0xAD, 0x29, 0x00, 0xF4, 0xA5, 0x2F, 0x00, 0xEA, 0xE8, 0x80, 0xF4, 0xA5, 0x4D, 0x00, 0xEA, 0xCA, 0xA0, 0x68, 0xE2, 0xC0, 0xE4, 0x44, 0x40, 0xAA, 0xAA, 0xE0, 0xAA, 0xAE, 0x40, 0xAA, 0xAA, 0xBE, 0x50, 0x00, 0xAA, 0x4A, 0xA0, 0xAA, 0xE4, 0x40, 0xE2, 0x48, 0xE0, 0x88, 0x42, 0x20, 0xE0, 0xE6, 0xAE, 0x00, 0x88, 0xEA, 0xE0, 0xE8, 0x8E, 0x00, 0x22, 0xEA, 0xE0, 0xEA, 0xCE, 0x00, 0x64, 0xE4, 0x40, 0xEA, 0xE2, 0x60, 0x88, 0xEA, 0xA0, 0x40, 0xC4, 0xE0, 0x20, 0x22, 0xAE, 0x8A, 0xAC, 0xA0, 0xC4, 0x44, 0xE0, 0xFA, 0xAA, 0xAA, 0x00, 0xEA, 0xAA, 0x00, 0xEA, 0xAE, 0x00, 0xEA, 0xAE, 0x80, 0xEA, 0xAE, 0x20, 0xE8, 0x88, 0x00, 0xE8, 0x6E, 0x00, 0x4E, 0x44, 0x60, 0xAA, 0xAE, 0x00, 0xAA, 0xE4, 0x00, 0xAA, 0xAF, 0x94, 0x00, 0xA6, 0xCA, 0x00, 0xAA, 0xE2, 0x60, 0xE6, 0xCE, 0x00, 0xAA, 0x80, 0x8A, 0x80, 0xEA, 0xE0, 0x00, 0xC4, 0x60, 0x00, 0xE6, 0xE0, 0x00, 0xAA, 0xE8, 0x40, 0x42, 0xE0, 0x24, 0xEA, 0xEA, 0x2E, 0x8E, 0x8E, 0x24, 0xE4, 0x4E, 0xF0, 0x25, 0xAB, 0x48, 0x24, 0x0E, 0xAE, 0x24, 0xE6, 0xAE, 0x24, 0xEA, 0xCE, 0x24, 0x0C, 0x4E, 0xE0, 0xEA, 0xAA, 0x24, 0x0E, 0xAE, 0x24, 0x0A, 0xAE, 0x64, 0xE4, 0x60};
DEBUG Registered variable uint8_t_id of type uint8_t
DEBUG Adding: static const font::GlyphData font_glyphdata_id[] = {font::GlyphData{
  .a_char = (const uint8_t *)" ",
  .data = uint8_t_id + 0,
  .offset_x = 0,
  .offset_y = 2,
  .width = 4,
  .height = 4,
}, font::GlyphData{
  .a_char = (const uint8_t *)"!",
  .data = uint8_t_id + 2,
  .offset_x = 0,
  .offset_y = 0,
  .width = 2,
  .height = 6,
}, [...], font::GlyphData{
  .a_char = (const uint8_t *)"\342\202\254",
  .data = uint8_t_id + 303,
  .offset_x = 0,
  .offset_y = 0,
  .width = 4,
  .height = 6,
}};
DEBUG Registered variable font_glyphdata_id of type font::GlyphData
DEBUG Adding global: font::Font *font_matrixchunky6;
DEBUG Adding: font_matrixchunky6 = new font::Font(font_glyphdata_id, 102, 6, 6, 1);
DEBUG Registered variable font_matrixchunky6 of type font::Font
DEBUG  -> finished
```

### Additional information

TTF fonts obtained from: https://github.com/trip5/Matrix-Fonts

trip5, 2025-03-13:
Hi there.  Matrix-Fonts dev here.  Just making sure I stay in the loop about this.  Have not tested this bug yet.

My fonts are fixed-height, created with FontForge, which may be involved in this problem, too.

I'm thinking freetype only works with hinted fonts instead of pixel-fixed fonts...?

skgsergio, 2025-03-13:
Tested with bdf variant and got a different error:

```
Failed config

font: [source /config/esphome/pixel-clock.yaml:92]
  
  Codepoints in bitmap fonts must be in the range 0-255.
  - id: font_matrixchunky6
    file: resources/fonts/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
  - id: font_matrixchunky8
    file: resources/fonts/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
```

skgsergio, 2025-03-13:
Interesting...

This works:
```yaml
font:
  - id: font_matrixchunky6
    file: https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/6-series/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_matrixchunky8
    file: https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/8-series/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
```

This fails with `Codepoints in bitmap fonts must be in the range 0-255.` error and is just a `curl -LO` of the URLs above:
```yaml
font:
  - id: font_matrixchunky6
    file: resources/fonts/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_matrixchunky8
    file: resources/fonts/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

```

trip5, 2025-03-13:
This feels like freetype can't properly handle unicode BDF files, too... a bit disappointing if ESPHome is implementing a "fix" for something that wasn't broken...

But the possibility exists that my font needs a change, too.

clydebarrow, 2025-03-13:
The limitation on BDF font codepoints was historic from the previous (Pillow based) version but there was a code comment querying whether it was a BDF or a Pillow limitation. If the BDF fonts work with Unicode when fetched as a URL then there's clearly no BDF limitation and Freetype must be happy so that check should be able to be removed for local files.

I'll look at the initial issue later.

clydebarrow, 2025-03-13:
> I'm looking at perhaps the TTF doesn't include correct info about ascent/descent / height/width,

No idea, I have a reasonable knowledge of using fonts but am clueless about what goes into them. But if it works with Pillow, it should be able to work with Freetype. It may be that different properties need to be used in different situations, but I will test with the info above.  Might be a day or two before I can do that though.

trip5, 2025-03-14:
To be honest, if I can contribute... I'm usually very happy to do so immediately.  But right now, I'm packing up to travel to my home country in a few days.  I'm bringing a clock as a gift so I'll spend some time experimenting but... it could be pretty random when I actually have time.

clydebarrow, 2025-03-14:
Test this please: https://github.com/esphome/esphome/pull/8407

```
external_components:
  - source: github://pr#8407
    components: [font]
```

skgsergio, 2025-03-14:
Regarding https://github.com/esphome/issues/issues/6863#issuecomment-2722334128 (bdf local vs url): as @trip5 said is the € symbol, once removed it doesn't throws error. But if I keep it and use the URL version it renders the € without issues, so is actually working.

[image]

Regarding https://github.com/esphome/issues/issues/6863#issuecomment-2723389882 (testing the pr):
- The TTF font works with it.
- The local BDF font works keeping the € symbol and renders it correctly.

trip5, 2025-03-14:
So that pr squashes it... Fantastic.

closed 2025-03-14 by clydebarrow