esphome/issues #6864: SEN5x reports incorrect values for SEN55 PMx

radeklat, 2025-03-13:
### The problem

I recently bought 3 SEN55 sensors and all of them report completely wrong PMx values. In a room with air circulation behing a PM filter, all 3 report the PMx values in thousands. Not only would that be more of a visible smoke, it should actually be even outside of the range of the sensor (Datasheet states 1000 is the upper range). Nearby Awair Element air quality sensor reports < 10 PMx concentration.

Thank you for all your help.

CC @martgras and @anbreww

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

firebeetle32

### Component causing the issue

sen5x

### YAML Config

```yaml
sensor:
  - platform: sen5x
    id: sen55
    pm_1_0:
      name: "SEN55 PM1"
      id: sen55_pm1
      accuracy_decimals: 0
    pm_2_5:
      name: "SEN55 PM2.5"
      id: sen55_pm2_5
      accuracy_decimals: 0
    pm_4_0:
      name: "SEN55 PM4"
      id: sen55_pm4
      accuracy_decimals: 0
    pm_10_0:
      name: "SEN55 PM10"
      id: sen55_pm10
      accuracy_decimals: 0
    temperature:
      name: "SEN55 Temperature"
      id: sen55_temperature
      accuracy_decimals: 1
    humidity:
      name: "SEN55 Humidity"
      id: sen55_humidity
      accuracy_decimals: 1
    voc:
      name: "SEN55 VOC"
      id: sen55_voc
      accuracy_decimals: 1
    nox:
      name: "SEN55 NOx"
      id: sen55_nox
      accuracy_decimals: 1
    store_baseline: true
    update_interval: 30s
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

A graph of the reported values without any conversion:

[image]

I noticed a few things:
- There a static offset between PM1 and the other PM values.
- The other PM values are always identical.
- The sensor doesn't react to smoke at all (I tried to blow a match in it's direction - no change).
- The VOC sensor does react to VOCs (such as open nail polish remover), so I don't think it's a connection issue.

I find it suspicious that all 3 sensors would be broken the same way. One of them was purchased months apart, so they are likely not even the same batch. What else can I do to diagnose the issue? From my limited knowledge of C code and comparing it to the Datasheet, I can't spot anything wrong.

mweinelt, 2025-07-06:
The PM values are just cumulative, that is why they are so close to another. I agree that, ideally, we would get buckets for the different sizes that do not overlap. For that you need to subtract the value of the next smaller PM size from each PM value.

[image]

https://sensirion.com/media/documents/6791EFA0/62A1F68F/Sensirion_Datasheet_Environmental_Node_SEN5x.pdf

radeklat, 2025-07-21:
Thank you for clarifying that the PM values are the cumulative. I updated the configuration:

```yaml
sensor:
  - platform: sen5x
    id: sen55
    pm_1_0:
      name: "SEN55 PM1"
      id: sen55_pm1
      accuracy_decimals: 0
    pm_2_5:
      name: "SEN55 PM2.5"
      id: sen55_pm2_5
      accuracy_decimals: 0
      filters:
        - lambda: return x - id(sen55_pm1).raw_state;
    pm_4_0:
      name: "SEN55 PM4"
      id: sen55_pm4
      accuracy_decimals: 0
      filters:
        - lambda: return x - id(sen55_pm2_5).raw_state;
    pm_10_0:
      name: "SEN55 PM10"
      id: sen55_pm10
      accuracy_decimals: 0
      filters:
        - lambda: return x - id(sen55_pm4).raw_state;
```

However, the values are still nonsensical:
- PM values don't seem to be reacting to environment. PM 1.0 and PM 2.5 is accumulating into what looks like a logarithm plot. PM 4 and 10 stay at 10.
- The values for PM 2.5 suggest a wildfire, not a air in closed office.
- The value for PM 1 is way outside of the sensor resolution (0-1000)

See the updated graphs with measurement over ~12 days:

Logarithmic scale
[image]

Linear scale
[image]

anthonysecco, 2025-10-01:
I just setup my SEN55 today and noticed the PM sensors are all saturated

[21:47:34.669][V][i2c.idf:153]: 0x69 RX bytes 24
[21:47:34.669][V][i2c.idf:172]: Sending 5 jobs
[21:47:34.669][VV][sen5x:350]: pm_1_0 = 0xfffe
[21:47:34.669][VV][sen5x:353]: pm_2_5 = 0xfffe
[21:47:34.669][VV][sen5x:356]: pm_4_0 = 0xfffe
[21:47:34.669][VV][sen5x:359]: pm_10_0 = 0xfffe
[21:47:34.669][VV][sen5x:362]: humidity = 0x121c
[21:47:34.678][VV][sen5x:365]: temperature = 0x12f8
[21:47:34.679][VV][sen5x:368]: voc = 0x01cc
[21:47:34.679][VV][sen5x:374]: nox = 0x000a

The datasheet says if any reading is unknown, it returns 0xffff, in this case we see 0xfff**e**

Here's the initial provision historical graph.   It continued to increase until saturation.

[image].

I'm continuing to troubleshoot but it's not clear to me what the issue could be.

closed 2026-05-09 by github-actions[bot]

radeklat, 2026-10-01:
I'm reporting back after I fixed the issue with the help of Claude Code / Opus 5.5:

Root cause found: **not a driver bug**, the sensor was powered from 3.3 V.                                                                          

The SEN5x needs **VDD 4.5–5.5 V** ([datasheet](https://sensirion.com/media/documents/6791EFA0/62A1F68F/Sensirion_Datasheet_Environmental_Node_SEN5x.pdf)). I powered it from the Gravity I2C port of the [DFRobot FireBeetle 2 IO shield (DFR0762)](https://www.dfrobot.com/product-2395.html). Per the [schematic](https://dfimg.dfrobot.com/wiki/18552/DFR0762_firebeetle-2-io-expansion-shield_schematics_V1.0.pdf), the `+` pin of all Gravity ports is  **3V3**. On 3.3 V, RH/T and VOC still work fine, but the PM part (fan + laser) gives garbage. This is the same as @malteh71 found.

**Evidence** (raw I2C reads with my own lambda, bypassing the driver; CRC OK, device status register `0xD206` = `0x00000000` in both cases):
                                                                                                                                
| | 3.3 V | 5 V |                                                                                                                                   
|---|---|---|                                                                                                                                       
| PM1 / PM2.5 (µg/m³) | 70–97 / 74–102, jumping | 11 / 12, stable |                                                                                 
| Number conc. 0.5 µm (#/cm³) | 480–660 | 74–83 |                                                                                                   
| Typical particle size (µm) | 0.31 | 0.47–0.51 |                                                                                                   
| Match smoke | no reaction on PM1/PM2.5 | PM1 3917, PM2.5 saturates at `0xFFFE`, then decays |                                                     

Note: the device status register does **not** flag the low supply voltage, so nothing in the logs hints at it.
                                                                                                                                                      
**Fix:** connect SEN55 VDD to 5 V (the shield's `VIN` screw terminal or the FireBeetle `VCC` pin when powered by USB). Keep SDA/SCL on the 3.3 V I2C bus.

The driver decodes the values correctly. Suggestions that could help other users:
- Docs: add a warning that VDD must be 5 V, even though I2C works on 3.3 V.
- `0xFFFE` (seen by @anthonysecco) is the saturated maximum (6553.4), not an error value.