esphome/issues #6866: H801 keep rebooting

FLchs, 2025-03-16:
### The problem

Hello,
I'm using a h801 (esp8266) to drive several led strips for a small greenhouse.
The device keeps being unavailable for a while, the available again. Strangely the light keeps on while doing so, but either HA, esphome logs or the webui keep reporting the reseting reason as `Software/System restart`.

Even the minimal config linked is not working. 
The h801 is powered by a 12v 4A power supply which is well over the 10 watts 12v led strip I'm using for testing and anyway even with every lights off and nothing plugged in, it keeps rebooting.

I tried disabling every components but nothing I do seems to change anything so I'm asking for help.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2025.3.2

### What platform are you using?

ESP8266

### Board

h801

### Component causing the issue

light

### YAML Config

```yaml
esphome:
  name: smart-greenhouse-light

esp8266:
  board: esp01_1m

logger:
# Enable Home Assistant API
api:
  password: ""

# web_server:
#   port: 80
#   include_internal: true
#   version: 2

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: "[redacted]"
  password: "[redacted]"

output:
  # - platform: esp8266_pwm
  #   pin: 12
  #   frequency: 1000 Hz
  #   id: pwm_b
  # - platform: esp8266_pwm
  #   pin: 13
  #   frequency: 1000 Hz
  #   id: pwm_g
  - platform: esp8266_pwm
    pin: 15
    frequency: 500 Hz
    id: pwm_center

light:
  - platform: monochromatic
    name: "Center light"
    output: pwm_center
```

### Anything in the logs that might be useful for us?

```txt
If really needed I will solder some wires to start a serial logging session but I would rather not :)
```

### Additional information

I <3 esphome

popy2k14, 2025-04-19:
I have the same issue with H801 controller. 
Sadly I don't see it often myself, but my son and wife told me that the lamp is flashing shortly white on reboot.

My yaml is basically the same, here are some differences: 

```
...
# Enable logging and disable serial uart (with baudrate: 0)
# workaround that the devices does not reboot so often

logger:
  baud_rate: 0         
...

#RGBW LED stripe + FET

light:
  - platform: rgbw
    name: Licht
    id: leds
    color_interlock: true
    default_transition_length:
        seconds: ${transition_length_seconds}
    red: output_red
    green: output_green
    blue: output_blue
    white: output_white
    <<: !include ../effects.yaml
...
```

@FLchs any news on this issue? 
I have added the "baud_rate: 0" a few months ago which I think was some improvement to this issue

FLchs, 2025-04-19:
Hello @popy2k14,
Yeah I finally got it working reliably, I had to declare all the outputs that are connected to the MOSFETs. 
When I have access to my computer I will share my working config file.

popy2k14, 2025-04-19:
@FLchs nice to hear that.
Looking forward to your config.

popy2k14, 2025-05-09:
> Hey [@popy2k14](https://github.com/popy2k14) sorry for the late reply, here is the important part :
> 
> output:
>   - platform: esp8266_pwm
>     pin: 12
>     frequency: 1000 Hz
>     id: pwm_b
>   - platform: esp8266_pwm
>     pin: 13
>     frequency: 1000 Hz
>     id: pwm_g
>   - platform: esp8266_pwm
>     pin: 15
>     frequency: 1000 Hz
>     id: pwm_r
>   - platform: esp8266_pwm
>     pin: 14
>     frequency: 1000 Hz
>     id: pwm_w1
>   - platform: esp8266_pwm
>     pin: 4
>     frequency: 1000 Hz
>     id: pwm_w2
> light:
>   - platform: monochromatic
>     id: "stripe_1"
>     name: "Stripe 1"
>     output: pwm_r
>   - platform: monochromatic
>     id: "stripe_2"
>     name: "stripe_2"
>     output: pwm_g
>   - platform: monochromatic
>     id: "stripe_3"
>     name: "stripe_3"
>     output: pwm_b
> I only use 3 output but if any output is not declared the board is stuck in a reboot loop.

thx, i already have all defined and also just using 4 (rgbw).
I dont know what changed but it now doesnt restart.