esphome/issues #6868: Component ID containing upper case altered / Component not available with genuine ID containing upper case letter

bugeaud, 2025-03-17:
### The problem

When creating a component with id field containing uppercase in the YAML, the component is not available as Web API using genuine ID (uppercase included) but instead the same string converted with lower case only.

IDs are defined by esphome.io/guides/configuration-types.html#config-id
My understanding was that ID of YAML has only C language name restriction but as XML ID are case sensitive.

Hence the following IDs are all valid but differents ones :

- test
- TEST
- Test
- TeSt
- ...

The problem is that if we define an ID that contains any uppercase letter (say TEST), this ID can not be used as-is to access the API to work with the component from a remote place using HTTP. Only the lower case version (here test) can be used.

Please note, that web UI (version 2 or 3 at minimum) do seem to use lower case only ID for the HTTP POST request sent down the wire, which means somewhere there was some convert was done.

Either clarify the documentation that only lower case ID are respected and shall be used or remove the forced lower case unnecessary convert perfrom at any step.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

KinCony E16P

### Component causing the issue

web

### YAML Config

```yaml
substitutions:
 roller_TEST_button_up: input16
 roller_TEST_button_down: input15
 roller_TEST_motor_up: output1
 roller_TEST_motor_down: output2 
 roller_TEST_open_duration: 10000ms
 roller_TEST_close_duration: 10000ms
 
# ESP32 Board config
esphome:
  name: kc868-e16p-test

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable IPv6
network:
  enable_ipv6: true

# Add a I2C entry
i2c:
  sda: 16
  scl: 15
  scan: true
  id: bus_a

# Either active ethernet or wifi, but not both due to IDF limitation :(
# LAN8720 configuration entry
#ethernet:
#  type: LAN8720
#  mdc_pin: GPIO23
#  mdio_pin: GPIO18
#  clk_mode: GPIO17_OUT
#  phy_addr: 0

wifi:
  ssid: !secret wifi-ssid
  password: !secret wifi-password

web_server:
  port: 80
  version: 3
  local: true
  auth:
    username: !secret web-user
    password: !secret web-password

ota:
  - platform: esphome
    password: !secret ota-password

# PCF8574 I2C IO Expanders configuration
pcf8574:
  - id: 'pcf8574_hub_out_1'  # for output channel 1-8
    address: 0x21

  - id: 'pcf8574_hub_out_2'  # for output channel 9-16
    address: 0x25

  - id: 'pcf8574_hub_in_1'  # for input channel 1-8
    address: 0x22

  - id: 'pcf8574_hub_in_2'  # for input channel 9-16
    address: 0x24

# Individual outputs
switch:
  - platform: gpio
    id: output1
    name: "Output 1"
    interlock: &interlock_TEST [output1, output2]
    internal: true
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 0
      mode: OUTPUT
      inverted: true

  - platform: gpio
    id: output2
    name: "Output 2"
    interlock: *interlock_TEST
    internal: true
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 1
      mode: OUTPUT
      inverted: true

binary_sensor:

  - platform: gpio
    id: input15
    name: "Input 15"
    on_press:
      - lambda: |-
          if (id(TEST).current_operation == COVER_OPERATION_CLOSING){
            id(TEST).stop();
          }
          else {
            id(TEST).close();
          }
    pin:
      pcf8574: pcf8574_hub_in_2
      number: 6
      mode: INPUT
      inverted: true

  - platform: gpio
    id: input16
    name: "Input 16"
    on_press:
      - lambda: |-
          if (id(TEST).current_operation == COVER_OPERATION_OPENING){
            id(TEST).stop();
          }
          else {
            id(TEST).open();
          }
    pin:
      pcf8574: pcf8574_hub_in_2
      number: 7
      mode: INPUT
      inverted: true

cover:
  - platform: time_based
    name: "TEST VR"
    id: TEST
    device_class: shutter
    
    open_action:
      - switch.turn_on: ${roller_TEST_motor_up}
    open_duration: ${roller_TEST_open_duration}

    close_action:
      - switch.turn_on: ${roller_TEST_motor_down}
    close_duration: ${roller_TEST_close_duration}

    stop_action:
      - switch.turn_off: ${roller_TEST_motor_up}
      - switch.turn_off: ${roller_TEST_motor_down}

# Enable logging
logger:

# Enable Home Assistant API
api:
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-20:
You have misread the docs.  The web server "ID" is generated from the **name:**, not the **id:**, and it is converted to all lowercase.

closed 2025-03-20 by ssieb

bugeaud, 2025-03-20:
Thanks Samuel for clarifying, my bad. I misunderstood the current way of doing here in web_api.

My understanding was that there are indeed an id on yaml elements for components (we can see them in the YAML tests for instance) to be identified that is used in esphome already.

I find odd that altough those id elements are created by the user in the YAML and are already used already in some area of esphome, the web_api code do not use them straight away but override them instead with synthetic id (see calc_object_id()) coming from the name. My expectation was that such a synthetic value from name would only be used if id was not set in the setup_entity.

Doing so has several harsh consequences on non english language names : loss of meaning and collisions.

For instance here is the impact of the sole str_sanitize used at this time on some languages :

Example in French (mostly ASCII name but some non ASCII)

cout << str_sanitize("Lumière salle à manger");
Prints :
Lumi__re_salle____manger

Example in Japanese (non ASCII name only)
cout << str_sanitize("部屋の照明");
Prints :
_______________

French example shows that depending on Unicode configuration you get unpredictable underscore numbers.
Japanese example goes one step further and shows that the "id" generated contains only underscores and will colide with an other "id" from another component with the same number of non ASCII glyphs.

I do thing that the way of using name as "id" on web_api has to evolve to something more classical such as : use id as an id if it exists, but if not keep the current way of doing (for legacy reason). Doing so, people could have I18Ned name in any language but still rely on id (ASCII only) for configuration and machine to machine interractions as id.

I was thinking to open a feature request on that one.

ssieb, 2025-03-20:
The `id:` is internal only.  It's literally the C++ variable name used for the compiled code, so it goes away at runtime.  But I do understand the issue with language characters and that's something that needs to be figured out.

pzich, 2025-03-20:
It looks like home assistant uses some version of "unicode slugify" that handles non-ascii characters a bit better:
https://github.com/home-assistant/core/blob/dev/tests/util/test_init.py#L40-L62
Maybe something similar can be used in ESPHome

ssieb, 2025-03-20:
Interesting.  That would have to be handled in the python ahead of time because there's no way we can include the unicode libraries and lookup tables on the device.

ssieb, 2025-03-21:
I think you misunderstood what I meant.  I'm suggesting that the "synthetic id" be generated by the python during the compilation process and then stored on the device, instead of being generated at runtime on the device.