esphome/issues #6872: Sprinkler Controller causes panic when running through automatic advance

paradox460, 2025-03-18:
### The problem

Been trying to build a sprinkler controller using a KinCony E16S, which is an ESP32 relay board.

When I set each output as regular switches, I can turn them on and off freely, all on, all off, without issues.

But if I try to run the sprinkler auto advance program, it fails and panics on the third "valve". I've tried enabling many different orders, changing the pin outputs around, it's consistently on the third one.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.3

### What platform are you using?

ESP32-IDF

### Board

KinCony KC868-E16S-V1.4 with ESP32-WROOM-32E

### Component causing the issue

Sprinkler

### YAML Config

```yaml
substitutions:
  name: "irrigation-controller-3243cc"
  friendly_name: Irrigation Controller

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev
  on_boot:
    then:
      - button.press: buzzer_beep

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
  - platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

[... 426 lines cut ...]
[13:42:21]A10     : 0x3ffc0e98  A11     : 0x00019bbc  A12     : 0x00000000  A13     : 0x00060c23  
[13:42:21]A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[13:42:21]EXCVADDR: 0x00000028  LBEG    : 0x40082f5d  LEND    : 0x40082f65  LCOUNT  : 0x00000027  
[13:42:21]
[13:42:21]
[13:42:21]Backtrace: 0x400dd003:0x3ffbf960 0x4015804d:0x3ffbf980 0x401580b7:0x3ffbf9a0 0x400e2457:0x3ffbf9c0 0x400e4a5e:0x3ffbf9f0 0x400d7fa2:0x3ffbfa10
[13:42:21]
[13:42:21]
[13:42:21]
[13:42:21]
[13:42:21]ELF file SHA256: d9ba86196112b4ff
[13:42:21]
[13:42:21]Rebooting...
[13:42:21]ets Jul 29 2019 12:21:46
[13:42:21]
[13:42:21]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:42:21]configsip: 0, SPIWP:0xee
[13:42:21]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:42:21]mode:DIO, clock div:2
[13:42:21]load:0x3fff0030,len:7112
[13:42:21]load:0x40078000,len:15624
[13:42:21]load:0x40080400,len:4
[13:42:21]load:0x40080404,len:3876
[13:42:21]entry 0x4008064c
```

muvo243, 2025-03-29:
Hey everyone, I dug into the issue and it looks like the problem is actually in how the valve operators are set up in the OpenSprinkler code. When running the auto advance routine, the code panics on the third valve because it only allocates two valve operator objects. Essentially, the vector in the code is hardcoded like this:

  std::vector<SprinklerValveOperator> valve_op_{2};

This means that while manually switching the valves works fine, the auto advance cycle tries to access a third valve operator that doesn’t exist, causing the crash.

To fix it, you’ve got a couple of options:

Increase the Number of Valve Operators:
Change the initialization to create as many operators as you need for your setup. Instead of hardcoding two, modify the vector’s size so that it matches the total number of valves. For example, if you have four valves, initialize it with four SprinklerValveOperator objects.

Reuse or Dynamically Allocate Operators:
If your design allows it, you might refactor the code so that valve operators are reused once a valve cycle is complete. This way, you can dynamically manage the pool of operators rather than fixing the number at startup.

In short, the fix is to ensure that the number of valve operators matches the number of valves that the auto advance cycle needs to handle. This adjustment should prevent the panic when the cycle reaches the third valve.

jstoaks2112, 2025-04-13:
Thanks to muvo243 - this is exactly the issue and the correct fix I was encountering. Although I could not wait for the fix to happen and merge. I created a local custom_component in ESPhome, basically duplicating this component locally. Included the fix on the valve operator, recompiled and the issue had resolved !!!

Kudos AGAIN to movu243 !!!! - You Rock for digging into this !!!!

> Hey everyone, I dug into the issue and it looks like the problem is actually in how the valve operators are set up in the OpenSprinkler code. When running the auto advance routine, the code panics on the third valve because it only allocates two valve operator objects. Essentially, the vector in the code is hardcoded like this:
> 
>   std::vector valve_op_{2};
> 
> This means that while manually switching the valves works fine, the auto advance cycle tries to access a third valve operator that doesn’t exist, causing the crash.
> 
> To fix it, you’ve got a couple of options:
> 
> Increase the Number of Valve Operators: Change the initialization to create as many operators as you need for your setup. Instead of hardcoding two, modify the vector’s size so that it matches the total number of valves. For example, if you have four valves, initialize it with four SprinklerValveOperator objects.
> 
> Reuse or Dynamically Allocate Operators: If your design allows it, you might refactor the code so that valve operators are reused once a valve cycle is complete. This way, you can dynamically manage the pool of operators rather than fixing the number at startup.
> 
> In short, the fix is to ensure that the number of valve operators matches the number of valves that the auto advance cycle needs to handle. This adjustment should prevent the panic when the cycle reaches the third valve.

PiervitoCreanza, 2025-05-10:
As I understand it, the system currently allocates only two SprinklerValveOperator instances, which is generally sufficient since typically only two valves are active at any given time, one finishing while the next begins, especially with valve_overlap in effect.

However, problems occur when the combination of valve_overlap and pump delay allows more than two valves to be active simultaneously. For instance, if valve_overlap is greater than or equal to twice the run duration, multiple valves can overlap beyond what the two allocated operators can manage. When the auto-advance routine attempts to activate a third valve during this window, it exceeds the vector’s bounds, resulting in a crash.

To prevent this, the simplest solution is to ensure the parameters are sized conservatively, specifically, valve_overlap and any delays should be significantly smaller than the valve run time. This guarantees that no more than two valves (and pumps) are active simultaneously.

jstoaks2112, 2025-05-12:
As I've physically run through this scenario.

For every 1 controller you could potentially have up to 4 valves/pumps
active at one time.

If you have 1 master valve, 1 pump and two valves overlapping.

So the vector needs to be sized large enough to accommodate the physical
setup.

muvo243, 2025-05-12:
Maybe we could ask the mainainers of esphome to do a merge with my suggested fix @jesserockz ?

jesserockz, 2025-05-26:
@muvo243 If you have a suggested fix, please make a pull request

muvo243, 2025-05-28:
@jesserockz Yeah, I didn’t fully understand the underlying code in ESPHome at the time, and I was hesitant to introduce potential issues like the one @kbx81 mentioned.

I agree it’s more nuanced than just adding more SprinklerValveOperator instances. I can see how that might mask the crash but introduce race conditions or sequencing problems with valve control. Your point about transitions and edge cases like overlap/run time mismatches makes total sense.

jstoaks2112, 2025-06-02:
> Well, I would not say it is and edge case, by introducing valve_delay and
> pump_start/stop_pump/valve_delay this is quite normal config that overlaps
> all valves to prevents pressure spikes. For now using overlap in useless in
> sprinker controller, but you can just build a script that involves valve
> closing/opening overlap and it works like a charm. So sadly the workaround
> to achieve desired behaviour is not to use sprinkler controller, despite
> its idea is awesome.

I've had no issues with valve overlap after changing the valve operator std
vector to anything larger than 4.

You have to download the component from ESPhome GitHub and create a local
component for this.

Regards,
J

jstoaks2112, 2025-12-12:
> I believe esphome/esphome#12436
> <https://github.com/esphome/esphome/pull/12436> fixes this -- anybody
> still here & care to confirm? 🙂

I dont believe it did, as this did not correct the number of valve
operators that caused the panic condition. unless the default number of
valve numbers was increased by default ?

i have not checked recently as I had to bring the repo local and make the
corrections for my specific setup of 3 valves.

regards J