esphome/issues #6874: builds_flags broken with latest esphome

yanivf38, 2025-03-19:
### The problem

prior to 2025.2.2 everything worked as expected were my build flags looked like this -
```
esphome:
  platformio_options:
    build_flags:
      - DARDUINO_USB_MODE=1
      - DARDUINO_USB_CDC_ON_BOOT=1
  ```
but with the latest build, I need to add an extra dash to the build flags and make it look like this
```
esphome:
  platformio_options:
    build_flags:
      # Need the two flags to be able to see the logs  (the logs will show up if I connect to the left usb port)
      - -DARDUINO_USB_MODE=1
      - -DARDUINO_USB_CDC_ON_BOOT=1
```

### Which version of ESPHome has the issue?

esphome 2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.3

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

clydebarrow, 2025-03-21:
> ```
>       - DARDUINO_USB_MODE=1
>       - DARDUINO_USB_CDC_ON_BOOT=1
> ```

I'm surprised that ever worked. There should be no space between the - and the D in a compiler argument.

yanivf38, 2025-03-21:
hi @clydebarrow ,
The dash signifies that each line is an individual item and not part of the flag itself. In the working example I have two dashes for each row. 1st for the item and 2nd for the flag

clydebarrow, 2025-03-21:
> In the working example I have two dashes for each row. 1st for the item and 2nd for the flag

Yes, that's how it should be. It's just coincidence that YAML uses a dash to signify an array element and command line tools like compilers use a dash to distinguish a flag from a file argument. You can also write it like this using JSON style array syntax.

```
esphome:
  platformio_options:
    build_flags: [-DARDUINO_USB_MODE=1, -DARDUINO_USB_CDC_ON_BOOT=1]
```

yanivf38, 2025-03-21:
> build_flags: [-DARDUINO_USB_MODE=1, -DARDUINO_USB_CDC_ON_BOOT=1]

Thank you. I like that option. Easier to read in my opinion

closed 2025-03-21 by clydebarrow

ssieb, 2025-03-21:
Why do you even need those build options?  That should be handled automatically by esphome.  What are you trying to do?

yanivf38, 2025-03-22:
Good question @ssieb . 
I use them because my esp by default acts as OTG so this forces it to USB mode so I can see the logs.  This is what the community recommended. I did try without the flags with the latest esphome and it works as well (in the 2024 builds it didn't work without them) I commented them out.