esphome/issues #6876: switch.on_turn_... actions causing failure to boot

Renegade605, 2025-03-19:
### The problem

When attempting to add switch.on_turn_on/off actions, some cause the chip to fail boot. ESPHome web shows it disconnected and all entities are unavailable in Home Assistant, but a new config can still be flashed OTA.

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.3.3

### What platform are you using?

ESP32-IDF

### Board

Apollo DEV-1

### Component causing the issue

switch

### YAML Config

```yaml
# Define Project
substitutions:
  name: thermostat
  version: "0.1"
  device_description: ${name} made by DGR - version ${version}.

esphome:
  name: ${name}
  friendly_name: Thermostat
  comment: Thermostat on Apollo DEV-1
  platformio_options: 
    board_build.flash_mode: dio
  
  project:
    name: "DGR.Thermostat"
    version: "${version}"
  
  on_boot:
    then:
      - light.turn_on:
          id: rgb_onboard_light
          effect: "Fast Flash"
          red: 1.0
          green: 0
          blue: 0
      

# Define Board
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  - platform: esphome
    password: "91d876399c4ed7f0455df4dff693fad5"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermostat Fallback Hotspot"
    password: !secret wifi_password
  
  on_connect:
    then:
      - light.turn_on:
          id: rgb_onboard_light
          effect: "None"
          brightness: 0.5
          red: 1.0
          green: 0
          blue: 0

captive_portal:

web_server:
  port: 80

# Switches
switch:
  - platform: gpio
    pin: GPIO4
    id: relay_heater
    name: "Furnace Heat"
    on_turn_on: 
      then:
        - switch.turn_on: relay_blower
        - light.turn_on: 
            id: rgb_onboard_light
            effect: "Slow Pulse"
            brightness: 1.0
            red: 1.0
            green: 1.0
            blue: 0.9
#    on_turn_off: 
#      then:
#        - light.turn_on: 
#            id: rgb_onboard_light
#            effect: "Slow Pulse"
#            red: 0
#            green: 1.0
#            blue: 0
    
  - platform: gpio
    pin: GPIO5
    id: relay_blower
    name: "Blower Fan"
#    on_turn_on: 
#      then:
#        - if: 
#            condition:
#              - switch.is_off: relay_heater
#            then: 
#              - light.turn_on: 
#                  id: rgb_onboard_light
#                  effect: "Slow Pulse"
#                  red: 0
#                  green: 1.0
#                  blue: 0
    on_turn_off: 
      then:
        - switch.turn_off: relay_heater
#        - light.turn_on: 
#            id: rgb_onboard_light
#            effect: "None"
#            brightness: 0.5
#            red: 1.0
#            green: 0
#            blue: 0

# onboard LED using pin GPIO3 as shown below
light:
  - platform: esp32_rmt_led_strip
    id: rgb_onboard_light
    name: "RGB Onboard Light"
    pin: GPIO3
    default_transition_length: 0s
    chipset: WS2812
    num_leds: 1
    rgb_order: grb
    effects:
      - pulse:
          name: "Slow Pulse"
          transition_length: 1000ms
          update_interval: 1000ms
          min_brightness: 0.5
          max_brightness: 1.0
      - pulse:
          name: "Fast Flash"
          transition_length: 100ms
          update_interval: 100ms
          min_brightness: 0.2
          max_brightness: 1.0

button:
  - platform: restart
    icon: mdi:power-cycle
    name: "ESP Reboot"

sensor:
  - platform: internal_temperature
    name: "ESP Temperature"
    id: sys_esp_temperature

  - platform: wifi_signal
    name: RSSI
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: homeassistant
    entity_id: sensor.hallway_climate_temperature
    id: hallway_temperature
    name: "Hallway Temperature"
    device_class: temperature
    state_class: measurement
    internal: false
    on_value_range: 
      - below: 20.4
        then: 
          - switch.turn_on: relay_heater
      - above: 20.4
        then:
          - switch.turn_off: relay_blower
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

In the 2 switch entries, there are commented out sections. Uncommenting any one of these sections causes the issue (ie. trying to enable the on_turn_off action of switch.relay_heater).

Also tried with arduino framework instead of esp-idf, same behaviour.

Renegade605, 2025-03-20:
After realizing I could get logs over USB, I have determined the cause.

The switch.on_turn_off is triggered on boot, the light.turn_on action causes a kernel panic because (I assume) it is not yet initialized.

Am I missing something here? Are triggers really intended to execute so early in the boot process that other components aren't ready yet? Does the light.turn_on action really not check if the target is ready? Why doesn't it handle the exception gracefully?

A ton of searching has yielded no obvious condition I can add that checks if these components are ready before calling them.

randybb, 2025-03-20:
Exactly, it is not initialized yet. you need to change priority, so this will be executed a bit later.

```yaml
esphome:
  on_boot:
    priority: -100
    then:
      ..
yaml

Renegade605, 2025-03-20:
on_boot isn't the trigger causing the problem.

on_turn_off of the switch is. There is no priority option for that trigger.

ssieb, 2025-03-20:
You can add `setup_priority: 700` to the switch or probably even move the light before the switch.  They both have the same priority.

ssieb, 2025-03-20:
1. That's intentional.  We don't want people using it unless absolutely necessary because it can easily cause problems.
2. It wasn't already off.  It didn't have a state at boot.  From your config, wouldn't you want it to trigger at boot?  How else is the LED going to get set?  You try using `restore_mode: disable` if you really don't want it to.
3. There are no exceptions.  They take up way too many resources.  It's not that it's not ready, it hasn't been initialized yet.

Renegade605, 2025-03-21:
> 1. That's intentional.  We don't want people using it unless absolutely necessary because it can easily cause problems.
Fair enough. This doesn't seem like an extraordinary circumstance though.
  
> 2. It wasn't already off.  It didn't have a state at boot.  From your config, wouldn't you want it to trigger at boot?  How else is the LED going to get set?  You try using `restore_mode: disable` if you really don't want it to.
That's only true programmatically. _Physically_, it is off if the device is unpowered, and it continues to be off after the device powers up.
In this particular case, the LED is set in on_boot and wifi.on_connect. I know that I'm new to ESPHome, but I'm not new to programming. Best practice is to initialize everything in one place, not rely on the initialization of one component to initialize others. If (like this case) an LED is to have an initial state, that shouldn't be reliant on a switch coming online and the spaghetti that comes with bouncing all around the program.
Thanks for the tip on `restore_mode: disable`. Again, the documentation doesn't hint at this being a side_effect of that. For that matter, the documentation of on_turn_off doesn't say it will be executed at boot either. If there is a note there that says "on_turn_on/off will execute on first power up when the state is restored", this entire issue becomes much more obvious to solve.
  
> 3. There are no exceptions.  They take up way too many resources.  It's not that it's not ready, it hasn't been initialized yet.
Fair enough also. The logs specifically said "the Exception was unhandled" but that must be the kernel talking rather than the application. Regardless, I do stand by the statement that calling a light.turn_on action shouldn't be capable of causing kernel panic in such an ordinary situation.

ssieb, 2025-03-21:
> That's only true programmatically. _Physically_, it is off if the device is unpowered, and it continues to be off after the device powers up.

Unpowered is different than off.  The pin has to be set to a state, so there's always going to be a change.  And if the pin is inverted, then off is with the pin on.  You are welcome to update the documentation if you want.

> Best practice is to initialize everything in one place, not rely on the initialization of one component to initialize others.

I disagree, but I guess that's personal opinion.  If one component is going to be the one controlling another one, why have an extra copy of the state setting code that you'll have to keep in sync if you make changes later.

The priority settings are tricky to get right and don't work in all cases.  However, I would suggest that the switch should have a lower priority than the light, because it's very much more likely that the switch would be affecting the light than the other way around.  It would also be a lot better if we could have a dynamically calculated priority instead of the current fixed priorities, but that's a big change.

> The logs specifically said "the Exception was unhandled" but that must be the kernel talking rather than the application.

That's the RTOS saying that nothing handled the CPU exception.

esphome makes embedded programming so easy that people forget that it's still embedded programming.  ;-)

Renegade605, 2025-03-21:
> Unpowered is different than off. 

I disagree. Unpowered is by definition off.

> I disagree, but I guess that's personal opinion. If one component is going to be the one controlling another one, why have an extra copy of the state setting code that you'll have to keep in sync if [...]

That's a fair point. In this case, the boot state and idle states are different, so setting the idle state at boot is obviously undesirable. I guess it's impossible to make a one-size-fits-all solution to that.

> The priority settings are tricky to get right and don't work in all cases. However, I would suggest that the switch should have a lower priority than the light, because it's very much more likely that the switch would be affecting the light than the other way around.

Doesn't that mean that my situation should never have happened? :)

>It would also be a lot better if we could have a dynamically calculated priority instead of the current fixed priorities, but that's a big change.

Definitely. I've been dreaming of handing my smart home over to the AI and just letting it figure out how I want it to behave since Eureka first aired.

> esphome makes embedded programming so easy that people forget that it's still embedded programming. ;-)

Sometimes when a framework makes programming too easy, it actually makes it harder for programmers to use. ;)

ssieb, 2025-03-21:
> I disagree. Unpowered is by definition off.

You don't see the difference between not connected and GNDed?  That can have a huge difference depending on what it's connected to.

Renegade605, 2025-03-21:
> > I disagree. Unpowered is by definition off.
> 
> You don't see the difference between not connected and GNDed? That can have a huge difference depending on what it's connected to.

I'm sure there's an exception that proves the rule, but in general no. A switch that is off is unpowered, and an unpowered switch is functionally equivalent to off. If 'unpowered' really is a different state than 'off' for a particular device, I'd argue that it should be considered something entirely different than a switch.

clydebarrow, 2025-03-21:
There is a general issue within ESPHome in that actions can be triggered during the `setup()` phase that affect other components that have not yet been setup. The setup priority can mitigate that, but it's a band-aid. I've previously proposed that any triggers invoked before _all_ components have been setup be deferred on a queue, but there are some implementation issues and I haven't had a lot of enthusiasm expressed for the change.

Renegade605, 2025-03-21:
> I've previously proposed that any triggers invoked before _all_ components have been setup be deferred on a queue, but there are some implementation issues and I haven't had a lot of enthusiasm expressed for the change.

That seems a pretty reasonable solution to me. If there's an official way to upvote somewhere let me know.