esphome/issues #6878: 2025.3.0 Breaks ILI9341 and CST816 LCD running LVGL causes a bootloop.

billmaterial, 2025-03-20:
### The problem

The update from 2025.2.2 to 2025.3.0 creates a bootloop on my ILI9341 and CST816 LCD board running on an ESP32.
I have removed the CST816 touchscreen definition and only have the ILI9341 driver with the basic yaml for LVGL test screen but it still bootloops.

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

LVGL

### YAML Config

```yaml
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev
  platformio_options: 
    board_build.flash_mode: dio

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    
# Enable logging
debug:
  update_interval: 5s

logger:
  level: INFO
  logs:
    component: ERROR

# Enable Home Assistant API
api: 
  encryption:
    key: "redacted"
  reboot_timeout: 0s

# Allow Over-The-Air updates
ota:
  - platform: esphome
    password: !secret ota_password

# Allow provisioning Wi-Fi via serial
# improv_serial:
[... 362 lines cut ...]
[12:29:20]I (602) cpu_start: Pro cpu start user code
[12:29:20]I (602) cpu_start: cpu freq: 160000000 Hz
[12:29:20]I (602) cpu_start: Application information:
[12:29:20]I (607) cpu_start: Project name:     esphome-web-476304
[12:29:20]I (613) cpu_start: App version:      2025.3.0
[12:29:20]I (618) cpu_start: Compile time:     Mar 20 2025 12:20:23
[12:29:20]I (624) cpu_start: ELF file SHA256:  4760a651fa044366...
[12:29:20]I (630) cpu_start: ESP-IDF:          5.1.5
[12:29:20]I (634) cpu_start: Min chip rev:     v0.0
[12:29:20]I (639) cpu_start: Max chip rev:     v3.99 
[12:29:21]I (644) cpu_start: Chip rev:         v3.1
[12:29:21]I (649) heap_init: Initializing. RAM available for dynamic allocation:
[12:29:21]I (656) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[12:29:21]I (662) heap_init: At 3FFB8738 len 000278C8 (158 KiB): DRAM
[12:29:21]I (668) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[12:29:21]I (675) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[12:29:21]I (681) heap_init: At 400979E4 len 0000861C (33 KiB): IRAM
[12:29:21]I (689) spi_flash: detected chip: generic
[12:29:21]I (692) spi_flash: flash io: dio
[12:29:21]I (697) app_start: Starting scheduler on CPU0
[12:29:21]I (701) app_start: Starting scheduler on CPU1
[12:29:21]I (701) main_task: Started on CPU0
[12:29:21]I (711) main_task: Calling app_main()
[12:29:21]I (863) main_task: Returned from app_main()
```

Bazily1, 2025-03-20:
Similarly, when using lvgl, it goes into a bootloop.

Perhaps something was changed with the update
“[lvgl] Fix initialisation race condition (Bugfix) esphome#8369 by @clydebarrow”

```
[D][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[D][lvgl:000]: flex_update: update 0x3ffc35f0 container 	(in lv_flex.c line #208)
[D][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[D][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[D][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[D][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[D][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[D][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[D][lvgl:000]: lv_obj_Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:52:04]
[17:52:04]Core  0 register dump:
```

clydebarrow, 2025-03-20:
What happens if you remore the webserver?

clydebarrow, 2025-03-20:
Try setting `color_palette: 8bit` in the display config.

billmaterial, 2025-03-20:
@clydebarrow

I have tried both of those (one of my edits removes the color_palette: 8bit I had at first) and it is the same, constantly bootloops so I am a bit lost now as it is a bare minimum config and it still does not seem to work.

tonyis49, 2025-03-20:
just updated to 2025.3.0 and   both my co2 sensors with an lvgl display have the same problem

clydebarrow, 2025-03-20:
Still looking into it, but it is caused by the PR referenced above, so here is a work-around to restore the previous behaviour:

```
external_components:
  - source: github://esphome/esphome@9bc4f68d87828d5c3cb75ca0d6e6776458a01eef
    components: [lvgl]
```

closed 2025-03-21 by ssieb

Valcob, 2025-03-21:
I guess this one was closed too soon this is not the end of story even after setting the proper buff size :)

to get into a boot loop just use 
```
widgets:
        - textarea:
              id: t_input
              x: 130
              y: 147
              width: 61
              height: 37
              text: "25"   # <-- this is your pain in the butt which gives you a division by 0 and a kernel panic

Valcob, 2025-03-21:
breaks in both SDL and esp32 ....

clydebarrow, 2025-03-21:
> breaks in both SDL and esp32

Works fine for me with the fix.

Valcob, 2025-03-21:
yes you are right I missed one line in my example
```
text_font: DG_32
```
DG is a font with just numbers
as soon as we have the combination of two text+text_font we have a div0 exception removing either of two will fix the issue

clydebarrow, 2025-03-21:
Try this - there was a separate issue.

```
external_components:
  - source: github://pr#8444
    components: [lvgl]
```

billmaterial, 2025-03-21:
Looking good now, @clydebarrow thanks for all your hard work.

tonyis49, 2025-03-21:
works for me too, many thanks

randonkade, 2025-03-22:
Having same issue with rpi_dpi_rgb display and added external_components just make the screen flash but stays in bootloop.

ssieb, 2025-03-22:
@randonkade come ask on the esphome discord and provide serial logs.

clydebarrow, 2025-03-22:
Did you try this: https://community.home-assistant.io/t/help-with-failing-esp32-s3-box-3-after-esphome-upgrade-to-2025-3-0/866875/8

atoulmin, 2025-03-22:
I tried that too, my screen wont boot after patch. I had to roll back esphome to get them back up and running.

randonkade, 2025-03-22:
> [@randonkade](https://github.com/randonkade) come ask on the esphome discord and provide serial logs.

Okay Ill get my logs and add them in Discord later today. 

> Did you try this: https://community.home-assistant.io/t/help-with-failing-esp32-s3-box-3-after-esphome-upgrade-to-2025-3-0/866875/8

Yeah I did. Both together and individually. 
- No change with just pr#8442
- Screen briefly flash white at the top portion with pr#8444
- Both together same as pr#8444 alone - Screen flash white at the top portion 

external_components:
  - source: github://pr#8442
    components: [lvgl]    
  - source: github://pr#8444
    components: [lvgl]

Valcob, 2025-03-22:
@randonkade it's hard to make assumptions without a trace  + elf file or without a minimum code to reproduce your issue 
I would suggest the following 
1. In case you have multiple pages defined try to comment all of them and uncomment one by one till you get the boot loop so you know which one fails 
2. After you have your page try to comment all the widgets one the page  and then uncomment one by one to see which one causes the boot loop 
3. Use SDL to not write to you real device so you can quickly compile and check the UI
4. When you have the page and widget that fails you may post the code here so we can reproduce the issue too

clydebarrow, 2025-03-23:
Further fix added to 8444:
```
external_components:
  - source: github://pr#8444
    components: [lvgl]
```