esphome/issues #6879: 2nd DNS server from DHCP left unconsidered

Elix-g, 2025-03-20:
### The problem

My DHCP server distributes options for my two DNS servers 192.168.179.7 and 192.168.179.8 properly. While both DNS servers are properly applied by ESPHome on ESP8266/ESP32, ESPHome on Tuya CB2S, ignores the 2nd DNS server option but sets the first server twice.

### Which version of ESPHome has the issue?

2024.x.x, 2025.x.x

### What type of installation are you using?

pip

### What platform are you using?

BK72XX

### Board

noname Tuya board

### Component causing the issue

wifi

### YAML Config

```yaml
bk72xx:
  board: cb2s
  framework:
    version: latest

wifi:
  ap:
    ap_timeout: 5min
    manual_ip:
      gateway: 192.168.4.1
      static_ip: 192.168.4.1
      subnet: 255.255.255.0
    ssid: Fallback_${friendly_name}
    password: !secret wifi_password_fb
  domain: .privnet.xmas
  fast_connect: true
  networks:
    - ssid: !secret wifi_ssid_iot
      password: !secret wifi_password_iot
  power_save_mode: NONE
  reboot_timeout: 15min
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.3.0
INFO Reading configuration /opt/esphome/dashboard/bk7231-02.yaml...
INFO Starting log output from 192.168.179.250 using esphome API
INFO Successfully connected to bk7231-02 @ 192.168.179.250 in 0.008s
INFO Successful handshake with bk7231-02 @ 192.168.179.250 in 0.235s
[15:14:03][I][app:100]: ESPHome version 2025.3.0 compiled on Mar 20 2025, 14:44:13
[15:14:03][I][app:102]: Project Beken.CB2S version 0.0.1
[15:14:03][C][status_led:019]: Status LED:
[15:14:03][C][status_led:020]:   Pin: 11
[15:14:03][C][wifi:600]: WiFi:
[15:14:03][C][wifi:428]:   Local MAC: 50:8B:B9:BA:AF:AD
[15:14:03][C][wifi:433]:   SSID: [redacted]
[15:14:03][C][wifi:436]:   IP Address: 192.168.179.250
[15:14:03][C][wifi:439]:   BSSID: [redacted]
[15:14:03][C][wifi:441]:   Hostname: 'bk7231-02'
[15:14:03][C][wifi:443]:   Signal strength: -62 dB ▂▄▆█
[15:14:03][C][wifi:447]:   Channel: 6
[15:14:03][C][wifi:448]:   Subnet: 255.255.255.0
[15:14:03][C][wifi:449]:   Gateway: 192.168.179.1
[15:14:03][C][wifi:450]:   DNS1: 192.168.179.7
[15:14:03][C][wifi:451]:   DNS2: 192.168.179.7
```

DAVe3283, 2025-04-22:
What DHCP server are you using? I have no problems with any of my devices (8266, ESP32, ESP32-C6, etc.). They all show both of my DNS servers, both WiFi and Ethernet devices.

You'll want to do a packet capture of the DHCP exchange with the ESP and verify it is actually being told of both of your DNS servers.

DAVe3283, 2025-04-23:
Hmm, that sounds like a LibreTiny bug then, not an ESPHome bug. Or it is the interaction between LibreTiny and ESPHome, but I'm not familiar with LibreTiny at all. Hopefully someone else is and can help, sorry.

closed 2025-12-06 by github-actions[bot]