esphome/issues #6880: Font size checking issue

asolochek, 2025-03-20:
### The problem

I have a font which the configuration checking claims only exists in size 7, but if you change it to size 7 you end up with an exception "invalid pixel size"

[Tamzen5x9b.bdf.gz](https://github.com/user-attachments/files/19371636/Tamzen5x9b.bdf.gz)

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  name: test-project
  friendly_name: Test Project
  board_type: esp32dev
  room: "General"
  log_level: DEBUG

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  area: ${room}

esp32:
  board: ${board_type}
  framework:
    type: esp-idf
#    type: arduino

font:
  - file: fonts/Tamzen5x9b.bdf
    id: tamzen5x9b
    size: 9

wifi:
  ssid: XX
  password: XX
```

### Anything in the logs that might be useful for us?

```txt
Error when size is 9:

  Font /Volumes/config/esphome/fonts/Tamzen5x9b.bdf only has size 7 available.
  - file: fonts/Tamzen5x9b.bdf
    id: tamzen5x9b
    size: 9

Error when size is 7:

INFO Generating C++ source...
Traceback (most recent call last):
  File "/opt/homebrew/bin/esphome", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 1057, in main
    return run_esphome(sys.argv)
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 1044, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 460, in command_compile
    exit_code = write_cpp(config)
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 213, in write_cpp
    generate_cpp_contents(config)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 225, in generate_cpp_contents
    CORE.flush_tasks()
    ~~~~~~~~~~~~~~~~^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
    ~~~~^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/__main__.py", line 205, in wrapped
    await coro(conf)
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/esphome/components/font/__init__.py", line 512, in to_code
    font.set_pixel_sizes(size, 0)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/opt/homebrew/Cellar/esphome/2025.3.0/libexec/lib/python3.13/site-packages/freetype/__init__.py", line 1311, in set_pixel_sizes
    if error: raise FT_Exception(error)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
freetype.ft_errors.FT_Exception: FT_Exception:  (invalid pixel size)
```

Zoogara, 2025-03-21:
Issues with fonts after update to 2025.3.0, True-Type fonts that use to scale now report: 

`  Font /config/esphome/fonts/calibri.ttf only has sizes 12, 13, 15, 16, 17, 19 available.
  - file: fonts/calibri.ttf
    id: calibri_20
    size: 20`

Example of font definition is:

`font:
  # Forecast text
  - file: "fonts/calibri.ttf"
    id: calibri_12
    size: 12
    glyphs: [
       '&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z', 'å', 'ä', 'ö', '/', '–']
  # Weather display detail text
  - file: "fonts/calibri.ttf"
    id: calibri_20
    size: 20
    `

Is this possibly an undocumented breaking change related to [PR#8407](https://github.com/esphome/esphome/pull/8407)?

asolochek, 2025-03-21:
Yeah, I thought fonts previously would scale (although they'd look like crap) but I couldn't remember for sure which font types would scale and which wouldn't, and I was too lazy to test.

Zoogara, 2025-03-21:
Ah - ok so bitmap fonts didn't scale in the previous version? How do I tell if a font is bitmap or vector?  

Calibri is a pretty standard font, I can use it in other apps at any size.

clydebarrow, 2025-03-21:
> Ah - ok so bitmap fonts didn't scale in the previous version?

Correct.

> How do I tell if a font is bitmap or vector?

Other than what ESPHome now tells you, I have no idea.

> Calibri is a pretty standard font, I can use it in other apps at any size.

Not with that font file. Yes, it's a common _font_, but your _file_ is evidently not a standard vector representation of it.

clydebarrow, 2025-03-21:
Fixed with https://github.com/esphome/esphome/pull/8443

```
external_components:
  - source: github://pr#8443
    components: [font]
```

clydebarrow, 2025-03-21:
> but your _file_ is evidently not a standard vector representation of it.

So turns out I was wrong - the file *does* have bitmaps but it _also_ has scalable vector glyphs. The PR referenced above now deals correctly with that. The scalable version will be used unless one of the bitmap sizes is requested at 1bpp.

Zoogara, 2025-03-21:
> So turns out I was wrong - the file _does_ have bitmaps but it _also_ has scalable vector glyphs. The PR referenced above now deals correctly with that. The scalable version will be used unless one [...]

So ok - does that mean it should compile or not.  I was just about to download the font again, but no point if the presence of a bitmap font stops me compiling.

Or am I misunderstanding how this works?

ssieb, 2025-03-21:
It works.  If you pick a size (and 1bpp which is the default) that has a bitmap version, it will use that now.  Otherwise, it will use the vector forms.  See the difference between sizes 19 and 20.

Zoogara, 2025-03-21:
Right - and what about the 25 point and 40 point fonts?  

```
  # weather display time and temp
  - file: "fonts/calibri.ttf"
    id: calibri_25
    size: 25
    
  # weather display wind speed
  - file: "fonts/calibri.ttf"
    id: calibri_40
    size: 40
```

Neither of those will compile.

It looks adequate on the 320 x 240 displays I use:

[image]

I the answer to try a different font?  Or another version of the Calibre tif file?

clydebarrow, 2025-03-21:
> Right - and what about the 25 point and 40 point fonts?

Those work if you are using the PR as an external component.

clydebarrow, 2025-03-21:
> Adding bpp doesn't solve the compiling error.

Nobody said it would.

xaxix, 2025-03-21:
> > Right - and what about the 25 point and 40 point fonts?
> 
> Those work if you are using the PR as an external component.

How do i use the PR as an external component?

xaxix, 2025-03-21:
As the procedure was not clear to me:

Add the code
`external_components:
  - source: github://pr#8443
    components: [font]`
into the YAML of the respective device configuration. This refers to a patch in Github and the compilation runs through without an error message.

For Example
`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxxxxx"
    password: "xxxxxxxx"

captive_portal:

external_components:
  - source: github://pr#8443
    components: [font]

switch:
  - platform: restart
    name: "Inkplate Reboot"
    id: reboot`

closed 2025-03-24 by kbx81