esphome/issues #6881: After latest update  2025.3.0 my hub75 display dont display text anymore or not always

jl-p, 2025-03-20:
### The problem

After i upgraded my wordclock and status screen wont show any text anymore. sometimes some lines work . some lines i couldnt reproduce, i added logging and that showed the text should had been there.
i downgraded my esphome to the 2025.2.2 and everything works again. 

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest official

### What platform are you using?

ESP32

### Board

devkitv1

### Component causing the issue

hub75 

### YAML Config

```yaml
esphome:
  name: wordclock
  friendly_name: WordClock

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
# Enable Home Assistant API

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

substitutions:
  DelayTimePage2: 10s
  xscrollpadding: "4" # in pix 

captive_portal:

external_components:
  - source: github://TillFleisch/ESPHome-HUB75-MatrixDisplayWrapper@main

 
 

globals:
- id: page
  type: int
  initial_value: "1"
- id: WhichCounter
  type: int
  initial_value: "0"
[... 398 lines cut ...]

script:
  - id: start
    mode: restart
    then:
      - logger.log: "Script started."
      - repeat:
          count: 15
          then:
            - delay: 2s
            - number.to_min: schermbrightness
            - logger.log: "Setting brightness to minimum."
            - delay: 500ms
            - number.to_max: schermbrightness
            - logger.log: "Setting brightness to maximum."
      - delay: 60s
      - display.page.show: !lambda |-
            return id(page1);
```

### Anything in the logs that might be useful for us?

```txt
the logs looks normal but no text on my display
```

truewombat, 2025-03-23:
I have the same problem for 2025.3.0 and also 2025.3.1. Random display lines were not printed and just blank making text barely readable. Downgrading to 2024.11.3 "fixed" the issue. 
The device is an airgradient  esp8266 d1 mini based.

yatush, 2025-03-24:
Might be connected - happened to me with CYD as well, I try to print a Google font with "center" attribute, and it seems that symbols that are as wide as it gets look good, but the ones that are not very wide, are not centered correctly.