esphome/issues #6883: web_server component added automatically in ESPHome2025.3.0 / esp32doit-devkit-v1 board

vit5041, 2025-03-21:
### The problem

I noticed with my real device, but in order to reproduce you can make an empty project with the following board:

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

you can make minimal sketch without adding web_server: line. Then compile and check that in main.cpp the web_server component has been added.

started in ESPHome 2025.3.0

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-23:
What's the full yaml?

vit5041, 2025-03-23:
> What's the full yaml?

the min sketch:

`
esphome:
  name: test

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

api:

captive_portal:
`

plplaaa2, 2025-03-23:
My case

```

#substitutions:
#  name: esp32-c6-test
#  board: esp32-c6-devkitc-1
#  device_description: esp32-c6-test
#  s_led: GPIO8  

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.7.3
  platformio_options:    
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    #board_upload.maximum_ram_size: 524288

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.1.5
    source: https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
    platform_version: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
    sdkconfig_options:
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y

# Enable logging
logger:
  level: DEBUG
  #baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  - platform: esphome
    password: !secret hotspot_pw

wifi:  
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: HIGH
  fast_connect: True
  output_power: 8.5dB     # ESP32 C3
  manual_ip: 
   static_ip: $s_ip
   gateway: 192.168.1.1
   subnet: 255.255.255.0
   dns1: 1.1.1.1
   dns2: 8.8.8.8
  on_connect:
    then:
      - light.turn_off: builtin_led

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $name
    password: !secret hotspot_pw

captive_portal:

time:
  - platform: sntp
    id: time_comp
    timezone: UTC-9
     - time.bora.net

text_sensor:
  - platform: version
    name: "${name} Esphome Version"
  - platform: wifi_info
    ip_address:
      name: ${name} ESP IP Address

binary_sensor:
  - platform: status
    name: "${name} Connection Status"
      
sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    internal: true
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "${name} WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""
    icon: mdi:wifi    
    
  - platform: internal_temperature
    name: '${name} Internal temperature'

  - platform: uptime         
    name: Uptime Sensor     
    id: up_time       
    filters:            
      - lambda : return x / 60;
  
button:    
  - platform: restart
    name: "${name} Restart"
    entity_category: diagnostic 

light:
  - platform: esp32_rmt_led_strip
    id: builtin_led
    rgb_order: GRB
    pin: $s_led
    num_leds: 1
    #rmt_channel: 0
    chipset: ws2812
    name: "on board light"
    effects:
      - pulse:
          name: "Slow Pulse"
          transition_length: 250ms
          update_interval: 250ms
          min_brightness: 50%
          max_brightness: 100%
      - pulse:
          name: "Fast Pulse"
          transition_length: 100ms
          update_interval: 100ms
          min_brightness: 50%
          max_brightness: 100%
```
I don't use web_server except in special cases.
However, in 2025.3.0, web_server is automatically built just by updating.
In my case, it's not just a specific device, it's all the devices.

ssieb, 2025-03-23:
It seems to be specific to esp-idf.  It doesn't happen with arduino.

ssieb, 2025-03-24:
Caused by https://github.com/esphome/esphome/pull/7538

chertvl, 2025-05-25:
Same here.
How can I compile bt-proxy device via esp-idf framework, such a dock says, if web_server should be disabled because of malfunction when both components enabled?

[image]

Smandurlo, 2025-05-25:
I found out you have to remove captive portal and the web server is gone

chertvl, 2025-05-25:
> I found out you have to remove captive portal and the web server is gone

Thanks, it works. But a little bit strange. Totally another behaviour, then arduino framwork, or then older esp-idf.

closed 2025-06-14 by bdraco