esphome/issues #6884: 2025.3.0 and SH1106 128x64 shows missing pixels on some font sizes

MallocArray, 2025-03-21:
### The problem

ESP32-C3 and ESP8266 with "SH1106 128x64" display and Open Sans font size 9 shows missing pixels in various letter and numbers when using 2025.3.0, where it was correct with 2025.2.1
Notice the gaps in CO2, PM2.5, TVOC, and NOx, but other fonts sizes are not impacted

[image]

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

ESP32-C3

### Component causing the issue

ssd1306_i2c with model: "SH1106 128x64"

### YAML Config

```yaml
font:
  - file: "gfonts://Open Sans"
    id: open_sans_9
    size: 9

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    id: oled_display
    address: 0x3C
    pages:
      - id: airgradient_default
        lambda: |-
          it.printf(0,19, id(open_sans_9), "CO2");
          it.printf(54, 19, id(open_sans_9), "PM2.5");
          it.printf(104,18, id(open_sans_9), "TVOC");
          it.printf(104,41, id(open_sans_9), "NOx");
```

### Anything in the logs that might be useful for us?

```txt
No errors present

[16:14:39][C][ssd1306_i2c:023]:   Rotations: 0 °
[16:14:39][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[16:14:39][C][ssd1306_i2c:024]:   Address: 0x3C
[16:14:39][C][ssd1306_i2c:025]:   Model: SH1106 128x64
[16:14:39][C][ssd1306_i2c:027]:   External VCC: NO
[16:14:39][C][ssd1306_i2c:028]:   Flip X: YES
[16:14:39][C][ssd1306_i2c:029]:   Flip Y: YES
[16:14:39][C][ssd1306_i2c:030]:   Offset X: 0
[16:14:39][C][ssd1306_i2c:031]:   Offset Y: 0
[16:14:39][C][ssd1306_i2c:032]:   Inverted Color: NO
[16:14:39][C][ssd1306_i2c:033]:   Update Interval: 1.0s
```

### Additional information

Full config file for the device can be found:

ESP32-C3
https://github.com/MallocArray/airgradient_esphome/blob/5.1.0/full_config/ag-one.yaml

ESP8266
https://github.com/MallocArray/airgradient_esphome/blob/5.1.0/full_config/ag-pro.yaml

MallocArray, 2025-03-21:
Same config with 2025.2.2

[image]

ssieb, 2025-03-21:
The font renderer has been changed to freetype.  At 9 pixels, you should be using a bitmap font anyway.

MallocArray, 2025-03-21:
I don't even know what a bitmap font is.  Are gfonts available as bitmap fonts? I don't want to have to include any external files outside of the main yaml file, so gfonts are important to me.

Does freetype not work at size 9 at all, or what is the limiting factor?

ssieb, 2025-03-21:
Bitmap fonts specify every pixel instead of shapes.  Rendering outline fonts at such small sizes with no anti-aliasing is not going to work well.  I don't know what pillow did differently.  I will look to see if there's an option to get freetype to work better.  But it will never be better than a bitmap font.

dsfsdbvxc, 2025-03-22:
Having the same issue since updating from 2025.2.2, to 2025.3.0

Same issue with text not displaying correctly on my [Waveshare display ](https://www.waveshare.com/wiki/2.42inch_OLED_Module) (embedded SSD1309 driver, SPI).

Display on the left  hasn't been updated and displays fine, one on the right has and doesnt show text correctly, pixilated and not aligned etc. Compiles and builds fine, no errors.

[image]

`# OLED Display (SSD1306 SPI)
display:
  - platform: ssd1306_spi
    model: SSD1306 128x64
    cs_pin: GPIO33
    dc_pin: GPIO14
    reset_pin: GPIO32
    spi_id: spi_bus
    update_interval: 10s
    id: oled_display
    lambda: |-
      if (id(oled_toggle).state) {  // If switch is ON, show content
        it.image(2, 24, id(icon_elec));
        it.printf(22, 9,  id(my_font), "Temp       : %.1f°C", id(rack_temp_sensor).state);
        it.printf(22, 22, id(my_font), "Unraid     : %.1f W", isnan(id(server_avg).state) ? 0.0 : id(server_avg).state);
        it.printf(22, 35, id(my_font), "UPS Left : %s", id(ups_runtime).state.c_str());
        it.printf(22, 48, id(my_font), "UPS Bat  : %s", id(ups_battery).state.c_str());
      } else {
        it.clear();  // Clears screen instead of displaying "TEST"
      }

# Image
image:
  - file: "images/electricity-alt-20.bmp"
    id: icon_elec
    type: BINARY
    resize: 20x20

# Font
font:
  - file: "fonts/Arial-bold.ttf"
    id: my_font
    size: 10`

ronnieSVK, 2025-03-22:
Same issue here with "Waveshare clone?" 2.42" i2c  128x64 display... For instance refsan.ttt with size "9" has errors, also some icons...

MallocArray, 2025-03-22:
@ronnieSVK Could you share some screenshots with what you are seeing?

dsfsdbvxc, 2025-03-25:
Just for info, this is still an issue in 2025.3.1

clydebarrow, 2025-03-25:
https://github.com/esphome/esphome/pull/8463

closed 2025-03-25 by clydebarrow