esphome/issues #6886: Update 2025.3.0 broke dallas sensor

aizmob, 2025-03-22:
### The problem

Dear friends! I have an issue with ds18b20 sensor after update to 2025.3.0. The second one is working fine.
Both sensors were OK during two years with current versions of esphome.

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

dallas

### YAML Config

```yaml
esphome:
  name: "esp-heater"
  friendly_name: Heater Temperature

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
# Example configuration entry for finding
# Service UUIDs and iBeacon UUIDs and identifiers
#  level: VERY_VERBOSE
#esp32_ble_tracker:

# Enable Home Assistant API
api:
  encryption:
    key: "gInXm3wPx/yEMneCf/aYN0iIqtA+OU3iS/phoBS+CZo="

ota:
  - platform: esphome
 #   password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.14.41
    gateway: 192.168.14.254
    subnet: 255.255.255.0
    dns1: 192.168.14.254

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
[... 182 lines cut ...]
[06:01:12][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[06:01:12][D][sensor:094]: 'ds18b20': Sending state nan °C with 2 decimals of accuracy
[06:01:27][D][sensor:094]: 'WiFi Signal dB': Sending state -40.00000 dB with 0 decimals of accuracy
[06:01:27][D][sensor:094]: 'WiFi Signal pct': Sending state 100.00000 % with 0 decimals of accuracy
[06:01:38][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=41.4°C
[06:01:38][D][sensor:094]: 'ds18b20_2': Sending state 41.43750 °C with 2 decimals of accuracy
[06:01:42][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[06:01:42][W][dallas.temp.sensor:139]: 'ds18b20' - Scratch pad checksum invalid!
[06:01:42][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[06:01:42][D][sensor:094]: 'ds18b20': Sending state nan °C with 2 decimals of accuracy
[06:02:08][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=41.0°C
[06:02:08][D][sensor:094]: 'ds18b20_2': Sending state 41.00000 °C with 2 decimals of accuracy
[06:02:12][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[06:02:12][D][dallas.temp.sensor:054]: 'ds18b20': Got Temperature=42.0°C
[06:02:12][D][sensor:094]: 'ds18b20': Sending state 42.00000 °C with 2 decimals of accuracy
[06:02:27][D][sensor:094]: 'WiFi Signal dB': Sending state -40.00000 dB with 0 decimals of accuracy
[06:02:27][D][sensor:094]: 'WiFi Signal pct': Sending state 100.00000 % with 0 decimals of accuracy
[06:02:38][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=40.5°C
[06:02:38][D][sensor:094]: 'ds18b20_2': Sending state 40.50000 °C with 2 decimals of accuracy
[06:02:42][D][dallas.temp.sensor:054]: 'ds18b20': Got Temperature=41.4°C
```

### Additional information

[image]

ssieb, 2025-03-22:
I don't know why one sensor wouldn't be working and there's nothing specific in the last release that should affect this.
But try this and see if it makes any difference:
```yaml
external_components:
  - source: github://ssieb/esphome@onewire2
    components: [ gpio, esp32, one_wire, dallas_temp ]
    refresh: 1min
```

aizmob, 2025-03-22:
Thanks for the promptness! Will try a little later...

MallocArray, 2025-03-22:
For what it is worth, I have 2 dallas sensors with the default update interval on ESP8266 and I'm not seeing the same behavior on 2025.3.0

aizmob, 2025-03-22:
> try this and see if it makes any difference
> 

1 hour of successful operation with your suggestion.  Let's see further.
Thanks again!

Shady313FN, 2025-03-24:
Hello!

I have the same issue. The versions are same as OP. I have 26 sensors and was using it in the last 6 months without any issue. In the 2025.3.0 some completely stopped working (about 10pcs). Sometimes I got values but mainly not. I can't send logs, because after trying some restarting and waiting for some hours, I downgraded back to 2025.2.2 and everything is working fine again. My sensors are mandatory for my floor heating system. It's laid into the concrete. I have 2 sensors on the same cable for redundancy. But redundancy here not helped, because both of the sensors were unavailable in the same place.

aizmob, 2025-03-24:
> I have the same issue.

Your case shows it wasn't coincidence as I thought yesterday! Shall try to rollback.
Thank You!

Chranislav, 2025-03-24:
Hi,
I have the same problem. Also in version 2025.3.1.

aizmob, 2025-03-24:
So we have _a real_ problem.

ssieb, 2025-03-25:
Did you try the external component shown earlier?

ramarro123, 2025-03-25:
if you mean 
external_components:
  - source: github://ssieb/esphome@onewire2
    components: [ gpio, esp32, one_wire, dallas_temp ]
    refresh: 1min

yes, and it didn't worked

don't know if it's a coincidence, but when it restart after OTA update, it worked for a while (i have an alert that send watsapp after 30 mins of missing data for temp [it's an aquarium i don't want to kill corals or fish so have some extra checks]) 
i don't know if it worked for 1 min or for 29.9 but i can say that after 30m it start acting nuts.

the above log is, for the record, without external_component mod

ssieb, 2025-03-25:
Yes, I know it was.  That's why I asked.  What does the log look like with it?

aizmob, 2025-03-25:
> Did you try the external component shown earlier?

Yes with no luck:
https://github.com/esphome/issues/issues/6886#issuecomment-2744874604

ramarro123, 2025-03-25:
i didn't saved the prev log, i will try when i come back home and let you know!

ssieb, 2025-03-25:
I'll see what the log shows, but if it doesn't work with that branch, then I'll need to see an oscilloscope or logic analyzer trace.

aizmob, 2025-03-25:
> I'll see what the log shows

Installed v 2025.3.1 again, recompiled fw with your settings and am waiting for sensor loss to get a log. It may take 1-2 hours. 
Shall show you logs asap.

aizmob, 2025-03-25:
@ssieb , caught an error:
```
INFO ESPHome 2025.3.1
INFO Reading configuration /config/esphome/esp-heater.yaml...
INFO Updating https://github.com/ssieb/esphome.git@onewire2
INFO Generating C++ source...
INFO Compiling app...
Processing esp-heater (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
RAM:   [=         ]  12.4% (used 40680 bytes from 327680 bytes)
Flash: [=====     ]  50.4% (used 924433 bytes from 1835008 bytes)
========================= [SUCCESS] Took 3.81 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.14.41 port 3232...
INFO Connected to 192.168.14.41
INFO Uploading /data/build/esp-heater/.pioenvs/esp-heater/firmware.bin (930208 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.97 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.14.41 using esphome API
INFO Successfully connected to esp-heater @ 192.168.14.41 in 7.186s
INFO Successful handshake with esp-heater @ 192.168.14.41 in 0.143s
[18:25:51][I][app:100]: ESPHome version 2025.3.1 compiled on Mar 25 2025, 17:50:56
[18:25:51][C][wifi:600]: WiFi:
[18:25:51][C][wifi:428]:   Local MAC: CC:DB:A7:12:4B:C4
[18:25:51][C][wifi:433]:   SSID: 'aiz'[redacted]
[18:25:51][C][wifi:436]:   IP Address: 192.168.14.41
[18:25:51][C][wifi:440]:   BSSID: 50:FF:20:AC:56:6C[redacted]
[18:25:51][C][wifi:441]:   Hostname: 'esp-heater'
[18:25:51][C][wifi:443]:   Signal strength: -46 dB ▂▄▆█
[18:25:51][C][wifi:447]:   Channel: 1
[18:25:51][C][wifi:448]:   Subnet: 255.255.255.0
[18:25:51][C][wifi:449]:   Gateway: 192.168.14.254
[18:25:51][C][wifi:450]:   DNS1: 192.168.14.254
[18:25:51][C][wifi:451]:   DNS2: 0.0.0.0
[18:25:51][C][logger:177]: Logger:
[18:25:51][C][logger:178]:   Max Level: DEBUG
[18:25:51][C][logger:179]:   Initial Level: DEBUG
[18:25:51][C][logger:181]:   Log Baud Rate: 115200
[18:25:51][C][logger:182]:   Hardware UART: UART0
[18:25:51][C][gpio.one_wire:022]: GPIO 1-wire bus:
[18:25:51][C][gpio.one_wire:023]:   Pin: GPIO23
[18:25:51][C][gpio.one_wire:084]:   Found devices:
[18:25:51][C][gpio.one_wire:086]:     0xc80821b0a5319c28 (DS18B20)
[18:25:51][C][gpio.one_wire:086]:     0x0900000016e44928 (DS18B20)
[18:25:51][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[18:25:51][C][dallas.temp.sensor:034]:   Address: 0xc80821b0a5319c28 (DS18B20)
[18:25:51][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[18:25:51][C][dallas.temp.sensor:036]:   Update Interval: 30.0s
[18:25:51][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[18:25:51][C][dallas.temp.sensor:034]:   Address: 0x0900000016e44928 (DS18B20)
[18:25:51][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[18:25:51][C][dallas.temp.sensor:036]:   Update Interval: 30.0s
[18:25:51][C][copy.sensor:015]: Copy Sensor 'WiFi Signal pct'
[18:25:51][C][copy.sensor:015]:   Device Class: 'power_factor'
[18:25:51][C][copy.sensor:015]:   State Class: 'measurement'
[18:25:51][C][copy.sensor:015]:   Unit of Measurement: '%'
[... 166 lines cut ...]
[18:38:14][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[18:38:14][D][dallas.temp.sensor:137]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[18:38:14][D][sensor:094]: 'ds18b20': Sending state nan °C with 2 decimals of accuracy
[18:38:23][D][sensor:094]: 'WiFi Signal dB': Sending state -46.00000 dB with 0 decimals of accuracy
[18:38:23][D][sensor:094]: 'WiFi Signal pct': Sending state 100.00000 % with 0 decimals of accuracy
[18:38:31][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=30.8°C
[18:38:31][D][sensor:094]: 'ds18b20_2': Sending state 30.81250 °C with 2 decimals of accuracy
[18:38:43][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[18:38:44][W][dallas.temp.sensor:132]: 'ds18b20' - Scratch pad checksum invalid!
[18:38:44][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[18:38:44][D][dallas.temp.sensor:137]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[18:38:44][D][sensor:094]: 'ds18b20': Sending state nan °C with 2 decimals of accuracy
[18:39:01][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=30.6°C
[18:39:01][D][sensor:094]: 'ds18b20_2': Sending state 30.56250 °C with 2 decimals of accuracy
[18:39:13][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[18:39:14][W][dallas.temp.sensor:132]: 'ds18b20' - Scratch pad checksum invalid!
[18:39:14][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[18:39:14][D][dallas.temp.sensor:137]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[18:39:14][D][sensor:094]: 'ds18b20': Sending state nan °C with 2 decimals of accuracy
[18:39:23][D][sensor:094]: 'WiFi Signal dB': Sending state -45.00000 dB with 0 decimals of accuracy
[18:39:23][D][sensor:094]: 'WiFi Signal pct': Sending state 100.00000 % with 0 decimals of accuracy
[18:39:31][D][dallas.temp.sensor:054]: 'ds18b20_2': Got Temperature=30.4°C
[18:39:31][D][sensor:094]: 'ds18b20_2': Sending state 30.37500 °C with 2 decimals of accuracy

```

ssieb, 2025-03-25:
The sensor just stops responding, which really looks like a hardware issue of some sort.  Does it keep working with the previous version?

aizmob, 2025-03-26:
@ssieb 
Could check few minutes ago only: NO, v 2025.2.2 shows the same bad picture
[image]
That is, we can well conclude it is a sensor problem. At least in my case.
It's very strange because my previous attempt to run 2025.2.2 was successful and sensor worked perfectly during ~30 hours. A coincidence again, sorry.

ramarro123, 2025-03-26:
Hello,

i did put 

```
external_components:
  - source: github://ssieb/esphome@onewire2
    components: [ gpio, esp32, one_wire, dallas_temp ]
    refresh: 1min
```

on top of my yml, build, restart and after a few mins that's what i got

```
[07:51:00][D][dallas.temp.sensor:054]: 'Temp 1': Got Temperature=25.7°C
[07:51:00][D][sensor:093]: 'Temp 1': Sending state 25.68750 °C with 1 decimals of accuracy
[07:51:33][D][dallas.temp.sensor:054]: 'Temp 2': Got Temperature=25.9°C
[07:51:33][D][sensor:093]: 'Temp 2': Sending state 25.87500 °C with 1 decimals of accuracy
...... then after a while.....

[08:37:47][D][esp32.preferences:114]: Saving 1 preferences to flash...
[08:37:47][D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[08:37:48][D][dallas.temp.sensor:054]: 'Temp 1': Got Temperature=25.5°C
[08:37:48][D][sensor:093]: 'Temp 1': Sending state 25.50000 °C with 1 decimals of accuracy
[08:37:58][D][dallas.temp.sensor:054]: 'Temp 2': Got Temperature=25.7°C
[08:37:58][D][sensor:093]: 'Temp 2': Sending state 25.68750 °C with 1 decimals of accuracy
....... (this is the last correct read, no other log are displayed with evident error).....

[08:49:47][D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[08:49:47][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:49:48][W][dallas.temp.sensor:132]: 'Temp 1' - Scratch pad checksum invalid!
[08:49:48][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:49:48][D][dallas.temp.sensor:134]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[08:49:48][D][sensor:093]: 'Temp 1': Sending state nan °C with 1 decimals of accuracy
[08:49:58][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:49:58][W][dallas.temp.sensor:132]: 'Temp 2' - Scratch pad checksum invalid!
[08:49:58][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:49:58][D][dallas.temp.sensor:134]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[08:49:58][D][sensor:093]: 'Temp 2': Sending state nan °C with 1 decimals of accuracy

```

Shady313FN, 2025-03-26:
> I have the same issue. [...]

Strange, in the afternoon I upgraded to 2025.3.2 and everything is working fine, I don't have any errors. Is anything changed? I had problems with 2025.3.0

gotnone, 2025-04-26:
Does this have anything to do with GTK rekeying issue #3415?

In that issue it seems like more than one DS18B20 was causing a problem.

jimf862, 2025-05-26:
up to version 2025.4.2 all my dallas sensors were fine, upgraded to latest 2025.5.0 and all failed to be recognised, rolled back & all working again. Just a comment, no way can i open them all up & restructure the build of each.

ssieb, 2025-05-26:
@jimf862 what ESPs? parasitic mode? external pull-up resistor?

gitglint, 2025-05-27:
Hello,

I have a similar problem. I hope it's okay to add this here, I came to this page via Google and couldn't find anything else about it.

I have several Dallas ds18b20 sensors connected to several D1-Mini (ESP8266). These sensors are connected (no external pull-up resistor) to the GPIO port D4, as well as to GND and +5V of the D1-Mini.

The sensors have been delivering data without any problems for over a year without any errors or failures.

Three days ago, on May 24, 2025, I updated Home Assistant ESPHome Device Builder to the 2025.5.0, and since then, some of the sensors have been delivering incorrect data or no data at all. The errors started simultaneously immediately after the update. I consider it highly unlikely that a hardware defect would cause the Dallas sensors on independent ESP8266s to "randomly" fail at the same time right at the time of the update.

On the ESP8266 with five Dallas sensors connected to GPIO pin D4, none of the sensors have been delivering data since the update.

For the ESP with three Dallas sensors connected to GPIO pin D4, two sensors provide correct data, while one sensor provides invalid data every few minutes (air temperature above 80 degrees Celsius instead of 20 degrees Celsius).

For ESPs with only one Dallas sensor connected to GPIO pin D4, no errors occur.

UPDATE: Today I updated to ESPHome Device Builder 2025.5.1, and there was no improvement in these errors.

Attached are the configuration and logs (from the update) of a D1-Mini with 5 Dallas sensors. Maybe this helps.

[heizungsraum.txt](https://github.com/user-attachments/files/20455875/heizungsraum.txt)

[logs_heizungsraum_run.txt](https://github.com/user-attachments/files/20455880/logs_heizungsraum_run.txt)

gitglint, 2025-05-27:
Hello,

adding external_component

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ esp8266 ] 
```
    
before the board configuration (like it was mentioned in https://github.com/esphome/issues/issues/7039) solved the problem for me.

jimf862, 2025-05-27:
in reply to ssieb they are all esp8266 and there is no pull up resistor fitted. I understand now it is recommended to have a 4k7 pull up, but as i say these are embedded in the house, to replace is not impossible obviously but is it worth it, if i remain on old version.
Not sure if they would continue to run if i update ESP but not each sensor software

andypnz, 2025-06-03:
Hi I have the same issue on 2025.5 on the same device (Pico W)….. all ok on 2025.4

> Hello everyone, I have a similar problem, my DS18B20 sensor stopped working with 2025.5.3 and ESPHome 2025.5.0. My sensor is running on the Pi Pico W and the part of the log looks like that:
> 
> ```
> [17:04:48][C][gpio.one_wire:022]: GPIO 1-wire bus:
> [17:04:48][C][gpio.one_wire:023]:   Pin: GPIO1
> [17:04:48][W][gpio.one_wire:082]:   Found no devices!
> [17:04:48][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
> [17:04:48][W][dallas.temp.sensor:031]:   Unable to select an address
> ```

ssieb, 2025-06-07:
2025.5.2 should be good for all except libretiny devices.