esphome/issues #6889: loger warning at deep sleep

CrazyUs3r, 2025-03-23:
### The problem

The problem is like deep sleep, if the esp go in deep sleep 1m läter kommt this log.

I use api

Logger: aioesphomeapi.reconnect_logic
Quelle: runner.py:154
Erstmals aufgetreten: 22. März 2025 um 08:09:19 (32 Vorkommnisse)
Zuletzt protokolliert: 06:57:53

Can't connect to ESPHome API for wetterstation @ 192.168.178.233: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.233', port=6053))]: [Errno 113] Connect call failed ('192.168.178.233', 6053) (SocketAPIError)
Can't connect to ESPHome API for wetterstation @ 192.168.178.233: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.233', port=6053))] (TimeoutAPIError)

### Which version of ESPHome has the issue?

all

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

api / deep sleep

### YAML Config

```yaml
Api:

Deep_sleep:
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-23:
What are you expecting?  The device isn't available.
But make sure you added the device to HA after you configured deep sleep.  If you didn't, then remove it from the integrations list and add it again.  It will handle it somewhat better in that case.

CrazyUs3r, 2025-03-23:
The value are save and after wake up ha show the New value but api make after join deep sleep every time this log fault

d1mdev, 2025-04-03:
i see same warnings after march update of esphome.
solution "remove then add again" don't help.
esp32, esp32c3

CrazyUs3r, 2025-04-11:
So same problem by ha 205.04.01 an os 15.1

CrazyUs3r, 2025-04-29:
So update to ESPHome 2025.4.1 and the same error log

d1mdev, 2025-04-29:
i suspect that its not an esphome problem, but home assistant. it uses the same component (aioesphomeapi), but another version.
As i see in esphome logs esp goes to sleep fine. End part of log looks ok:
```
[22:00:35][I][deep_sleep:060]: Beginning Deep Sleep
[22:00:35][I][deep_sleep:062]: Sleeping for 600000000us
[22:00:35][D][deep_sleep:064]: Delay 300 ms for API to settle a bit
[22:00:36][D][esp32.preferences:114]: Saving 4 preferences to flash...
[22:00:36][D][esp32.preferences:143]: Saving 4 preferences to flash: 1 cached, 3 written, 0 failed
INFO Processing expected disconnect from ESPHome API for esp32-c3-zero-1 @ 192.168.X.X
WARNING Disconnected from API
```
And about one minute later Home Assistant writes new warning:
```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:154
First occurred: 09:08:30 (76 occurrences)
Last logged: 22:01:18

Can't connect to ESPHome API for esp32-c3-zero-1 @ 192.168.X.X: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.X.X', port=6053))] (TimeoutAPIError)
Can't connect to ESPHome API for esp32-c3-zero-1 @ 192.168.X.X: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.X.X', port=6053))]: [Errno 113] Connect call failed ('192.168.X.X', 6053) (SocketAPIError)
```

closed 2025-12-05 by github-actions[bot]