esphome/issues #6891: Error: ESP does not have enough space to store OTA file (2025.3.1)

pickonedev, 2025-03-23:
### The problem

Hi!

Till 2025.3.0, everything was ok, I coulud update all the devices, but from 2025.3.1, I get "Error: ESP does not have enough space to store OTA file" for my Athom Human Presence devices. This is the ESPHome code:
Any solution? Thanks!

```
substitutions:
  name: bedroom_1_presence
  friendly_name: Bedroom 1 Presence

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp8266:
  board: esp8285
  restore_from_flash: true

preferences:
  flash_write_interval: 1min

# Enable logging
logger:

api:
  encryption:
    key: "xxx"

ota:
  platform: esphome

mdns:
  disabled: false

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

light:
  - platform: status_led
    name: "Status LED"
    id: led
    pin: GPIO16
    disabled_by_default: true
    entity_category: config

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 115200
  # debug:
  #   direction: BOTH
  #   dummy_receiver: true
  #   after:
  #     delimiter: "\n"
  #   sequence:
  #     - lambda: UARTDebug::log_string(direction, bytes);
[... 286 lines cut ...]
Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

ESP8266

### Board

esp8285

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

pickonedev, 2025-03-23:
After removing the webserver, all good, but still, maybe I need the webserver, if someone knows a better workaround, please share.

AberDino, 2025-03-23:
I see the same issue with my Localbytes smart plugs (also ESP8266): "ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)". Unfortunately no webserver to disable for those plugs, so hopefully ESPHome 2023.3.2 will fix this...

MichaelMKKelly, 2025-03-23:
@AberDino 
> I see the same issue with my Localbytes smart plugs (also ESP8266): "ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove [...]

for the LocalBytes plug if you flash a minimal firmware you are then able to flash the normal firmware after.

if you change the package to the minimal file compile and write then revert it will work:

example where the main package is commented out and minimal is added

```
packages:
  #localbytes.plug-pm: github://LocalBytes/esphome-localbytes-plug/localbytes-plug-pm.yaml@main
  localbytes.plug-pm: github://LocalBytes/esphome-localbytes-plug/minimal.yaml@main
```
after writing then just revert the change and compile and write again

It would be a pain to do this every time but for now it is a workaround

swoboda1337, 2025-03-23:
> A file. Please try flashing a minimal firmware (remove everything except ota)". Unfortunately n

Remove stuff like `captive_portal`

AberDino, 2025-03-23:
@MichaelMKKelly and @swoboda1337, thank you for your responses. I should clarify that I purchased my Localbytes smart plugs in March 2024, pre-flashed with ESPHome. They have been part of my Home Assistant setup since, and generally when there is an ESPHome update, I apply it to all my ESPHome devices, and I've never had an issue updating these plugs. I'm aware of storage issues when migrating from Tasmota for example, but not when the device is already ESPHome. Going back to minimal firmware as an interim step each time there is an update is not very practical (I've got 11 of these plugs), so I should probably just revert to 2025.3.0 for now.

MichaelMKKelly, 2025-03-23:
@AberDino yeah its definitely an issue I have a few of them myself. I have raised 2 draft PR's in the LocalBytes repo with possible solutions to lower the firmware size. hopefully Adam will look into those or other possible solutions for a long term fix. in the coming days.

feel free to add to the discussion over there if you particularly like or dislike either solution:

https://github.com/LocalBytes/esphome-localbytes-plug/pull/12
https://github.com/LocalBytes/esphome-localbytes-plug/pull/13

AberDino, 2025-03-23:
@MichaelMKKelly, thanks for the additional information. I had a look at the draft PRs, and between the two, my vote would also be to remove IPv6 support (even if it is temporary), as I don't use it on my IoT VLAN. Interesting how a small tweak like that can make all the difference.

MichaelMKKelly, 2025-03-23:
@AberDino go mention it on the PR then. so Adam can see where the community sits

garyjohnwalker, 2025-03-25:
When I updated to 2025.3.0, I had to do the fix suggested above. Flash the minimal firmware, add the device back to the network, flash with 2025.3.0. I managed to figure that out myself but at that stage, I thought it was just me.

I now come to update to 2025.3.1 and I can't even flash the minimal firmware. Insufficient space.

I have 2 plugs like this. Are these things now stuck at 2025.3.0 or is there a way to clear space?

Failing that, is the tasmota minimal firmware smaller than the minimal ESP? I'm wondering if I could flash that (if I can find it).

pickonedev, 2025-03-25:
> When I updated to 2025.3.0, I had to do the fix suggested above. Flash the minimal firmware, add the device back to the network, flash with 2025.3.0. I managed to figure that out myself but at that [...]

Remove portal and/or web server from esp, at least till they will solve it

garyjohnwalker, 2025-03-25:
Tried removing the web server (as suggested higher up) but it's still too big.
I guess I'm just going to have to wait and cross my fingers.