esphome/issues #6892: LVGL align RIGHT_MID positions incorrectly

nottledim, 2025-03-23:
### The problem

Change from 2025.2.2 to 2025.3.1 the position of a widget has changed. 

I have a LVGL widget 
```         - led:
                  id: led_pulse
                  align_to:
                    id: show_time
                    align: RIGHT_MID
```
The widget it aligns to has width 100% and its contents are positioned correctly.  The led widget (above) was always aligned on the right hand edge. Since the update it has moved to left of centre.

### Which version of ESPHome has the issue?

2025.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

ESP32-IDF

### Board

Waveshare esp32-s3-touch-lcd-7

### Component causing the issue

lvgl

### YAML Config

```yaml
number:
  - platform: template
    name: LVGL Screen timeout
    optimistic: true
    id: display_timeout
    unit_of_measurement: "s"
    initial_value: 60
    restore_value: true
    min_value: 10
    max_value: 3600
    step: 5
    mode: box

font:
  - file: "gfonts://Roboto"
    id: roboto24
    size: 24
    bpp: 4
  - file: "gfonts://Roboto"
    id: roboto18
    size: 18
    bpp: 4
    
color:
  - id: clr_pv
    hex: 33FF44
  - id: clr_ev
    hex: 3333ff
  - id: clr_used
    hex: FF3340
  - id: clr_border
    hex: FF0000
  - id: clr_import
    hex: FF0000
  - id: clr_export
    hex: 00FF00
  - id: clr_bg_hdr
    hex: '111111'
  - id: clr_txt_hdr
    hex: ff7700
  - id: clr_txt_col_hdr
    hex: 'bbbbbb'
  - id: clr_bg_grid
    hex: 111111
  - id: clr_bg_cell
    hex: 7777ff
  - id: clr_txt_hdr2
    hex: 'bbbbbb'
  - id: clr_bg_page
    hex: '000000'
  - id: clr_volts
    hex: 'bbbbbb'
  - id: clr_unit
    hex: 777777

lvgl:
  displays:
    - wavey
  touchscreens:
    - touchy
  theme:
    button:
      bg_color: 0x333333
      bg_grad_dir: VER
      text_color: 0xFFFFFF
      checked:
        bg_color: 0xcc9900
        text_color: 0x000000
        bg_grad_color: 0x664d00
      pressed:
        border_color: 0xff6600
  pages:
    - id: pageish
      layout:
        type: flex
        flex_flow: COLUMN
      bg_color: clr_bg_page
      bg_opa: cover
      widgets:
        - obj:
            align: TOP_MID
            height: SIZE_CONTENT
            width: 100%
            bg_color: clr_bg_hdr
            widgets:
              - label:
                  id: title_text
                  width: 100%
                  text_align: CENTER
                  text_font: montserrat_24
                  text_color: clr_txt_hdr
                  text: FouterLec
              - label:
                  id: show_time
                  width: 100%
                  align_to:
                    id: title_text
                    align: OUT_BOTTOM_MID
                  text_align: CENTER
                  text_font: montserrat_16
                  text_color: clr_txt_hdr2
                  pad_top: 8
                  pad_bottom: 0
                  text: "now"
              - led:
                  id: led_pulse
                  align_to:
                    id: show_time
                    align: RIGHT_MID
        - obj:
            flex_grow: 1
            width: 100%
            height: SIZE_CONTENT
            bg_color: clr_bg_grid
            layout:
              type: grid
              grid_rows: [FR(1), FR(2), FR(2), FR(2), FR(1)]
              grid_columns: [FR(1), FR(1), FR(1), 48]
              grid_row_align: END
              pad_row: 24
            widgets:
```

### Anything in the logs that might be useful for us?

```txt

```

clydebarrow, 2025-03-25:
Although I can reproduce the difference between 2025.2.2 and the current dev, the thing is, nothing in the LVGL component changed between those, other than unrelated display driver changes. The generated code is identical between the two versions, so I have no idea why the difference, except...

The `align_to` feature in LVGL is notoriously fickle, and I recommend not using it. For your case I would suggest using a flow layout to wrap the two text lines, and the LED can simply be a child of the label it's associated with:

```
      widgets:
        - obj:
            align: TOP_MID
            layout:
              type: flex
              flex_flow: column
            width: 100%
            bg_color: clr_bg_hdr
            widgets:
              - label:
                  id: title_text
                  width: 100%
                  text_align: CENTER
                  text_font: montserrat_24
                  text_color: clr_txt_hdr
                  text: FouterLec
              - label:
                  id: show_time
                  width: 100%
                  text_align: CENTER
                  text_font: montserrat_16
                  text_color: clr_txt_hdr2
                  pad_top: 8
                  pad_bottom: 0
                  text: "now"
                  widgets:
                  - led:
                      id: led_pulse
                      align: RIGHT_MID
```

closed 2025-03-25 by clydebarrow

nottledim, 2025-03-25:
That's really helpful - and a much neater solution too.  Many thanks.