esphome/issues #6894: home assistant esphome 2025.3.0 and SH1306 64x48 shows corrupted fonts

SamuelCarson, 2025-03-23:
### The problem

After upgrade from 2025.2.2 to 2025.3.1 fonts are corrupted (missing pixels). sh1306 64x48 pixel oled display.

### Which version of ESPHome has the issue?

2025.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

20244.12.5

### What platform are you using?

ESP8266

### Board

D1 mini

### YAML Config

```yaml
esphome:
  name: "airgradient-test"

#  project:
#    name: ajfriesen.ESPHome-AirGradient
#    version: "1.0"

esp8266:
  board: d1_mini
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
- platform: esphome  
  password: !secret ota_password

# Create a switch for safe_mode in order to flash the device
# Solution from this thread:
# https://community.home-assistant.io/t/esphome-flashing-over-wifi-does-not-work/357352/1
switch:
  - platform: safe_mode
    name: "Flash Mode (Safe Mode)"

#dashboard_import:
## package_import_url: github://esphome/esphome-project-template/project-template-esp32.yaml@v6
#  package_import_url: github://ajfriesen/ESPHome-AirGradient/main/air-gradient.yaml

wifi:
  networks:
  - ssid: !secret wifi_iot_ssid
    password: !secret wifi_iot_password
  fast_connect: true  
# Optional manual IP
  #manual_ip:
  #  static_ip: 192.168.1.195
  #  gateway: 192.168.1.1
  #  subnet: 255.255.255.0
    
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Air-Gradient Fallback Hotspot"
    password: !secret fallback_ssid_password

captive_portal:

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      id: esp_ip_address

i2c:
  sda: D2
  scl: D1
  frequency: 400kHz  # 400kHz eliminates warnings about components taking a long time other than SGP40 component: https://github.com/esphome/issues/issues/4717

#font:
#  - file: "gfonts://Arimo@Regular"
#    id: arimo9
#    size: 9
#  - file: "gfonts://Arimo@Regular"
#    id: arimo10
#    size: 10

font:
  - file: "gfonts://Source+Code+Pro@Extra-Light"
    id: roboto
    size: 18

display:
  - platform: ssd1306_i2c
    id: oled
    model: "SSD1306 64x48"
    pages:
      - id: page1
        lambda: |-
          it.printf(0, 0, id(roboto), "CO2:");
          it.printf(2, 24, id(roboto), "%.0fppm", id(co2).state);
      - id: page2
        lambda: |-
          it.printf(0, 0, id(roboto), "PM25:");
          it.printf(2, 24, id(roboto), "%.0f", id(pm25).state);
      - id: page3
        lambda: |-
          it.printf(0, 0, id(roboto), "Hmdty:");
          it.printf(2, 24, id(roboto), "%.1f%%", id(humidity).state);
      - id: page4
        lambda: |-
          it.printf(0, 0, id(roboto), "Temp:");
            it.printf(2, 24, id(roboto), "%.fF", (id(temp).state)*9.0/5.0+32.0);

#    pages:
#      - id: page1
#        lambda: |-
#          it.printf(0, 0, id(arimo10), "CO2: %.0fppm", id(co2).state);
#          it.printf(0, 12, id(arimo10), "PM25: %.0f", id(pm25).state);
#          it.printf(0, 24, id(arimo10), "Hmdty: %.1f%%", id(humidity).state);
#          it.printf(0, 36, id(arimo10), "Temp: %.1fF", (id(temp).state)*9.0/5.0+32.0);
#      - id: page2
#        lambda: |-
#          it.printf(0, 0, id(arimo9), "%s", id(esp_ip_address).state.c_str());
#          it.printf(0, 12, id(arimo10), "PM10 : %.0f", id(pm10).state);
#          it.printf(0, 24, id(arimo10), "PM25 : %.0f", id(pm25).state);
#          it.printf(0, 36, id(arimo10), "PM100: %.0f", id(pm100).state);
# Maybe add a page later

# Provide a /metrics endpoint for Prometheus
prometheus:
web_server:
  port: 9926
  version: 1

interval:
  - interval: 5s
    then:
      - display.page.show_next: oled
      - component.update: oled

uart:
  - rx_pin: D5
    tx_pin: D6
    baud_rate: 9600
    id: uart_1
    
  - rx_pin: D4
    tx_pin: D3
    baud_rate: 9600
    id: uart_2

sensor:
#   - platform: sht3xd
#     temperature:
#       id: temp
#       name: "Temperature"
#     humidity:
#       id: humidity
#       name: "Humidity"
#     address: 0x44
#     update_interval: 5s

  - platform: pmsx003
    # type can be PMSX003, PMS5003S, PMS5003T, PMS5003ST
    # https://esphome.io/components/sensor/pmsx003.html
    type: PMS5003T
    uart_id: uart_1
    update_interval: 120s
    pm_1_0:
      id: pm10
    #   filters:
    #     - timeout: 100ms
      name: "Particulate Matter <1.0µm Concentration"
    pm_2_5:
      id: pm25
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      id: pm100
      name: "Particulate Matter <10.0µm Concentration"
    temperature:
     id: temp
     name: "Temperature in degrees Celcius"
    humidity:
     id: humidity
     name: "Relative Humidity in %"

  - platform: senseair
    uart_id: uart_2
    co2:
      id: co2
      name: "SenseAir CO2 Value"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
all looks normal
```

lcnittl, 2025-03-24:
Same on a `SSD1306 128x64` with the `Ubuntu` font, which rendered correctly pre `2025.3.0`. I suspect it has to do with https://github.com/esphome/esphome/pull/8300

There is another issue about the same problem https://github.com/esphome/issues/issues/6863 and it seems there is a fix on the way with https://github.com/esphome/esphome/pull/8407.

ssieb, 2025-03-24:
Yes, it's due to the change to freetype, but 8407 will not fix this.  A bitmap font would work much better at these sizes, but I'm poking around freetype a bit to see if there's a way to get better rendering.

ssieb, 2025-03-25:
Try the linked PR.

lcnittl, 2025-03-26:
@ssieb Fixed on my side (with 2025.3.2). Thanks for providing the PR!