esphome/issues #6896: Pip complaints during install

Sc00tz, 2025-03-24:
### The problem

I'm trying to compile a yaml config for a muse proto that I want to use as a voice assistant. I am getting this error when I go to install it:

INFO ESPHome 2025.3.1
INFO Reading configuration /config/esphome/muse-voice-assistant.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING `attenuation: 11db` is deprecated, use `attenuation: 12db` instead
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Compiling app...
Processing raspiaudio-muse-proto (board: esp-wrover-kit; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Installing ESP-IDF's Python dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
*** Error 1
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/raspiaudio-muse-proto/.pioenvs/raspiaudio-muse-proto/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:463 (message):
  /usr/bin/python3: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.19 seconds ==========================

### Which version of ESPHome has the issue?

2025.3.1

[... 359 lines cut ...]
    initial_option: "12 dB"
    on_value:
      - lambda: |-
          if (x == "3 dB") {
            pinMode(23, OUTPUT);
            digitalWrite(23, HIGH);
          } else if (x == "6 dB") {
            pinMode(23, OUTPUT);
            digitalWrite(23, LOW);
          } else if (x == "9 dB") {
            pinMode(23, INPUT);  // Disconnected
          } else if (x == "12 dB") {
            pinMode(23, OUTPUT);
            digitalWrite(23, LOW);
          } else if (x == "15 dB") {
            pinMode(23, OUTPUT_OPEN_DRAIN);
            digitalWrite(23, LOW);  // Connect to GND through 100kΩ
          }
```

### Anything in the logs that might be useful for us?

```txt
I did try this again with the current beta version of the add-on, same results so I went back to the normal release version.
```

MallocArray, 2025-03-26:
I'm struggling to see what the real error from pip is, but have you considered addressing several of the deprecated messages, such as the attenuation value?

Is there a reason to keep your frameworks pinned to specific versions that the messages say aren't recommended?
```
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.3.0
```

I'm running various devices with just straight esp-idf and letting esphome get the recommended one.  I did have to recently remove the rmt_channel parameter as it wasn't supported with the newest version, but was an easy modification.

Sc00tz, 2025-03-26:
@MallocArray I did try that too, but it didn't help. The problem is that it needs to install some pip packages, regardless what version I use, and for some reason the Add-On's container won't let me due to the package management, and there doesn't seem to be a virtual environment installed.

ssieb, 2025-03-26:
Did you try re-installing the add-on?

Sc00tz, 2025-03-27:
Yes, and tried the beta, and dev versions too, they all did the same thing. It was very weird.

closed 2025-08-08 by github-actions[bot]