esphome/issues #6897: ILI9488 [E][component:082]: Component display is marked FAILED

mateusz-jaworski, 2025-03-24:
### The problem

After configure 3,5" lcd screen 480x320 with ILI9488 driver to esp32 I've got an blank, white screen.
Everytime, when I try compile i've got the message: [E][component:082]:   Component display is marked **FAILED**

INFO ESPHome 2025.3.1
INFO Reading configuration /config/esphome/esph8.yaml...
INFO Starting log output from 192.168.5.114 using esphome API
INFO Successfully connected to esph8 @ 192.168.5.114 in 0.124s
INFO Successful handshake with esph8 @ 192.168.5.114 in 0.095s
[20:30:31][I][app:100]: ESPHome version 2025.3.1 compiled on Mar 24 2025, 20:12:26
[20:30:31][C][wifi:600]: WiFi:
[20:30:31][C][wifi:428]:   Local MAC: 40:91:51:FC:9F:20
[20:30:31][C][wifi:433]:   SSID: [redacted]
[20:30:31][C][wifi:436]:   IP Address: 192.168.5.114
[20:30:31][C][wifi:440]:   BSSID: [redacted]
[20:30:31][C][wifi:441]:   Hostname: 'esph8'
[20:30:31][C][wifi:443]:   Signal strength: -61 dB ▂▄▆█
[20:30:31][C][wifi:447]:   Channel: 6
[20:30:31][C][wifi:448]:   Subnet: 255.255.255.0
[20:30:31][C][wifi:449]:   Gateway: 192.168.5.1
[20:30:31][C][wifi:450]:   DNS1: 192.168.5.1
[20:30:31][C][wifi:451]:   DNS2: 9.9.9.9
[20:30:31][C][logger:177]: Logger:
[20:30:31][C][logger:178]:   Max Level: DEBUG
[20:30:31][C][logger:179]:   Initial Level: DEBUG
[20:30:31][C][logger:181]:   Log Baud Rate: 115200
[20:30:31][C][logger:182]:   Hardware UART: UART0
[20:30:31][C][spi:068]: SPI bus:
[20:30:31][C][spi:069]:   CLK Pin: GPIO18
[20:30:31][C][spi:070]:   SDI Pin: 
[20:30:31][C][spi:071]:   SDO Pin: GPIO23
[20:30:32][C][spi:076]:   Using HW SPI: SPI
[20:30:32][C][ili9xxx:091]: ili9xxx
[20:30:32][C][ili9xxx:091]:   Rotations: 0 °
[20:30:32][C][ili9xxx:091]:   Dimensions: 320px x 480px
[20:30:32][C][ili9xxx:092]:   Width Offset: 0
[20:30:32][C][ili9xxx:093]:   Height Offset: 0
[20:30:32][C][ili9xxx:102]:   Color mode: 8bit 332 mode
[20:30:32][C][ili9xxx:106]:   18-Bit Mode: YES
[20:30:32][C][ili9xxx:110]:   Reset Pin: GPIO17
[20:30:32][C][ili9xxx:111]:   CS Pin: GPIO16
[20:30:32][E][component:082]:   Component display is marked FAILED
[20:30:32][C][captive_portal:089]: Captive Portal:
[20:30:32][C][mdns:116]: mDNS:
[20:30:32][C][mdns:117]:   Hostname: esph8
[20:30:32][C][esphome.ota:075]:   Version: 2
[20:30:32][C][esphome.ota:078]:   Password configured

### Which version of ESPHome has the issue?

2025.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

ESP32

### Board

esp-wroom-32

### Component causing the issue

ili9xxx

### YAML Config

```yaml
esphome:
  name: esph8
  friendly_name: ESPH8

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Bmlm82WZzcQ7oIdsI4R9o5CZIK3kqOGeoCFOeu79E1Y="

ota:
  - platform: esphome
    password: "9d0683a5d3aac20809d597d04daefe7b"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esph8 Fallback Hotspot"
    password: "1LMzX9NNnhK6"

captive_portal:

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%

font:
  - file: "gfonts://Open Sans"
    id: roboto10
    size: 10

display:
  - platform: ili9xxx
    model: ILI9488
    dc_pin: GPIO4
    reset_pin: GPIO17
    cs_pin: GPIO16
    invert_colors: false
    show_test_card: true
    color_palette: 8BIT
    lambda: |-
      it.printf(64, 0, id(roboto10), "WOT Statistics");
      it.rectangle(1, 1, 100, 100, id(my_red));
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-03-24:
Can you get serial logs from boot?

clydebarrow, 2025-03-25:
Basically the 9488 is just too big a display for an ESP32 without PSRAM. Even in 8 bit mode, the buffer for a 480x320 display needs 153600 bytes, and there is not a big enough block in the heap for that when Wi-Fi is in use.

It will work with LVGL and a `buffer_size: 25%` but not with the native display functions, which need a full-size buffer.

closed 2025-03-31 by clydebarrow