esphome/issues #6898: waveshare_epaper 2.90inv2-r2 does not work after ESP32C3 deep sleep

GraffJosh, 2025-03-24:
### The problem

Using the waveshare_epaper 2.90inv2-r2 with a Xaio ESP32C3 works great until I put the device in deep sleep. After wake it no longer updates the display until I power cycle the entire device. 

I tried using https://github.com/esphome/esphome/pull/6209 with this and 4.2" display. The 4.2 will work after deepsleep (though it does go through a full refresh sequence which is less than ideal). This 2.9 won't work at all. 

### Which version of ESPHome has the issue?

2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3

### What platform are you using?

ESP32

### Board

SEEED Xaio ESP32C3

### Component causing the issue

waveshare_epaper

### YAML Config

```yaml
esphome:
  name: dinghy
  friendly_name: Dinghy
  on_boot:
    - priority: 600
      then:
      - deep_sleep.prevent: very_deep_sleep

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: arduino
deep_sleep:
  id: very_deep_sleep
  run_duration: 
    default: 
      minutes: 2
  sleep_duration: 
    hours: 1
  wakeup_pin_mode: KEEP_AWAKE
  wakeup_pin: 
    number: GPIO2
    allow_other_uses: true
    inverted: true

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO2
      allow_other_uses: true
      inverted: True
      mode:
        input: True
        pullup: True
    name: "Button"

    on_double_click: 
      then:
      - globals.set: 
          id: last_click
          value: "2"
      - light.turn_on:
          id: button_light
          brightness: 100%
          red: 100%
          green: 0%
          blue: 0%
      - delay: 2s
      - light.turn_off:
          id: button_light
      - deep_sleep.allow: very_deep_sleep
      - deep_sleep.enter: very_deep_sleep
spi:
  clk_pin: GPIO9
  mosi_pin: GPIO10
display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: 
      number: GPIO7
    dc_pin: GPIO5
    reset_pin: GPIO21
    busy_pin: GPIO6
    reset_duration: 2ms
    model: 2.90inv2-r2
    rotation: 90
    update_interval: 10s
    full_update_every: 30
    auto_clear_enabled: True
    pages:
      - id: page1
        lambda: |-
          int x = 10;
          int y = 0;
          it.strftime(x, y, id(robotoSmall), TextAlign::TOP_LEFT, "%H:%M", id(platformtime).now());
```

### Anything in the logs that might be useful for us?

```txt

```

dsamarin1, 2025-04-17:
I don't have a solution, but FWIW, I am using the same setup as you: xiao esp32-c3 and 2.9inch display with "V2" sticker and "rev2.1" on the board; though i'm fighting a different problem, mirrored text and graphics along x-axis, at the moment. Which version markers does your display have?

I just copied your `display:` config (besides pin numbers and the lambda) and I am able to `it.print()` a hello world message just fine. So there's a chance you are fighting some other issue.

BTW, I don't believe you need the busy_pin, at least I don't.

GraffJosh, 2025-04-17:
I'm able to run the display just fine, the issue is in deep-sleep recovery. Are you able to deep-sleep your device?