esphome/issues #6903: Wrong font TextAlignment in 2025.3.x

yatush, 2025-03-26:
### The problem

TextAlign::CENTER is not rendered correctly after the upgrade to 2025.3.x

2025.2.0 output:
[image]

2025.3.2 output:
[image]

### Which version of ESPHome has the issue?

2025.3.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.x

### What platform are you using?

ESP32

### Board

ESP32-2432S028R

### YAML Config

```yaml
substitutions:
  device_name: "PLACEHILDER"
  friendly_name: Placeholder

esphome:
  name: $device_name
  friendly_name: $friendly_name

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  id: api_server

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

time:
  - platform: homeassistant
    id: esptime

spi:
  - id: tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

uart:
  tx_pin: GPIO22
  rx_pin: GPIO27
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

font:
  - file: "gfonts://Material+Symbols+Outlined"
    id: mdi_large
    size: 80
    glyphs: [
      "\U0000eacf", # Arrow up
      "\U0000eb97", # Display settings
    ]

display:
  - platform: ili9xxx
    update_interval: 1s
    model: ili9341
    spi_id: tft
    color_palette: 8BIT
    cs_pin: GPIO15
    dc_pin: GPIO2
    id: disp
    transform: { mirror_y: true }
    dimensions:
      width: 320
      height: 240
    invert_colors: false
    lambda: |-
      it.rectangle(10,  10, 100, 100, id(Color::WHITE));
      it.print(60, 60, id(mdi_large), Color::WHITE, TextAlign::CENTER, "\U0000eacf");
      it.rectangle(10,  110, 100, 100, id(Color::WHITE));
      it.print(60, 160, id(mdi_large), Color::WHITE, TextAlign::CENTER, "\U0000eb97");
```

### Anything in the logs that might be useful for us?

```txt
Logs look good, the output display does not look as expected
```

yatush, 2025-03-26:
@clydebarrow might be related to recent changes made to font rendering?

MallocArray, 2025-03-26:
2025.3.2 came out last night that addressed some other font related issues.  Have you tested with that specific version?

yatush, 2025-03-26:
Yes, this was tested a few hours ago with 2025.3.2 - I was hoping it would fix it, but this still happens in the latest version.

clydebarrow, 2025-03-26:
> [@clydebarrow](https://github.com/clydebarrow) might be related to recent changes made to font rendering?

It is quite likely related, I will investigate.

clydebarrow, 2025-03-26:
Turned out to be a pre-existing bug in the display component, was never triggered until the font generation got smarter.

closed 2025-03-27 by kbx81