esphome/issues #6904: Issue with BLE since 2025.3

gapthorpe, 2025-03-27:
### The problem

This was raised on the following ticket:
https://github.com/home-assistant/core/issues/141016#issuecomment-2747574532

ESPHome upgraded to 2025.3.0 on both my Bluetooth proxy devices. Since then I am getting this error:
Failed setup, will retry: Unable to connect to device Xx:XX:XX due to esp32 [XX:XX:XX]: XX:XX is not connected

Looking in the ESP32 logs can see the following errors which weren’t there before:
[13:30:50][I][esp32_ble_client:147]: [0] [XX:XX:XX:XX] Disconnecting (conn_id: 0).
[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752041) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752045) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752051) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752059) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752070) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[13:30:50][D][esp32_ble_client:191]: [0] [04:EE:03:15:2D:1E] ESP_GATTC_CLOSE_EVT
[13:30:50][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[13:30:50][D][esp-idf:000][BTU_TASK]: E (104752314) BT_BTM: Device not found

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752317) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[13:30:50][D][esp-idf:000][BTU_TASK]: W (104752319) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: XX:XX:XX:XX

Any advice on this issue?  Thanks!

### Which version of ESPHome has the issue?

2025.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3

### What platform are you using?

ESP32-IDF

### Board

M5 Atom

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

rinyheuvelman, 2025-03-30:
Same here: everything worked perfectly fine until I updated to 2025.3.0
Since then I’ve spent hours trying to figure out what goes wrong and how to fix this. 
I’ve 3 Bluetooth proxy devices, the issue disappears with a system reboot, only to return over and over again: 

Logger: homeassistant.components.esphome.manager
Bron: components/esphome/manager.py:385
integratie: ESPHome ([documentatie](https://www.home-assistant.io/integrations/esphome), [problemen](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+esphome%22))
Eerst voorgekomen: 00:19:02 (180 gebeurtenissen)
Laatst gelogd: 11:25:10

Bluetooth Proxy 0d7d08: [W][api.connection:135]: ESPHome Logs 2025.3.2 (192.168.178.44): Connection closed
Bluetooth Proxy 52c168: [W][esp32_ble_client:137]: [0] [74:8D:E8:3C:9F:49] Disconnecting before connected, disconnect scheduled.
Aquarium Lights: [W][component:237]: Component api took a long time for an operation (53 ms).
Aquarium Lights: [W][component:238]: Components should block for at most 30 ms.
Aquarium Lights: [W][api.connection:135]: ESPHome Logs 2025.3.2 (192.168.178.44): Connection closed

Please work this out!
For now I’m reverting to the previous version

Ladida1, 2025-04-27:
I have the same issue. Tried updating ESP to 2025.4.0 but still the same issue.

[20:47:36][I][bluetooth_proxy:286]: [0] [60:98:66:EB:C7:34] Connecting v3 without cache
[20:47:36][D][esp32_ble_tracker:115]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
[20:47:36][D][esp32_ble_tracker:231]: Pausing scan to make connection...
[20:47:36][I][esp32_ble_client:110]: [0] [60:98:66:EB:C7:34] 0x00 Attempting BLE connection
[20:47:36][D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_CONNECT_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_OPEN_EVT
[20:47:37][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[20:47:37][D][esp32_ble_tracker:282]: Starting scan...
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_SEARCH_CMPL_EVT
[20:47:37][I][esp32_ble_client:337]: [0] [60:98:66:EB:C7:34] Connected
[20:47:37][D][esp32_ble_client:296]: [0] [60:98:66:EB:C7:34] cfg_mtu status 0, mtu 65
[20:47:37][I][esp32_ble_client:442]: [0] [60:98:66:EB:C7:34] auth complete. remote BD_ADDR: 609866ebc734
[20:47:37][E][esp32_ble_client:445]: [0] [60:98:66:EB:C7:34] auth fail reason = 0x52
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_READ_CHAR_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_READ_CHAR_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_READ_CHAR_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_READ_CHAR_EVT
[20:47:37][W][bluetooth_proxy.connection:067]: [0] [60:98:66:EB:C7:34] Error reading char/descriptor at handle 0x E, status=15
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_READ_CHAR_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_REG_FOR_NOTIFY_EVT
[20:47:37][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_WRITE_DESCR_EVT
[20:47:37][W][bluetooth_proxy.connection:085]: [0] [60:98:66:EB:C7:34] Error writing char/descriptor at handle 0x F, status=15
[20:47:42][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_WRITE_CHAR_EVT
[20:47:42][D][esp32_ble_client:191]: [0] [60:98:66:EB:C7:34] ESP_GATTC_WRITE_CHAR_EVT
[20:47:52][D][esp32_ble_client:416]: [0] [60:98:66:EB:C7:34] Event 39
[20:47:52][I][esp32_ble_client:147]: [0] [60:98:66:EB:C7:34] Disconnecting (conn_id: 0).
[20:47:52][D][esp-idf:000][BTU_TASK]: W (229013) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

bdraco, 2025-04-27:
I recently fixed a race that would prevent connections from happening if the device took a long time to connect
https://github.com/esphome/esphome/pull/8597. Its scheduled for 2025.4.1 (not yet released)

There is race in the scanner as well which is addressed in https://github.com/esphome/esphome/pull/8601. It is scheduled for 2025.5.0 (not yet released)