esphome/issues #6908: Multiple displays with pages can not resolve which display to use when switching pages

pgtest, 2025-03-30:
### The problem

Using an ESP32 with two  GC9A01A (driver ili9xxx) displays on SPI bus. 

Works and compiles when I do not add page switching, loading or similar. 

When adding for example lvgl.page.next the compiler throws : Too many candidates found for 'lvgl_id' type 'lvgl::LvglComponent' Some are 'lvgl_1', 'lvgl_2'. 
These are defined as ids under lvgl: and each have displays: display_1 and display_2 respectively. so the pages should be callable.

In the code at the bottom.
use logger ... no problem.
comment out logger and add lvgl.page.next instead - compiler error as above. (lvgl.page.load etc same ... can not find unique reference)

might be user error.... 

### Which version of ESPHome has the issue?

2025.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

ESP32

### Board

ESp32 dev kit C

### Component causing the issue

Lvgl multiple displays and pages

### YAML Config

```yaml
esphome:
  name: esp32-round-display-1
  friendly_name: ESP32-Round-display-1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "zZ0qZ781+X1f+8IOyPRI0dfSGIRYzWHpxhNbWttZhDk="

ota:
  - platform: esphome
    password: "4f9e6e243285579954b791adabbe8c14"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Round-Display-1"
[... 307 lines cut ...]
          id: sunpower_needle2
          value: !lambda return x;
      - lvgl.label.update:
          id: sunpower_text2
          text:
            format: "%.3f kW"
            args: [ 'x/1000' ]

  - platform: rotary_encoder
    name: "Rotary Encoder"
    pin_a: GPIO13
    pin_b: GPIO14
    on_clockwise:
#      then:
#        lvgl.page.next: 
      - logger.log: "Turned Clockwise"
```

### Anything in the logs that might be useful for us?

```txt
Attempted usage: 2 displays ... one show "static" info i.e. no pages, and the other multiple pages switchable with rotary knob.

SPI bus is single with two devices with their own Chip Select.
```

ssieb, 2025-03-30:
You have to specify the lvgl id to use.  The error message tells you what it needs and it's explained in the docs:
https://esphome.io/components/lvgl/#actions

closed 2025-03-30 by ssieb

pgtest, 2025-03-31:
Thx for the clarification... I did not understand that it was implied to work for any of the other commands in "Actions" than noted: lvgl.widget.redraw, lvgl.pause and lvgl.resume has notes about it, but NOT on the rest of the commands on the documentation page. 

Works... when you knows it..
Thx