esphome/issues #6909: generic-rtl8720cf-2mb-992k doesn't seem to have valid pins for hardware UART 0 and 1

fcol95, 2025-03-31:
### The problem

Trying to use UART0 on a rtl8720cf config (WBR3) board.
It seems whatever I do, I can't seem to have it working.
Both for a generic uart component and for the logger component with UART0 as hardware_uart.
The first gives an error when setupping (see below), the second just doesn't output any logging on the interface.
When trying to set it up for the generic uart component, getting this error where it can't associate to the proper hardware pins, and trigger the "SoftwareSerial is not implemented for this chip" error.
The log below can't seem to find and print the proper pins for UART 0 or 1 too (UART 2 seems to be correct though, and I manage to have the logger output on that one, but my component is on UART0).
Here's the error:
```
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][scheduler:063]: set_interval(name='', interval=60000, offset=2966)
[C][uart.lt:049]: Setting up UART...
[E][uart.lt:081]:   SoftwareSerial is not implemented for this chip. Only hardware pins are supported:
[E][uart.lt:083]:     TX=255, RX=255
[E][uart.lt:086]:     TX=255, RX=255
[E][uart.lt:089]:     TX=16, RX=15
[E][component:119]: Component uart was marked as failed.
[E][component:164]: Component uart set Error flag: unspecified
[C][wifi:048]: Setting up WiFi...
```
Here's the config extract:
```
uart:
 - id: hlw8112_uart
   tx_pin: PA14
   rx_pin: PA13
   baud_rate: 38400
   data_bits: 8
   stop_bits: 1
```
Pin association found in the WBR3 specification here: https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy.

Odly enough it seems to be only board component in the code base that has a "_0" and "_1" suffixes for both 
SERIAL0_RX/SERIAL0_TX and SERIAL1_RX/SERIAL1_TX entries in the dictionary in "boards.py" file. Could be related?
[image]

### Which version of ESPHome has the issue?

2025.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

RTL87XX

### Board

WBR3

### Component causing the issue

UART

### YAML Config

```yaml
esphome:
  name: pow-meter-pc311
  friendly_name: pow-meter-pc311
  on_boot:
    - priority: 600
[... 283 lines cut ...]
[V][mdns:118]:   Services:
[V][mdns:120]:   - _esphomelib, _tcp, 6053
[V][mdns:122]:     TXT: friendly_name = pow-meter-pc311-water-heater
[V][mdns:122]:     TXT: version = 2025.3.2
[V][mdns:122]:     TXT: mac = b8060d4a7dfb
[V][mdns:122]:     TXT: platform = RTL8720CF
[V][mdns:122]:     TXT: board = generic-rtl8720cf-2mb-992k
[V][mdns:122]:     TXT: network = wifi
[V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[V][mdns:120]:   - _http, _tcp, 80
[C][esphome.ota:073]: Over-The-Air updates:
[C][esphome.ota:074]:   Address: pow-meter-pc311-water-heater.local:8892
[C][esphome.ota:075]:   Version: 2
[C][esphome.ota:078]:   Password configured
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:019]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:022]:   Remain in safe mode for 300 seconds
[C][api:140]: API Server:
[C][api:141]:   Address: pow-meter-pc311-water-heater.local:6053
[C][api:143]:   Using noise encryption: YES
[C][lt.component:013]: LibreTiny:
[C][lt.component:014]:   Version: v1.7.0+sha.55aacc8 on generic-rtl8720cf-2mb-992k, compiled at Mar 30 2025 22:12:02, GCC 10.3.1 (-Os)
[C][lt.component:015]:   Loglevel: 3
```

plplaaa2, 2025-04-16:
I've never used WBR3, but I've uploaded libretiny esphome to the mhcw02p chip that used the rtl8720cn chip as the base.  At that time, I used GPIO16, GPIO15 in TX/RX and it worked well.  When I tested 13, 14, I didn't try it, but I knew it was a UART pin.

[image]

As a dual ch wifi meter user, I hope your work goes well.

fcol95, 2025-04-17:
Yes UART2 through pins 15 and 16 seems to work well (as seen in the log message I've shared above), however would like to keep them free for the boatload and logger usage.
Would be curious if you could give a go to use the gpio13/14 on your mhcw02p chip!
But i think we might need to create a board config for those chip and assign serial0_rx/tx to gpio13/14 only in their board.py file.

plplaaa2, 2025-04-18:
> Yes UART2 through pins 15 and 16 seems to work well (as seen in the log message I've shared above), however would like to keep them free for the boatload and logger usage. Would be curious if you [...]

Unfortunately, the chip is now a brick. I think I made a mistake during the test. It is a communication chip originally attached to a Xiaomi device, but it has become a brick and is being used as an ESP32. You'd better consider it as one of the options if you can't.

Pwedge, 2025-07-04:
@fcol95 I just ran into the same issue today.  My digging shows that the reason that this is happening is because the "generic-rtl8720cf-2mb-992k" has the following #defines in libretiny:
`$LIBRETINY/boards/variants/generic-rtl8720cf-2mb-992k.h:
#define PIN_SERIAL0_RX_0 12u // PIN_A12
#define PIN_SERIAL0_RX_1 13u // PIN_A13
#define PIN_SERIAL0_TX_0 11u // PIN_A11
#define PIN_SERIAL0_TX_1 14u // PIN_A14
// [pwedge-note] PIN_SERIAL0_RX and PIN_SERIAL0_TX are never defined when using this board type

$LIBRETINY/cores/common/base/lt_pins.h:
#define PIN_INVALID 255
...(snip)...
#if LT_HW_UART0
#ifndef PIN_SERIAL0_RX
#define PIN_SERIAL0_RX PIN_INVALID
#endif
#ifndef PIN_SERIAL0_TX
#define PIN_SERIAL0_TX PIN_INVALID
#endif
#endif`

You could change the #defines in libretiny or hack up esphome, but I took an easier approach.  I changed the board type to "bw15", since that is also RTL8720CF-based and has the following #defines:
`#define PIN_SERIAL0_RX   13u // PIN_A13
#define PIN_SERIAL0_TX   14u // PIN_A14`

After re-running esphome with the bw15 board type, I now seem to have a functional UART using pins GPIO13/GPIO14!