esphome/issues #6911: Media player does not go back to "idle" after playing TTS announcement since 2025.3.3

MacrosorcH, 2025-04-01:
### The problem

Since ESPHome 2025.3.3, after playing a TTS announcement, the media player of my Raspiaudio Muse Luxe remains in status "playing" and doesn't go back to "idle" unless I send a stop command. It worked well until 2025.3.2.

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Raspiaudio Muse Luxe

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

MacrosorcH, 2025-04-01:
Looking at the 2025.3.3 changelog, it might be a regression caused by https://github.com/esphome/esphome/pull/8474 and/or  https://github.com/esphome/esphome/pull/8488

MacrosorcH, 2025-04-01:
It looks like the problem is actually caused by HA 2025.4.0b12. I first tried rolling back to ESPHome 2025.3.2 but it did not help, while rolling back to HA 2025.3.4 resolved the issue.

MacrosorcH, 2025-04-01:
Closed issue here because not ESPHome-specific. I opened a new issue on HA Core instead: https://github.com/home-assistant/core/issues/142022

ammgws, 2025-04-18:
@MacrosorcH 
>rolling back to HA 2025.3.4 resolved the issue.

Can you please clarify the version? The latest 2025.3.x release I can find on https://github.com/esphome/esphome/releases is 2025.3.3

There are not that many changes between 2025.3.2 and 2025.3.3 so we should be able to bisect it easily:
https://github.com/esphome/esphome/compare/2025.3.2..2025.3.3

MacrosorcH, 2025-04-18:
> [@MacrosorcH](https://github.com/MacrosorcH)
> 
> > rolling back to HA 2025.3.4 resolved the issue.
> 
> Can you please clarify the version? The latest 2025.3.x release I can find on https://github.com/esphome/esphome/releases is 2025.3.3

@ammgws HA 2025.3.4, not ESPHome. Regardless of the ESPHome version, the problem started with HA 2025.4.0 and is still unresolved on HA 2025.4.2. If you rollback HA to any release before April you'll see the problem disappear.

RichardA1, 2025-05-12:
I added a feature request that could help solve this issue:
https://github.com/esphome/feature-requests/issues/3148

MacrosorcH, 2025-05-19:
I am reopening this issue since the problem, extensively discussed in https://github.com/home-assistant/core/issues/142022, turned out being actually a ESPHome bug that this PR is meant to fix: https://github.com/esphome/ESP32-audioI2S/pull/19

Hoping someone from the ESPHome team can finally take a look at the PR and release the fix with ESPHome 2025.5.0?