esphome/issues #6912: sn74hc595 missing Implementation of get_flags()

bbadaboom, 2025-04-02:
### The problem

When trying to compile my previously working yaml which uses a sn74hc595 I get the error below.
I assume sn74hc595 is missing the implementation of get_flags() from GPIOPin.

`src/main.cpp: In function 'void setup()':
src/main.cpp:521:67: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id = new sn74hc595::SN74HC595GPIOPin();
                                                                   ^
In file included from src/esphome.h:52,
                 from src/main.cpp:3:
src/esphome/components/sn74hc595/sn74hc595.h:46:7: note:   because the following virtual functions are pure within 'esphome::sn74hc595::SN74HC595GPIOPin':
 class SN74HC595GPIOPin : public GPIOPin {
       ^~~~~~~~~~~~~~~~
In file included from src/esphome/core/hal.h:4,
                 from src/esphome/core/application.h:7,
                 from src/esphome/components/api/api_connection.h:9,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/gpio.h:61:23: note: 	'virtual esphome::gpio::Flags esphome::GPIOPin::get_flags() const'
   virtual gpio::Flags get_flags() const = 0;
                       ^~~~~~~~~
src/main.cpp:550:69: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_2 = new sn74hc595::SN74HC595GPIOPin();`

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.3.4

### What platform are you using?

ESP32

### Component causing the issue

sn74hc595

### YAML Config

```yaml
sn74hc595:
  - id: 'sn74hc595_hub'
    data_pin: GPIO14
    clock_pin: GPIO13
    latch_pin: GPIO12
    oe_pin: GPIO05
    sr_count: 2
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function 'void setup()':
src/main.cpp:521:67: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id = new sn74hc595::SN74HC595GPIOPin();
                                                                   ^
In file included from src/esphome.h:52,
                 from src/main.cpp:3:
src/esphome/components/sn74hc595/sn74hc595.h:46:7: note:   because the following virtual functions are pure within 'esphome::sn74hc595::SN74HC595GPIOPin':
 class SN74HC595GPIOPin : public GPIOPin {
       ^~~~~~~~~~~~~~~~
In file included from src/esphome/core/hal.h:4,
                 from src/esphome/core/application.h:7,
                 from src/esphome/components/api/api_connection.h:9,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/gpio.h:61:23: note: 	'virtual esphome::gpio::Flags esphome::GPIOPin::get_flags() const'
   virtual gpio::Flags get_flags() const = 0;
                       ^~~~~~~~~
src/main.cpp:550:69: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_2 = new sn74hc595::SN74HC595GPIOPin();
```

clydebarrow, 2025-04-07:
Are you using any external components? The error does not reproduce.

bbadaboom, 2025-04-12:
Hi, sorry for the late reply. I had been using an external component, but removed it about 3 Months ago. Had to come home to clean the build files and recompile, just to be sure there are no traces of it left. I still get the error.
Full build logs here:

```
INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/pool-steuerung.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing pool-steuerung (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.4
INFO Installing esphome/AsyncTCP-esphome @ 2.1.4
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
INFO AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.3.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.3.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/cover/cover.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/dht/dht.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling .pioenvs/pool-steuerung/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
[... 118 lines cut ...]
src/main.cpp:835:70: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_11 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
src/main.cpp:864:70: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_12 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
src/main.cpp:893:70: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_13 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
src/main.cpp:922:70: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_14 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
src/main.cpp:951:70: error: invalid new-expression of abstract class type 'esphome::sn74hc595::SN74HC595GPIOPin'
   sn74hc595_sn74hc595gpiopin_id_15 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
[... cut ...]
   sn74hc595_sn74hc595gpiopin_id_16 = new sn74hc595::SN74HC595GPIOPin();
                                                                      ^
Compiling .pioenvs/pool-steuerung/lib64d/WiFi/WiFiClient.cpp.o
Compiling .pioenvs/pool-steuerung/lib64d/WiFi/WiFiGeneric.cpp.o
Archiving .pioenvs/pool-steuerung/lib579/libAsyncTCP-esphome.a
Compiling .pioenvs/pool-steuerung/lib64d/WiFi/WiFiMulti.cpp.o
*** [.pioenvs/pool-steuerung/src/main.cpp.o] Error 1
========================= [FAILED] Took 101.47 seconds =========================
```

clydebarrow, 2025-04-12:
Post your complete yaml. The fragment you showed above does not generate the error on its own.

bbadaboom, 2025-04-13:
I stripped down the yaml removing all the logic stuff - this gives the same error messag:

```
substitutions:
  name: "pool-steuerung"
  friendly_name: "Pool Steuerung"
  friendly: "Pool"

  relay_salz:         "relay01"
  relay_uv:           "relay02"
  relay_robot:        "relay03"
  relay_licht:        "relay04"
  relay_pump_slow:    "relay05"
  relay_pump_med:     "relay06"
  relay_pump_fast:    "relay07"
  relay_salz_flow:    "relay08"
  relay_cover_close:  "relay09"
  relay_cover_open:   "relay10"
  relay_cover:        "relay11"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pool-Steuerung Fallback Hotspot"
    password: "xxxx"

captive_portal:

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime"
    update_interval: 5min
    disabled_by_default: true

  - platform: dht
    pin: GPIO32
    model: DHT22
    temperature:
      name: "${friendly} Temperatur Innen"
    humidity:
      name: "${friendly} Luftfeuchtigkeit Innen"
      accuracy_decimals: 1
    
sn74hc595:
  - id: 'sn74hc595_hub'
    data_pin: GPIO14
    clock_pin: GPIO13
    latch_pin: GPIO12
    oe_pin: GPIO05
    sr_count: 2

switch: 
  ## INTERNAL RELAYS
  - platform: gpio
    id: relay01
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
  - platform: gpio
    id: relay02
    pin:
      sn74hc595: sn74hc595_hub
      number: 1
  - platform: gpio
    id: relay03
    pin:
      sn74hc595: sn74hc595_hub
      number: 2
  - platform: gpio
    id: relay04
    pin:
      sn74hc595: sn74hc595_hub
      number: 3
  - platform: gpio
    id: relay05
    pin:
      sn74hc595: sn74hc595_hub
      number: 4
  - platform: gpio
    id: relay06
    pin:
      sn74hc595: sn74hc595_hub
      number: 5
  - platform: gpio
    id: relay07
    pin:
      sn74hc595: sn74hc595_hub
      number: 6
  - platform: gpio
    id: relay08
    pin:
      sn74hc595: sn74hc595_hub
      number: 7
  - platform: gpio
    id: relay09
    pin:
      sn74hc595: sn74hc595_hub
      number: 8
  - platform: gpio
    id: relay10
    pin:
      sn74hc595: sn74hc595_hub
      number: 9
  - platform: gpio
    id: relay11
    pin:
      sn74hc595: sn74hc595_hub
      number: 10
  - platform: gpio
    id: relay12
    pin:
      sn74hc595: sn74hc595_hub
      number: 11
  - platform: gpio
    id: relay13
    pin:
      sn74hc595: sn74hc595_hub
      number: 12
  - platform: gpio
    id: relay14
    pin:
      sn74hc595: sn74hc595_hub
      number: 13
  - platform: gpio
    id: relay15
    pin:
      sn74hc595: sn74hc595_hub
      number: 14
  - platform: gpio
    id: relay16
    pin:
      sn74hc595: sn74hc595_hub
      number: 15

```

clydebarrow, 2025-04-13:
Compiles just fine for me - you didn't show your build log.

clydebarrow, 2025-04-13:
That compiles just fine for me, but it's evident from your build log that the copy of the source you are building from is not up-to-date:

```
src/esphome/components/sn74hc595/sn74hc595.h:46:7
```

This line number is not (now) the class declaration, so the copy you have is out of date. Did you clean the build? You might also want to re-install the add-on, because something is wrong with your setup.

bbadaboom, 2025-04-13:
Yes, I cleaned the build files several times...
I have now uninstalled and reinstalled the addon which did the trick! Many thanks and sorry for the confusion - must have been somthing wrong in my update.

closed 2025-04-13 by clydebarrow