esphome/issues #6913: MR60BHA2: missing people_exist and target_num

rotilho, 2025-04-03:
### The problem

According to the documents, MR60BHA2 should support `people_exist` and `target_num`; however, that doesn't seem to be the case. The UI says they don't exist, and the compilation also fails.

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

esp32-c6-devkitm-1

### Component causing the issue

seeed_mr60bha2

### YAML Config

```yaml
esphome:
  name: example-mr60bha2
  friendly_name: Example MR60BHA2

esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  framework:
    type: esp-idf

# api:
#   encryption:
#     key: !secret encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  use_address: 192.168.1.45
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "RGB Light"
    pin: GPIO1
    num_leds: 1
    # rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:

binary_sensor:
  - platform: seeed_mr60bha2
    people_exist:
      name: "MR60BHA2 People Exist"

sensor:
  - platform: bh1750
    name: "BH1750 Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "MR60BHA2 Breath Rate"
    heart_rate:
      name: "MR60BHA2 Heart Rate"
    distance:
      name: "MR60BHA2 Distance"
    target_num:
      name: "MR60BHA2 Target Number"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.3.3
INFO Reading configuration /config/example-mr60bha2.yaml...
Failed config

binary_sensor.seeed_mr60bha2: [source /config/example-mr60bha2.yaml:53]
  platform: seeed_mr60bha2
  
  [people_exist] is an invalid option for [binary_sensor.seeed_mr60bha2]. Please check the indentation.
  people_exist: 
    name: MR60BHA2 People Exist
```

lpadumorvan, 2025-04-12:
Hi

Change "people_exist" to "has_target"
Change "target_num" to "num_targets"

source : https://esphome.io/components/seeed_mr60bha2.html

DV

rotilho, 2025-04-12:
That works!! Very thanks 🙏

closed 2025-04-12 by rotilho

lpadumorvan, 2025-04-13:
latest code here :

https://github.com/limengdu/MR60BHA2_ESPHome_external_components/blob/main/example/mr60bha2.yaml

DV