esphome/issues #6918: OTA Brickes several chip models, eg esp8285 and bk72xx

stofakiller, 2025-04-04:
### The problem

Hi,

After the several updates (im always on the latest version) often i just do an update and/or sometimes just change a few linis of code... But after max 5 updates they get totally bricked and the onboard blinks once every 1 minute

The strange is that i then solder TX,RX and power and ground and flash them trough a serial FTL and then the same code that "got the device bricked" works again, but then again after a couple up to 5 times, it get bricked again...

But also a have found out that i have to clear build files almost every time i use an esp8266/32v1 with i2c and spi display, because it didnt update my change on the display, but after clear build files, my update works

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.4.0

### What platform are you using?

BK72XX

### Board

Tuya branded

### Component causing the issue

gpio

### YAML Config

```yaml
Even with only wifi in the code it gets bricked after a couple up for exactly same code... And So far only with bk72xx and 8285 chips
```

### Anything in the logs that might be useful for us?

```txt
No logs, because they are bricked, but after solder lines to the chip and flash the same code with FTl and ESPHome WEB, they work again
```

### Additional information

Im running esphome on Proxmox PC with plenty of power, esphome can use up to 8 cores and 64GB of RAM

ssieb, 2025-04-08:
What do the serial logs show at that point?

stofakiller, 2025-04-08:
I can try with the esp8285, but I can't find anything about the bk72xx a Realtek RMS w302 chip (ubranded board), So I can't connect/reflash those (lost 5 of them the last couple of months)...Venlig hilsen/Best Regards,Jan Larsen

clydebarrow, 2025-04-11:
I tested OTA to an ESP8266 multiple times with no problem.

clydebarrow, 2025-04-11:
Well, the 8285 is an ESP8266 - zero difference in the software.

clydebarrow, 2025-04-11:
and I just tested on an 8285, same result - 15 OTA uploads in a row, no problem.

stofakiller, 2025-04-11:
I normally don't have problems, but occasionally they die after an OTA... Do you use on_boot and priority?Venlig hilsen/Best Regards,Jan Larsen

stofakiller, 2025-05-20:
Just happen again...

I updated 2 similar BW-SHP7... And they have same code... Only difference is name and manual IP... One worked and the other one is now bricked... I'll disassemble it later and soilder a serial usb on it to se if there is any log comming out...

But it has happened SO often and is very annoying and some other devices I can't find the pins to connect and manual reflash them... Because they were flashed with cloud cutter and OTA with esphome...